nodemcu/nodemcu-firmware #696: dev140: file.format() results in wdt reset.

karrots, 2015-10-20:
Calling file.format() now seems to trigger a WDT reset. Worked prior to the 1.4.x SDK integration #687 pull request.

eku, 2015-10-24:
Ack. Does not even any format. Files are still there.

3dot3, 2015-10-25:
it happens to me as well

Aeprox, 2015-10-25:
Strange, I haven't encountered this issue. Could you give some more info on this? eg. compiled modules, flash size etc.

eku, 2015-10-25:
@Aeprox  flashing esp_init_data_default.bin does not help.

jmattsson, 2015-10-27:
I can reproduce the issue here, I'll take a look as soon as I find a little bit of time!

karrots, 2015-10-27:
ESP-12 with a 32Mbit flash for me. It doesn't always reboot but it does
most of the time. It seems to be dependent on how many files and other
randomness. Sometimes it formats fine, others it reboots but the format
seems to complete, and other times you're just stuck and it won't format.

jmattsson, 2015-10-27:
Okay, it should be fixed now (see #703 for details).

TerryE, 2015-10-27:
One issue with the 12e and some other parts is that they run in DIO mode rather than QIO mode -- they trade off halving the flash bandwidth to make another couple of I/O pins available at an application level.  Most developers don't notice this having in bandwidth, but it does tip some timeout-critical code over the edge.

karrots, 2015-10-28:
Seems to be fixed by the wdt SDK update in #703. Still need to test a bit more to be sure.

eku, 2015-10-28:
I can confirm that #703 fixes the file.format() problem.

3dot3, 2015-10-28:
Yes, it has been fixed.

closed 2015-10-28 by TerryE