nodemcu/nodemcu-firmware #698: The wifi.sta module connects if a loop is running?

domingosl, 2015-10-22:
Im trying to detect when the module actually connects to my wifi AP, since .connect does not have a callback im doing something simple like this:
```
wifi.sta.config("SSID","password")
wifi.sta.connect()
tmr.delay(1000000)
i = 0
while(wifi.sta.status() ~= 5 and i < 10) do
  print("Waiting")
  print(wifi.sta.status())
  i = i + 1
  tmr.delay(1000000) 
end
```

But the output of .sta.status() is always 1 inside the loop. When it finish, if I send the command =wifi.sta.status() manually from the IDE it tells me 5. Why?

Aeprox, 2015-10-22:
During tmr.delay(1000000) the chip does nothing at all, so it can't connect. I suggest you read TerryE's awesome (unofficial nodeMCU FAQ)[http://www.esp8266.com/wiki/doku.php?id=nodemcu-unofficial-faq] and close this issue.

Aeprox, 2015-10-22:
FYI: https://github.com/nodemcu/nodemcu-firmware/wiki/nodemcu_api_en#wifistaeventmonreg implements a callback that does what you need without polling wifi.sta.status() in a loop.

domingosl, 2015-10-22:
@Aeprox Thanks that is more clean :). Im getting an error when calling monreg functions; 
attempt to call field 'eventMonStart' (a nil value)
stack traceback:
	init.lua:12: in main chunk
	[C]: in function 'dofile'
	stdin:1: in main chunk

Is like there is no reference to that function in the firmware, but Im using the last one compiled using http://frightanic.com/nodemcu-custom-build/

Aeprox, 2015-10-22:
It seems like this feature hasn't made it to the master branch yet (be79d1d3b6795e1216fc046b543dc99278e37468), so you'll need to build from dev branch.

TerryE, 2015-11-02:
Your code will never work.  Sorry but this isn't because of any bug in the nodemcu firmware, but because your are attempting to do something that it was never intended to do.  Connections are async so you have to return control to the SDK for them to be actioned.  We can't help here.  You need to understand the constraints of the system.  This issues list isn't for teaching apps programmers how the the SDK works.

closed 2015-11-02 by TerryE