nodemcu/nodemcu-firmware #701: tmr.time is broken with 0.9.6-dev_20150704

danlu01, 2015-10-25:
With firmware 0.9.6-dev_20150704, if I run =tmr.time() it jumps up and down and up.  This is what I get:

> =tmr.time()
862
> =tmr.time()
863
> =tmr.time()
857
> =tmr.time()
857
> =tmr.time()
859
> =tmr.time()
862
> =tmr.time()
862
> =tmr.time()
864
> =tmr.time()
865
> =tmr.time()
873
> =tmr.time()
871
> =tmr.time()
871
> =tmr.time()
872
> =tmr.time()
874
> =tmr.time()
882
> =tmr.time()
874
> =tmr.time()
874
> =tmr.time()
876

Aeprox, 2015-10-25:
Duplicate of #690 , please close this issue.

danlu01, 2015-10-25:
it seems to work better with 0.9.5 except it would run 1 sec slow for every 80sec.  I had to upgrade to latest firmware because ran into NET ON SENT callback crash.

closed 2015-10-25 by danlu01