nodemcu/nodemcu-firmware #704: Calling wifi.sta.getap() makes the MCU resetted by WDT

cebrax, 2015-10-27:
I am trying to create a webserver and an easy config user interface based on Marcos Kirsch's [nodemcu-httpserver](https://github.com/marcoskirsch/nodemcu-httpserver). I am using nodemcu-master build v0.9 with flashid of 1261768. I have downloaded the master build from http://frightanic.com/nodemcu-custom-build/.

I want to send list of APs back to the user with a script file "get_nw.lua", however whenever the callback that is defined in the argument of function wifi.sta.getap(), the system reboots with below output:

```
Requested URL: /nw.html
Method: GET
File: nw.html
Begin sending:	nw.html
Sent	nw.html	1024
Sent	nw.html	2048
Sent	nw.html	3072
Sent	nw.html	4096
Sent	nw.html	5120
Sent	nw.html	5426
Finished sending:	nw.html
 
Requested URL: /get_nw.lua
Method: GET
File: get_nw.lua
Begin sending:	get_nw.lua
Finished sending:	get_nw.lua
 

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 28608, room 16 
tail 0
chksum 0x5b
load 0x3ffe8000, len 3020, room 8 
tail 4
chksum 0x61
load 0x3ffe8bcc, len 8, room 4 
tail 4
chksum 0xb1
csum 0xb1
rL
```

I have tried various things with the callback function including the function in the examples of nodemcu wiki. All the things I've tried works when  run in a standalone script file or somewhere else in the program. However, it creates problem when called within the function in "get_nw.lua"

Here are the various files:

init.lua:
```
tmr.delay(4000000)
print("Started nodeMCU...")

-- Set UART Configuration
uart.setup(0,115200,8,0,1)

-- Work at 160 MHz
node.setcpufreq(node.CPU160MHZ)

-- Begin WiFi configuration

local wifiConfig = {}

-- wifi.STATION         -- station: join a WiFi network
-- wifi.SOFTAP          -- access point: create a WiFi network
-- wifi.wifi.STATIONAP  -- both station and access point
wifiConfig.mode = wifi.STATIONAP  -- both station and access point

wifiConfig.accessPointConfig = {}
wifiConfig.accessPointConfig.ssid = "ahsa-IOT-node-"..node.chipid()   -- Name of the SSID you want to create
wifiConfig.accessPointConfig.pwd = "abdullah12345"    -- WiFi password - at least 8 characters

wifiConfig.accessPointIpConfig = {}
wifiConfig.accessPointIpConfig.ip = "192.168.25.1"
[... 211 lines cut ...]
	local fin = 0
	local info_table = {}
	local json_string = '{';
	info_table.ssid, info_table.pwd = wifi.sta.getconfig()
	info_table.ip,
	 info_table.netmask,
	 info_table.gateway = wifi.sta.getip()
	if (wifi.sta.status() == 5) then
		info_table.connected = "true"
	else
		info_table.connected = "false"
	end
	for key, val in pairs(info_table) do
		json_string = json_string .. '"' .. tostring(key) .. '" : "' .. tostring(val) .. '", '
	end
	wifi.sta.getap(print)
	json_string = string.sub(json_string, 1, -3)
	json_string = json_string .. '}'
	connection:send("HTTP/1.0 200 OK\r\nContent-Type: application/json\r\nCache-Control: private, no-store\r\n\r\n")
    connection:send(json_string)
    fin = nil
    info_table = nil
    json_string = nil
end
```

pastukhov, 2015-10-27:
Same for me

TerryE, 2015-10-27:
Hi @cebrax.  This issues list is for issues relating to the development of the firmware.  The nodeMCU guys have "gone fishing" and the community contributors are all working to rebaseline the firmware against a SDK 1.4 based build.  No one who is actively working on the firmware will look at issues with a package that is stretching the esp8266 capabilities against an old SDK version wihch has loads of known bugs, so this issue falls directly in the "won't fix" category.  Sorry, but I'd be a liar if I promised anything else.

closed 2015-10-27 by TerryE

cebrax, 2015-10-28:
Hello @TerryE. I see. Thank you for the info.