nodemcu/nodemcu-firmware #706: PWM glitch changing duty from 1023

nickandrew, 2015-10-29:
If PWM duty is set to 1023 (100% on) and subsequently changed to any other value, the PWM output glitches and has a low value for approximately 1 full PWM cycle before adopting the new duty cycle.

I put a logic analyser on D6 (Channel 0, PWM output) and D7 (Channel 1, trigger mechanism) and ran this snippet:

``` lua
pwm.setup(6, 50, 1023)
pwm.start(6)
gpio.mode(7, gpio.OUTPUT)
gpio.write(7, gpio.HIGH)

tmr.alarm(3, 5000, 0, function()
    gpio.write(7, gpio.LOW)
    pwm.setduty(6,900)
    tmr.delay(80000)
    gpio.write(7,gpio.HIGH)
    pwm.setduty(6,1023)
end)
```

Outputs observed:

[image]

Note there is no glitch changing the duty *back* to 1023, which happens after the Channel-1 trace goes high: the last duty=900 cycle has just begun, and it completes, and the output remains high. Here's a 100 Hz trace showing no glitch changing it back:

[image]

tort32, 2016-01-23:
Faced with this issue too.
But I see no such glitch when at least one other PWM output is working (with duty != 1023)
According https://github.com/nodemcu/nodemcu-firmware/issues/124#issuecomment-70983439 PWM is software based.
So I guess CPU need some time to start generate PWM
```
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 547ebdf3ff07caa1bc20b71d0483e80d7e80fd26
	SSL: true
	modules: node,file,gpio,wifi,net,pwm,tmr,uart,bit,cjson,crypto,rtcmem,rtctime,rtcfifo,sntp
 build 	built on: 2016-01-06 15:44
 powered by Lua 5.1.4 on SDK 1.5.0
```

nickandrew, 2016-01-23:
> So I guess CPU need some time to start generate PWM

This happens while PWM is active (changing duty to 1023, 1022, 1023 etc) so I don't think the CPU is *starting* PWM as such. But even if it is, the output needs to not have a glitch like this.

My understanding of the PWM waveform is that the __high__ output is the first part of each cycle (for *duty* slots) and then __low__ for *1023 - duty* slots, where if *duty* is 1023, there's no low part. So when PWM starts up, it needs to start with a __high__ value.

pjsg, 2016-03-05:
Does this still happen after the pwm rework in the dev branch?

danlu01, 2016-10-03:
Might be related to "Duty cycle limited to 90% (at 1kHz PWM period)".  Came across Stefan Bruens's version of PWM.  I don't know how easy would it be to test it on nodemcu env.  Here's link to his new PWM:
https://github.com/StefanBruens/ESP8266_new_pwm

danlu01, 2016-10-03:
I would be interested to get PWM freq above 1khz.

pjsg, 2016-10-03:
We don't use the timer code from the SDK...

I would think that we could just change the limit on the frequency to be higher if we wanted. Of course, the higher the frequency, the greater the overhead. 

It wouldn't be that difficult to improve the resolution either.... 

What frequency do you want?

danlu01, 2016-10-04:
Maybe 10kHz or even 20kHz would be nice.

closed 2019-06-21 by stale[bot]