nodemcu/nodemcu-firmware #707: is MQTT Publish slow to send?

crazycoder1999, 2015-10-29:
Hi,
 I'm using esp8266 with the firmware produced with  Marcel's NodeMCU custom builds.
I tested the "dev" branch and the "master".
I changed a little bit the "Connect to MQTT Broker" code here https://github.com/nodemcu/nodemcu-firmware :
- removing the subscriber
- adding a loop of 100 messages around --> **m:publish("/topic","hello",0,0, function(conn) print("sent") end)**

I'm using mosquitto as a mqtt broker and I have launched a subscriber on all topic **#**.

The result is: the messages arrives correctly but they are really *slow* to send (around 1 second each)... 
why?

Thanks you for this great project!

TerryE, 2015-10-29:
Hi @crazycoder1999, this list is for community developers how are enhancing and fixing the firmware.  End-developers are welcome to post specific bugs in the firmware itself, but you are better posting more general "how-to" Qs on a developer forum such as Stackoverflow or esp8266.com.  Since the original developers have "gone fishing", the remaining community developers don't have the bandwidth to answer this type of query.  Sorry.

closed 2015-10-29 by TerryE

crazycoder1999, 2015-10-29:
ok, thanks you!
conversation moved here http://stackoverflow.com/questions/33414441/nodemcu-and-esp8266-slow-mqtt-publish