nodemcu/nodemcu-firmware #709: TMR API Request

eku, 2015-10-30:
The TMR modul provides 7 timers (0-7). The caller has to take care which timer is in use and which is free. I suggest to manage this in the modul itself. Reason: if you have a lot of packages that are loaded
on demand and a package does not know of other packages it cannot know which timers are free.

It the tmr.start would be changed from 

```(void) tmr.start(timer, delay, repeat, function)``

to

```handle = tmr.start(delay, repeat, function)```

the lua scripts can use timers without the knowledge of the number of timers in general and the number of free timers.

devsaurus, 2015-10-30:
Appealing idea. It calls for some sort of free/release method then as well. Maybe automatic garbage collection would be enough.

marcelstoer, 2015-10-30:
+1, this bothered me as well just the other day.

What would you do about `tmr.stop()`? Should the handle be an object with a `stop()` method or should we pass the handle to `tmr.stop()`?

eyaleb, 2015-10-30:
While this is a "nice to have" feature, being an embedded system with (very) limited resources I am ready to give up such niceties. However, if each module clearly documents shared resources that it needs ownership of then the task will already be simpler. There is a similar issue with `rtcmem` and probably more.

Heck, just today I had a problem that boiled down to my module using the same global name as one introduced in 1.4.0:-(

eku, 2015-10-30:
> What would you do about tmr.stop()? Should the handle be an object with a stop() method or should
> we pass the handle to tmr.stop()?

The handle is the number of the timer. It will be used with tmr.stop().

devsaurus, 2015-10-30:
I also had the object oriented approach in mind when looking at your description. So `tmr.function(handle, ....)` would be equivalent to `handle:function(...)`.

eku, 2015-10-30:
> While this is a "nice to have" feature, being an embedded system with (very) limited resources I am ready to give up such niceties. 

I do not agree with this. Keeping track of timers in use needs only a bit-array that fits in to an unsigned char. Doing the bit setting, resetting an testing are less than 10 C-instructions.

>  However, if each module clearly documents shared resources that it needs ownership of then the task will already be simpler. 

What if more than 7 modules that need a timer, but never more than 7 at the same time? Documenting does not help in such a situation.

dnc40085, 2015-10-31:
I don't know about getting rid of the ID, I like having the option to select a specific timer. 

However it would be nice to have the timer limit be dynamic instead of just being limited to what is configured at build time( currently 7 timers).

TerryE, 2015-10-31:
Whatever we do, it *must* be backwards compatible.  Most apps will keep the current API and there's absolutely no point in breaking loads of apps because a few developers  prefer more of a functional or OO paradigm.

dvv, 2016-06-16:
That's what I drafted (a bit JS-ish) but never had a chance to validate and hard test:

```Lua
------------------------------------------------------------------------------
-- Timer helpers
--
-- LICENCE: http://opensource.org/licenses/MIT
-- Vladimir Dronnikov <dronnikov@gmail.com>
--
-- NB: occupies physical tmr #0
--
-- Example:
-- setInterval(1000, print, "many", "times") -- 1
-- setTimeout(1000, print, "one", "shot")
-- clearInterval(1)
------------------------------------------------------------------------------

--local 
handlers = { }

local function handler()
  local now = tmr.now() / 1000
  local nextTime
  local k, v = next(handlers, nil)
  while k do
    --print("DEBUG", "CHECK", now, v[1])
    -- on time?
    if now >= v[1] then
      -- callback valid?
      local fn = v[2]
      if fn then
        --print("DEBUG", "CALL", now, fn)
        -- do call
        fn(unpack(v[4]))
      end
      -- rearm if interval
      if v[3] then
        v[1] = now + v[3]
        if not nextTime or v[1] < nextTime then
          nextTime = v[1]
        end
      else
        handlers[k] = nil
      end
    -- get restart time
    else
      if not nextTime or v[1] < nextTime then
        nextTime = v[1]
      end
    end
    k, v = next(handlers, k)
  end
  -- active timers?
  if nextTime then
    --print("DEBUG", "REARM", nextTime - tmr.now() / 1000)
    nextTime = nextTime - tmr.now() / 1000
  else
    nextTime = 1000
  end
  pcall(tmr.alarm, 0, nextTime, 0, handler)
end

local new = function(repeating, timeout, fn, ...)
  local a = { ... }
  -- TODO: random names
  local n = #handlers + 1
  handlers[n] = { tmr.now() / 1000 + timeout, fn, repeating and timeout, a }
  handler()
  return n
end

setInterval = function(...)
  return new(true, ...)
end

clearInterval = function(n)
  handlers[n] = nil
  handler()
end

setTimeout = function(...)
  return new(false, ...)
end

clearTimeout = clearInterval
```

devsaurus, 2016-07-24:
Topic continued in #1416.

closed 2016-07-24 by devsaurus