nodemcu/nodemcu-firmware #710: dev140: Bug in SSL module

apESP8266, 2015-10-30:
Hello!

I'm testing the following NodeMCU firmware:

NodeMCU custom build by frightanic.com
	branch: dev140
	commit: 340ef8fc0ae00be1960acebe0276702dc7a00cb0
	SSL: true
	modules: node,file,gpio,wifi,net,pwm,i2c,spi,tmr,adc,uart,ow,bit,mqtt,cjson,crypto
build 	built on: 2015-10-26 19:36
powered by Lua 5.1.4 on SDK 1.4.0

After being connected to the Internet, I run the following commands:

local SMTP_SERVER = "smtp.gmail.com"
local SMTP_PORT = "465"
conn=net.createConnection(net.TCP, 0)
conn:on("reconnection", function(co, payload) print("Reconnected") end )
conn:on("disconnection", function(co, payload) print("Disconnected") end )
conn:on("receive", function(co, payload) print("received -> ") print(payload) end )
conn:on("sent", function(co, payload) print("sent -> ") end )
conn:connect(SMTP_PORT,SMTP_SERVER)
conn:dns(SMTP_SERVER,function(sck,ip) print("SMTP Server: "..ip) end)     
conn:on("connection", function(co, payload)
		print("Connected")
		local IP_ADDRESS = wifi.sta.getip()
                	conn:send("HELO "..IP_ADDRESS.."\r\n")
		end )

And I get the following results on the console:

SMTP Server: 64.233.184.109
connected
sent -> 
received -> 
  
Disconnected

Changing the third line of the commands for:
conn=net.createConnection(net.TCP, 1)

I just get:
SMTP Server: 64.233.184.109

Apparently, using the secure connection (conn = net.createConnection (Net.Tcp, 1)) none of the events is detected.
How can we deal with events in the case of secure connections?
Are we in the presence of a firmware bug in the SSL module?

Thanks in advance to those who answer these questions and help overcome the difficulty in connecting to the email service of Google.

TerryE, 2015-10-31:
duplicate of #520

closed 2015-10-31 by TerryE

i68040, 2015-10-31:
I reviewed issue #520 and the closest comment I could find to a resolution was made by marcelstoer where he said "Did you try the dev/dev140 branches which are based on the 1.4 SDK?".

It looks like this issue was created by someone using the dev140 branch who is having SSL issues. I am new to nodemcu (Just got my ESP8266 boards yesterday) and have not found a clear statement around SSL support in Lua. Are there examples that work?

Thanks!

TerryE, 2015-10-31:
@i68040 Rion, The original creators of the firmware have "gone fishing" and it's mainenance is being continued by community committers.  NodeMCU is a thin Lua implementation wrapped arounf the Espressif SDK and the current sdk (1.4) fixes a significant number of bugs in the 0.9x SDKs.  Our priority is to get a production quality version based on the current SDK asap.  The limitation of SSL support in Lua is to do with key negotiation and exchange rather than SSL itself.  You'll have to research the details I'm sorry to say.  There's infor here; at StackOverflow and esp8266.com.  Seek and ye shall find.

i68040, 2015-10-31:
I appreciate the clear and concise response. I've been all around trying to orient myself and this helps me get a good sense of where we are at the present.

Thanks again. :)