nodemcu/nodemcu-firmware #711: Was tmr.now() changed in 1.4.0?

eyaleb, 2015-10-30:
This is not a complaint. Having ran my application for a few hours now, I noticed that it is reporting the exact same timing (at ms resolution) for hundreds of consecutive reports. Each report includes 8 measures of different phases. Initially I suspected that I have a bug where the same report is resent - but not so, looking carefully I could find one or two 1ms differences.

It used to be that time measurement was rather noisy. If the timer was actually fixed then I wish to thank the person/team involved, it makes tuning an application much easier.

cheers

jmattsson, 2015-11-02:
Nothing has explicitly changed that I can see, though since it is using system_get_time() which is an SDK function, any improvements in the SDK would flow through. I don't see anything in the official release-notes however.

eyaleb, 2015-11-02:
Fair enough. To make the point clear, here is a sample of the readings. Each line is the time it took the application to go through a phase. Each entry is a letter followed by the time as `s.mmm`. The application executes once a minute then deep sleeps.

Note that 1.4.0 is drastically faster (the first `L` entry is the total time of the last cycle), and see how variable 0.9.5 was compared to 1.4.0.

0.9.5 records:
```
L3.664,T100330,s0.609,u0.045,r0.000,w2.199,F0.000,S0.004,d0.580,t2.880 
L3.639,T100333,s0.612,u0.045,r0.000,w2.192,F0.000,S0.004,d0.580,t2.873 
L3.639,T100337,s0.608,u0.045,r0.000,w2.202,F0.000,S0.004,d0.580,t2.884 
L3.664,T100341,s0.613,u0.045,r0.000,w2.199,F0.000,S0.004,d0.580,t2.881 
L3.664,T100344,s0.612,u0.045,r0.000,w2.152,F0.000,S0.004,d0.580,t2.834 
L3.614,T100348,s0.608,u0.045,r0.000,w2.197,F0.000,S0.004,d0.580,t2.878 
L3.639,T100352,s0.604,u0.045,r0.000,w2.149,F0.000,S0.004,d0.580,t2.830 
L3.589,T100355,s0.605,u0.045,r0.000,w2.201,F0.000,S0.004,d0.580,t2.882 
L3.639,T100359,s0.608,u0.045,r0.000,w2.200,F0.000,S0.004,d0.580,t2.881 
L3.639,T100362,s0.608,u0.045,r0.000,w2.197,F0.000,S0.004,d0.580,t2.879 
```
1.4.0 records:
```
L0.500,T1444,s0.095,u0.060,r0.002,w0.063,F0.000,S0.006,d0.137,t0.324 
L0.500,T1444,s0.095,u0.060,r0.002,w0.063,F0.000,S0.006,d0.137,t0.324 
L0.500,T1445,s0.095,u0.060,r0.002,w0.063,F0.000,S0.006,d0.137,t0.323 
L0.500,T1445,s0.095,u0.060,r0.002,w0.063,F0.000,S0.006,d0.137,t0.324 
L0.500,T1446,s0.095,u0.060,r0.002,w0.063,F0.000,S0.006,d0.137,t0.324 
L0.500,T1446,s0.095,u0.060,r0.002,w0.063,F0.000,S0.006,d0.137,t0.324 
L0.500,T1447,s0.095,u0.060,r0.002,w0.063,F0.000,S0.006,d0.137,t0.323 
L0.500,T1447,s0.095,u0.060,r0.002,w0.063,F0.000,S0.006,d0.137,t0.324 
L0.500,T1448,s0.095,u0.060,r0.002,w0.063,F0.000,S0.006,d0.137,t0.324 
L0.500,T1448,s0.095,u0.060,r0.002,w0.063,F0.000,S0.006,d0.137,t0.324 
```

TerryE, 2015-12-13:
Not an open issue.

closed 2015-12-13 by TerryE