nodemcu/nodemcu-firmware #714: Unreliable File System Isssue

georeb, 2015-10-31:
I using an ESP03 with NodeMCU (latest version)
I am using it as a server to serve a webpage and images.

When I save files to the flash (png images and html files) it can sometimes become flaky in the sense that...

Files are improperly saved and are malformed
Files aren't removed using 'file.remove()'
Using 'file.format()' sometimes doesn't clear the entire free flash space
File names are sometimes incorrect
Only part of the file is saved
After replacing a file, maybe 3 or 4 times, it is improperly saved
Is anyone else experiencing this problem??
It appears to get worse after time!
Is it a software bug or am I not correctly flashing NodeMCU initially?

TerryE, 2015-10-31:
The "latest" version is chronically out of date.  Try a dev build.

georeb, 2015-11-03:
Hi TerryE,

I'm not entirely sure how to do that (I'm new to github) - if I choose 'dev' or 'dev096' from the drop-down menu, the download link still gives me the master version (20150704)...

1 - what is the difference between master, dev and dev096 please?
2 - is there a later version than 20150704 please?

Many thanks

marcelstoer, 2015-11-03:
Build your own firmware against the dev branch:

- http://frightanic.com/nodemcu-custom-build/index.php
- https://github.com/marcelstoer/docker-nodemcu-build

georeb, 2015-11-03:
I have used frightanic to build from the master but never the dev.
@marcelstoer Can you put a date to the dev version?? (is it later than 20150704?)

marcelstoer, 2015-11-03:
You'll be building on-demand against the latest source in the respective branch. Hence, if you build against the dev branch the binary is going to include whatever you see here: https://github.com/nodemcu/nodemcu-firmware/commits/dev

georeb, 2015-11-03:
WOW - so I am massively behind with my version of NodeMCU?!?
That would mean the dev version was updated only 13 hours ago!?!

TerryE, 2015-11-03:
> Lastly - should I ignore dev096?? What is it??
A (i) yes (ii) soon to be no more.

closed 2015-11-03 by TerryE