nodemcu/nodemcu-firmware #715: Unable to send multicast UDP packets

xhs, 2015-10-31:
I'm trying to implement mDNS in Lua.

With the information I got from [Issue 398](https://github.com/nodemcu/nodemcu-firmware/issues/398) and [nodemcu-ssdp](https://github.com/pastukhov/nodemcu-ssdp), I can now receive and parse mDNS queries successfully. But I am unable to send response to the multicast address (224.0.0.251).

Here's the testing code:
```lua
wifi.setmode(wifi.STATION)
wifi.sta.config('ssid', 'password')
ip = wifi.sta.getip()
print(ip)
net.multicastJoin(ip, '224.0.0.251')
sock = net.createConnection(net.UDP, 0)
sock:connect(5353, '224.0.0.251')
sock:send('test broadcast', function() print('Sent') end)
```

I can see "Sent" printed but get nothing in Wireshark.

Tested on master branch.

Is there anything I should do before sending multicast packets?
Or does NodeMCU support this at all?

pastukhov, 2015-11-01:
I will test tomorow

nanikjava, 2015-11-19:
I've tested this on my ESP12E DevKit 1.0 with the following firmware

NodeMCU 0.9.6 build 20150704  powered by Lua 5.1.4

and I can't see it in Wireshark too. When I use Linux machine to multicast using the command

iperf -c 224.0.0.251 -u -T 32 -t 3 -i 1

I can see it in Wireshark.

Is this a bug ?

pastukhov, 2015-11-19:
Try with more fresh FW.

pastukhov, 2015-11-19:
1.4 version is actual now.

3dot3, 2015-11-19:
This is highly related to issue #596:https://github.com/nodemcu/nodemcu-firmware/issues/596
Smartconfig only doing half of the job because of this issue.

nanikjava, 2015-11-20:
@pastukhov not sure I understand what you mean "Try with more fresh FW" as I flashed the latest file from https://github.com/nodemcu/nodemcu-firmware/releases https://github.com/nodemcu/nodemcu-firmware/releases/tag/0.9.6-dev_20150704

karrots, 2015-11-20:
Those releases are old and outdated. Test a fresh 'dev' branch build from http://frightanic.com/nodemcu-custom-build/

klokik, 2015-11-28:
Fresh build, same bug

3dot3, 2015-11-30:
hope sdk1.5 upgrade will solve this.

nickandrew, 2015-11-30:
Once again, the "releases" are leading people astray.

marcelstoer, 2015-11-30:
@klokik You did use the `dev` branch rather than `master`, right?

klokik, 2015-11-30:
Right. I used a mentioned earlier service.

marcelstoer, 2016-04-10:
> I'm trying to implement mDNS in Lua.

We've had an [mDNS module](http://nodemcu.readthedocs.org/en/dev/en/modules/mdns/) in the `dev` branch for a while now -> closing.

closed 2016-04-10 by marcelstoer