nodemcu/nodemcu-firmware #716: Removing dev096 dev120 dev140 branches

TerryE, 2015-10-31:
All are no longer being maintained, and their existence simply confuses developers who want to use nodeMCU.

*   `dev096` with the exception of a couple of tweaks by @dnc40085 ( bfa1749 and bea5e90 ) dev includes all relevant changes and even thes e have had parallel work.
*    `dev120` only changes are @vowstars work that has been superseded in `dev` by @jmattsson Johny's work.
*    `dev140` is a clone of the `DiUS/dev140` branch and again all of this work has been merged into dev. 

MY recommendation is that I delete these ~~tomorrow~~ on Monday, unless any other committer says no.  I'll keep a copy of these on my local copy just in case we want to refer to them. 

Comments?  Objections?

marcelstoer, 2015-10-31:
What a relief, thanks.

dnc40085, 2015-10-31:
I think that's a great idea.

As for the event monitor tweaks... I have been working on exposing the new event monitor included in the SDK, but I'm not sure how to make a clean swap from the one I wrote previously and the new one without causing too many waves.

I could include a preprocessor directive to allow switching between new event monitor and old one so people who still want to use the old one have the option.

On a side note, as a user of the 512k ESPs, I was wondering...since with every new SDK there is less and less flash available to the NodeMCU firmware and to the end user, Would it not be a good idea to have a way to choose which functions, within modules, to include in a build similar to the file "user_modules.h"?

TerryE, 2015-10-31:
I personally hate the way that each module addition needs to modify `user_modules.h`.  I would personally prefer having a simple ini-style file to define the configuration, which some tool to process it and generate the user files.  For example:
```
flash=512K
debug=
core=string,math,table
modules=node,file,gpio,wifi,net,tmr,bit,ow
...
```
and yes this should be a separate issue :smile:

dnc40085, 2015-11-01:
Thanks for the suggestions
Which should be default? the new or old event monitor?
How about all the disconnect reasons? Should I make lua table entries, 28 in total, for all disconnect reasons adding a minimum of 420-620 bytes of text? 
If so, should I also add a method to exclude reason entries if desired?

TerryE, 2015-11-01:
@dnc40085, I'll have to look through your fork because I don't know what your new proposed API is.  As far as properties go, why not  use a metatable and an `__index` function?  Where are you publishing your new event monitor code.  I can't find it in your nodemcu fork.

dnc40085, 2015-11-01:
My apologies, I was working on it locally. 
I've created a new branch named [dev_New_event_monitor](https://github.com/dnc40085/nodemcu-firmware/tree/dev_New_event_monitor) on my github fork.

TerryE, 2015-11-02:
If you use pairs to enumerate **_G** then you'll not find any rotables listed,  Yet they are still accessible.  Why?  review `app/lua/lbaselib.c`and also net uses these techniques; see `app/lua/lbaselib.c`.

TerryE, 2015-11-02:
All, you will see that dev120 and dev140 are no more.  Dev includes all 1.4 code.  I'll wait until later just in case anyone can think of a reason for not removing 096.

saitejalakkimsetty, 2015-11-02:
dev096 seems to be the current stable branch. Maybe it is a good idea to leave 096 as it is until we get a prebuild for 140.

TerryE, 2015-11-02:
@saitejalakkimsetty. No, don't confuse tags and branches in git.   `dev096` isn't the current stable branch.  Tag `0.9.6-dev_20150627` was the last "stable" build.  You can still fetch this source whether or not the branch exists.  No one is working on `dev096` and we won't be accepting any more PRs for it.   However, I will delay deleting this branch another couple of days in case I get other positions logged.

saitejalakkimsetty, 2015-11-02:
@TerryE Sounds good.

karrots, 2015-11-02:
My question is similar to one expressed above. How do we get `master` and `dev` in sync? There are recent patches to `master` that need to end up in `dev` for example.

TerryE, 2015-11-02:
> The next job will be to reconcile dev against master.

Say no more :smile:

karrots, 2015-11-03:
Cool

TerryE, 2015-11-07:
Done.  `dev096` is no more and we are now down to `dev` and `master`

TerryE, 2015-11-08:
All done and no comments so this one is closed.

closed 2015-11-08 by TerryE