nodemcu/nodemcu-firmware #719: Policy for maintaining open issues

TerryE, 2015-11-03:
The nodeMCU team have suspended active contributions to this project until further notice.  However,the support and maintenance of the project is being continued by a small number of independent contributors until such a time as the nodeMCU team can re-engage.  Our priority is to produce a stable release:

*    Based on the current SDK (now at 1.4 ot time of posting).   
*    Including Johny Mattson's unaligned access to flash patch.
*    Including Terry Ellison's strip_debug patch
*    John and Terry's refactoring of duplicated string and memory utilities.
*    Improved integration with Marcel Stör's, [custom build service](http://frightanic.com/nodemcu-custom-build/)
 
The nodfeMCU firmware is a fairly thin [eLua](http://www.eluaproject.net/overview)-based implementation layered over this SDK, and by rebaselining to this current version, this will address many report firmware issues that are in practice a result of known SDK issues (see [SDK Release notes](http://espressif.com/new-sdk-release/)). Our other enhancements will significantly increase the effective RAM available to  Lua programs.

However, in order to achieve these aims with our limited resources we have to enforce a strict scope on issues that we will accept here.  We will specifically close:

*    Any questions of a "how do I ..." or a "this doesn't work ..." (Please refer these application developer forums such as StackOverflow with the `esp8266` and `Lua` tags, or esp8266.com
*    Any issues relating to 0.9x builds as all work on these has stopped
*    Any enhancement requests where the originating poster is not proposing to resource the development him or herself as we don't have the effrot to resource additional enhancements
*    Any bug reports with a SDK 1.4 based build that does not define the build baseline and include a standalone example (in the form of a gist or inline code) which demonstrates the bug.

We will also be reviewing all existing open issues and closing any issues which fail to achieve these criteria.  However anyone is free to reopen such issues if they can provide the necessary supplementary information (e.g. by providing an example which shows that the bug is still present on the current `dev` build).

We apologise for what may seem very strict criteria, but we have to focus our limited resources to achieve the greatest common good.

marcoskirsch, 2015-11-03:
Shouldn't this be in a read me rather than in an open issue?

Thanks for your efforts!

TerryE, 2015-11-03:
I should think it's open until we've done the cull.  Some contributors and developers might have an issue with this.  However, the only real alternative as I see it is to allow the project to go inactive.

dwery, 2015-11-04:
are binary releases of the dev (1.4) series available somewhere?

dnc40085, 2015-11-04:
You can compile the dev branch using Marcel Stör's [NodeMCU Custom build service](http://frightanic.com/nodemcu-custom-build/).

TerryE, 2015-11-04:
If you have 512Kb Flash then you have to do a local or cloud build and only select the modules that you typically use.

dwery, 2015-11-04:
Thank you very much

devsaurus, 2015-11-04:
> Improved integration with Marcel Stör's custom build service

We should to revisit #386 then. It's been unattended for too long.

nickandrew, 2015-11-05:
Thanks for your efforts!

I rebased my OneWire fixes against dev140 branch and I noticed some of them have already made it into the branch in some manner. I'll re-test the remainder against dev140 within the next few days and submit another pull request.

TerryE, 2015-11-05:
@nickandrew, you'll need to use a branch rebaselined against `dev` and cherry pick your commits against that.  That's what I did with #722.  The first two commits in this are cherry picked from `master` and the third fixed a functional merge issue against the 1.4 changes, plus a bug fix that I hadn't yet raised a PR for,  This way the PR is a fast-forward merge.

ghusson2, 2015-12-09:
Just a remark, I think that this project is moving fast this lasts weeks (BSP 1.5, documentation changes, structural questions).
I don't know how is the core contribution team organized and maybe paid.
I don't know the history of the project (initial team, "official hardware", sources of incomes...)
I have three remarks : 
1. https://en.wikipedia.org/wiki/NodeMCU : "NOTICE: Active development is suspended that is KAPUT!" : gives a false impression over the project
2. what is the future of the project (for me NodeMCU is a very clever platform for quick and comfortable DIYers and it is currently the best for ESP8266 ; Espressif will release the ESP32 and I hope that NodeMCU will be ported) ?
3. is NodeMCU a source of income for somebody here, if it is the case : 
  1. how are you involved (how many time do you work for the project for example)
  2. how can we help you (for example howto donate, where to buy "official hardware")
I didn't see those information (maybe I missed something) so I just wanted to ask here :)

nodemcu, 2015-12-09:
well, I think I can answer part of your question.
> The nodeMCU team have suspended active contributions to this project until further notice. 

When this issue was opened, I and the original team were busy in something else, and kind of leave this project alone for many days. especially not focus on the github issues and not answer/reply to community questions.
My apologize to everyone.

>  https://en.wikipedia.org/wiki/NodeMCU : "NOTICE: Active development is suspended that is KAPUT!" : gives a false impression over the project

After this issue. there were very constructive email discussions between committers.
we sort things out and made some goals like stable releases, rework documents, guidelines. 
and here it is, this project I think now is back to its track.
@TerryE @marcelstoer @devsaurus @jmattsson @vowstar are now the collaborators of this repo.
Here many thinks to them. they offer many help and professional skills to this project.
of course also many thinks to every one who pushed PRs and issues.

>  what is the future of the project (for me NodeMCU is a very clever platform for quick and comfortable DIYers and it is currently the best for ESP8266 ; Espressif will release the ESP32 and I hope that NodeMCU will be ported) ?

NodeMCU will be ported to ESP32 if we can, there is rtos_sdk for esp32 "leaked".
but no info for the non-OS sdk right now. and espressif will beta the chip very soon I think.

>  is NodeMCU a source of income for somebody here, if it is the case :

non of us is full time work. mostly as interest/hobby.
and nobody gets paid from this project.
there is some income from the nodemcu dev-kit hardware before which can afford some host and domain name for www.nodemcu.com, bbs.nodemcu.com.
but it become little there are many "copies" hardware out there.

ghusson2, 2015-12-09:
Thanks for this clear explanation.
I think the suspension phase for stabilization and organization was mandatory.

marcelstoer, 2016-01-22:
I think we should close this and stop sending people who ask questions on the issues list over here. The initial description is only partially correct anymore and may, therefore, be misleading. This project is more active than ever and as a "don't ask howto questions here"-pointer we could use http://nodemcu.readthedocs.org/en/dev/en/support/.

TerryE, 2016-01-22:
@marcelstoer, Marcel, approaching 50% of new issues are outside the scope of this issues list, and should really be redirected to esp8266.com, SO or the like.  As far as the intro para is concerned: whilst zeroday is a stakeholder -- and we really appreciate his continued support -- his time available to work on this list is very limited, and none of the other NodeMCU employees contribute any more. We have only a limited number of committers and the work load in terms of moderation of this list isn't evenly divided.

Yes, we are steadily building up a core of active contributors and many have only actively started to engage in the last 6 months, and we're now keeping track of Espressif issues, sorting the documentation and adding important functionality.  So we should all take pleasure in that.  However, on a personal level I have a lot of competing demands on my time, so I don't spent as much time as I would like on the NodeMCU project.  Given my time is constrained, I would prefer to spend more of it developing new core functionality and less acting as the bad guy telling people that they are asking the wrong Qs or at least the right Qs but in the wrong place. 

I've done a quite rework to bring the lede up todate.By all means rework the intro to add anything that is missing, or close the issue and develop a simple and single URI that we can refer such posters to.  IMO, the great advantage of using an issue is that it's integration is within the core markup engine, so in responding I only need to type `#719` to link to this policy _and_ tracking issues closed because of it.  This is a lot easier than labelling them as say `Out of Issues scope`, posting in a link which is of the form `[Policy for Closing Out of Scope Issues](http://nodemcu.readthedocs.org/en/dev/en/support/#Policy_for_closing_out_of_scope_issues)`.  

It would really help us if Zeroday would promote a couple more of the active contributors to committer so we can share the load.  In the meantime, why don't we kill a couple of "birds with one stone"?  I'll take a couple of weeks off moderating the list and you take over, then maybe you will see why I prefer the "Please refer to \#719" option. :smile:

marcelstoer, 2016-01-22:
Nothing beats `#719` that's pretty obvious. I'm not even gonna argue that keeping a (potentially shortened like http://j.mp/1SClr70) URL somewhere handy takes only 3s longer to copy-paste because I know it's not the same thing. Also, we can't just edit the initial description of this issue as some of the comments wouldn't make sense then.

So, if you want to stick with using an issue, nothing against that, then we should create a new one with an up-to-date description but _lock it_ ('Lock conversation' link). That way you can adjust the description anytime without destroying a conversation thread because there won't be one. But hurry before the issue numbers move from 3-digit to 4-digit :wink:

TerryE, 2016-01-22:
Marcel, looks like you drawn the short straw for ?963? Or should just wait until \#999?  But I could do with a break from being Mr Bad Guy. :cry:

marcelstoer, 2016-07-28:
Superseded by #1010.

closed 2016-07-28 by marcelstoer