nodemcu/nodemcu-firmware #720: Android unable to connect to NodeMCU

divye1995, 2015-11-03:
Hey i have had this issue for a long time please someone help me solve it.
 My android phone never connects to my nodeMCU in AP mode
it keeps on  connecting and disconnecting , it connects to laptops fine though , I went through many forums and many of them said its an issue with SDK of ESP used in the firmware , but for some reason many users who are not using nodeMCU but only ESP are able to connect their android /ios phones , please help me out there , there seems no solution anywhere else online
If this is a bug with the Firmware please fix it asap 
I m current using nodeMCU v1.0 and NodeMCU 0.9.6 build

TerryE, 2015-11-03:
See #719

closed 2015-11-03 by TerryE

karrots, 2015-11-03:
Seems like this may be fixed by the 1.4.0 sdk patch that was released. 

http://bbs.espressif.com/viewtopic.php?f=46&t=1268