nodemcu/nodemcu-firmware #721: Problem with powersaving

NiklausS, 2015-11-03:
Hi,
just upgraded to from 0.9.5 to 0.9.6 and found that esp does not go to LIGHT_SLEEP or MODEM_SLEEP after dsleep -> the powerconsumption remains at 50mA with 0.9.5 this was working fine.

tested with:
ok - nodemcu_float_0.9.5_20150318.bin  (NodeMCU 0.9.5 build 20150318  powered by Lua 5.1.4)

bad - nodemcu-dev-18-modules-2015-10-27-15-11-48-float.bin (Lua 5.1.4 on SDK 1.4.0)
bad - nodemcu-dev140-19-modules-2015-10-28-20-33-03-float.bin

tmr.alarm(1, gsleep*1000, 0, function()
print("timer1 trigger dsleep - "..node.heap())
node.dsleep(1000000, 4)
end)

Thanks for the greate software.
 Nik

dnc40085, 2015-11-04:
I'm having the same issue with the current dev branch(SDK ver 1.4.0).
The SDK function `deep_sleep_set_option()` used by `node.dsleep()`, is returning a 0, which indicates a failure according to the SDK programming guide.

dnc40085, 2015-11-06:
The new function `wifi.sleep` should help you accomplish your task until the `node.dsleep` issue is resolved

NiklausS, 2015-11-06:
Thanks,
already tested it yesterday and it works so far.
I've places a wifi.sleep(0) at line 1 in the init.lua. I see a small  peek of approx. 60mA. 
So far this works as workaround.
I guess this goes in line with the readvdd33(), returning always 65535 in dev branches.

closed 2015-11-06 by NiklausS