nodemcu/nodemcu-firmware #723: Different results computing hmac in different platforms

drio, 2015-11-04:
Hi all, 

I'm having problems matching the results of computing the hmac on the same
string/key between nodemcu and another platform (python 2.7). I was wondering
if anyone else has run into this and may want to help me out. Details here:

```
Lua 5.1.4 on SDK 1.4.0
```
```
	print(crypto.toHex(crypto.toBase64("md5", "message", "secret")))
end

jmattsson, 2015-11-04:
Your issue is that you're not actually using crypto.hmac() in your command.
``> print(crypto.toHex(crypto.hmac("md5", "message", "secret")))``
``7e0d0767775312154ba16fd3af9771a2``
``> print(crypto.toBase64(crypto.hmac("md5", "message", "secret")))``
``fg0HZ3dTEhVLoW/Tr5dxog==``

closed 2015-11-04 by jmattsson

drio, 2015-11-04:
:sweat:  Thank you @jmattsson