nodemcu/nodemcu-firmware #726: Issue with createConnection to intranet IP

happyiphone, 2015-11-04:
I have my nodemcu connected to my wifi network. I can connect just fine to pretty much any server I want on the internet but I've been trying to connect to an intranet IP and I always get a disconnection event. The server is there and I can connect just fine with any other tool from within the same network (there is no problem with the server or reaching the server).

Trying to find the solution to this issue I've tried to "trap" the error but I can't find anything helpful. There is a parameter on the disconnection event but all I get from that parameter is this:

userdata: 3fff3ce0

Does that mean anything to anybody? I've been trying to find any information about that reason on the disconnect but no luck, any hint?

jmattsson, 2015-11-05:
A few questions:
  1. Are you using deep sleep and connecting right after that?
  1. If you continue retrying the connection, does it eventually work?
  1. When you connect to non-local IPs, do you go through a NAT?

TerryE, 2015-11-05:
I've just had a look at the code in [net.c](https://github.com/nodemcu/nodemcu-firmware/blob/master/app/modules/net.c#L78) and it does return the `self_ref` field as userdata.  Not sure why becuase the Lua callback routine can't do anything with userdata!

marcelstoer, 2016-04-10:
No feedback from OP in months -> closing.

closed 2016-04-10 by marcelstoer