nodemcu/nodemcu-firmware #732: PANIC: unprotected error in call to Lua API (stdin:8: attempt to call global 'run' (a nil value))

confile, 2015-11-05:
Using master commit: 9306e76ae17a2455315b73fc0774d92c4d589062

Here is the code I run: 

```
ssid = "www.mydomain"
pass = "234234234432"

gpio.write(0, gpio.LOW)

print("Startup up wifi mode")

wifi.setmode(wifi.STATION)
wifi.sta.config(ssid, pass)

wifi.sta.autoconnect(1)
wifi.sta.connect()

tmr.alarm(3, 1000, 1, function() 
    if (wifi.sta.status() < 5) then
        print("Connecting...")        
    else 
        tmr.stop(3)
        print("Connected having IP "..wifi.sta.getip())
        gpio.write(0, gpio.HIGH)        
        run()      
    end
end)

gpio.write(0, gpio.HIGH)

 

function run() 
    print("run")

    myhost="www.adafruit.com"
    mypage="testwifi/index.html"
    myip=""
    
    sk=net.createConnection(net.TCP, 0)
    sk:dns(myhost,function(conn,ip) 
    myip=ip
    sk=net.createConnection(net.TCP, 0)
    sk:on("receive", function(sck, c) print(c) end )
    sk:connect(80,myip)
    sk:send("GET / " .. mypage .." HTTP/1.1\r\nHost: " .. myhost .."\r\nConnection: keep-alive\r\nAccept: */*\r\n\r\n")
    sk=nil

end
```

I got this error: 

```
PANIC: unprotected error in call to Lua API (stdin:8: attempt to call global 'run' (a nil value))
�m���!����!1���
Waiting answer from ESP - Timeout reached. Command aborted.
```

devsaurus, 2015-11-05:
You're probably suffering from a race condition when downloading this code via serial connection (using ESPlorer?). Ensure that `run()` is fully defined before the timer expires for the first time.

Closing this now for reasons stated in #719. If you need further support then please post your issue with full log on esp8266.com. Looking forward to following up there...

closed 2015-11-05 by devsaurus

confile, 2015-11-05:
@devsaurus Yes I use ESPlorer. Another thing is that there are these strange characters in the output see ``����m������!������!�1���``. What does this mean?

What do you mean by run is fully defined. Do you mean place the method above the timer?

TerryE, 2015-11-06:
> You're probably suffering from a race condition when downloading this code via serial connection (using ESPlorer?). Ensure that `run()` is fully defined before the timer expires for the first time.

What @devsaurus means is that ESPLorer downloads snippets chunk by chunk.  To force this into one Lua chunk, bracket the snippet by `do` ... `end`. Please read #719.

confile, 2015-11-19:
@TerryE thank you!