nodemcu/nodemcu-firmware #736: Current dev won't compile for board revision 0.9

UserXYZ, 2015-11-06:
When enabling DEVKIT_VERSION_0_9 in user_config.h, READLINE_INTERVAL is not defined and compilation crashes almost instantly.
In previous dev revisions, even with SDK 1.4, this was explicitelly defined, regardles of DEVKIT_VERSION and so compilation works.
Further, with DEVKIT_VERSION_0_9 enabled, compilation crashes on node.c with following error:

node.c:246:6: error: conflicting types for 'key_long_press' [-Werror]
 void key_long_press(void *arg) {
      ^
node.c:183:7: note: previous implicit declaration of 'key_long_press' was here
       key_long_press(NULL);
       ^
node.c:258:6: error: conflicting types for 'key_short_press' [-Werror]
 void key_short_press(void *arg) {
      ^
node.c:187:7: note: previous implicit declaration of 'key_short_press' was here
       key_short_press(NULL);
       ^
cc1: all warnings being treated as errors

I tried rebuilding esp-open-sdk with crostools from https://github.com/pfalcon/esp-open-sdk.git (which worked before and is, as far as I know, suggested set for building the firmware) but it made no difference.

It seems that somebody made a change without checking output on both board revisions and that rendered compilation for old board impossible.
Building is done on Fedora 21 and Ubuntu 12.04, no difference.

For some reason, frightanic's dev build still passes, don't know how if he also had his dev tools/SDK updated...?

jmattsson, 2015-11-09:
This looks like something I broke. I'll have a look.

jmattsson, 2015-11-09:
Yep, that was my fault. Apologies.

TerryE, 2015-11-09:
@UserXYZ please rebuild and retest.  If OK, then please close this issue.  If we have not had confirmation that there is still an issue, then we will close this in 7 days time.

drawkula, 2015-11-09:
My 0.9 is in a reboot loop after flashing current nodemcu dev...

marcelstoer, 2015-11-09:
> For some reason, frightanic's dev build still passes, don't know how if he also had his dev tools/SDK updated...?

I don't need to as NodeMCU (i.e. this repository) ships with its own version of `esp-open-sdk.tar.gz` in `/tools` that matches the repository code.

UserXYZ, 2015-11-09:
I did the rebuild, and it passed just fine.
But, I'm kinda reluctant to use it since somebody reported they have constant reboot loop on board v0.9?

drawkula, 2015-11-09:
> But, I'm kinda reluctant to use it since somebody reported they have constant reboot loop on board v0.9?

Here... 

    $ TZ=GMT LANG= ls -l /opt/nodemcu-firmware/bin/
    total 484
    -rw-r--r-- 1 yeti yeti  32704 Nov  9 14:18 0x00000.bin
    -rw-r--r-- 1 yeti yeti 462499 Nov  9 14:18 0x10000.bin

...tested not really long ago...

drawkula, 2015-11-09:
> I did the rebuild, and it passed just fine.

It compiles...

> But, I'm kinda reluctant to use it since somebody reported they have constant reboot loop on board v0.9?

Here!

    $ TZ=GMT LANG= ls -l bin
    total 484
    -rw-r--r-- 1 yeti yeti  32704 Nov  9 14:18 0x00000.bin
    -rw-r--r-- 1 yeti yeti 462499 Nov  9 14:18 0x10000.bin
    $ git pull
    Already up-to-date.
    $ git branch -a
    * dev
      master
      remotes/origin/HEAD -> origin/master
      remotes/origin/dev
      remotes/origin/master

TerryE, 2015-11-09:
The worst that occurs with a constant reboot is that you will have to flash the old binaries.  This is your choice, but if you don't want to use it, then we can't keep this issue open.

closed 2015-11-09 by TerryE

UserXYZ, 2015-11-09:
Well if this isn't stupid reasoning...if you don't like my product being bad and faulty, then don't use it...or use old version...
What if Espressif said that to you when they published new SDK and you found out that it has a lot of neat things in it that you need but it makes your esp reboot and you dared to complain?
Would you feel happy about their attitude - meh, don't use it if you don't like it...

My point being - if current code is causing v0.9 boards to reboot, then it's the code that needs to be fixed, and not the hardware, since previous code didn't behave like that.

drawkula, 2015-11-09:
Ok... the "does not compile issue is gone", but it still is unusable. Perhaps it is a new issue now?

@UserXYZ try to run it... the reboot loop do not prevent flashing the old firmware again...

UserXYZ, 2015-11-09:
I will do it later tonight and test it, but as you say, maybe we have another issue now.
If that is the case, it should be fixed, or if nobody wants to do it because they do not have v0.9 and don't care about the problem, then previous dev should be freezed and a new version forked that will explicitelly be marked as "usable for boards revision 1.x ONLY" so we know not to expect any further help/updates.
That would suck big time, but at least, our next git pull would not generate totally unusable firmware.

TerryE, 2015-11-09:
@UserXYZ, sorry I should have referenced this: please read #719.  Also plesae avoid personal criticisms here.  I and the other maintainers of the firmware have no financial links with the NodeMCU Inc.  We are just doing our best to maintain the development of the firmware for the common good of the esp8266 Lua community, and we have to make priorities.  Getting flamed won't make us change them.

UserXYZ, 2015-11-09:
Dear TerryE, if you get something cool to play with from somebody smarter than you, and it stops working, it is logical that you will turn to that smarter one to help you.
Now, if he says "meh, can't bother with it", then it is normal to feel angry, mad, dissapointed, whatever, because somebody better than you has kind of let you down.

So, I am fully aware that nobody here is coding/testing/keeping this project alive for money and personal criticism was not my intention, but I do believe I have, like many that depend on you and other developers, right to feel at least dissapointed when we hear "use something old" because that is the worst solution - and that's why I said that either a freeze/fork should be done or the issue solved.

If I knew how to code C, I would most definitelly help/fix it myself...but, since you are smarter than I/we am/are for such things, I/we depend on you.

TerryE, 2015-11-09:
Your last comment isn't relevant to whether this issue is open.  The title on the issue is "Current dev won't compile for NodeMCU board revision 0.9" -- it  now does.  However, you have also said " I'm kinda reluctant to use it", so the call is closed until you come up with a reason to reopen it, or you raise another valid issue.

jmattsson, 2015-11-09:
For the record, I obviously tested this fix as much as I could before I pushed it. On the 0.9 dev board I have access to, it does not cause reboot loops. I can't fully test the button functionality as this board has been modified for deep-sleep, so if I press the User button it triggers a hardware reset rather than allowing the Lua to execute.

@UserXYZ If you're not willing to re-test after my fix, then that's entirely up to you, but we will not be keeping this issue open. If you do test, and do discover that something isn't working, please do raise an issue here, and we will endeavour to fix it. As @TerryE said though, we're all volunteers here, so there's a limit to how much we can do.

As for the bundled tool-chain, it's a STANDALONE=n build of esp-open-sdk. Currently the dev branch is built against SDK 1.4.0.

UserXYZ, 2015-11-11:
Ok guys, I managed to grab some time to test it, and it builds and installs fine, and I don't have reboot issue. As a matter of fact, it is working right now on my unmodified development  v0.9 board, and I can't see any problems... although, there are some memory leak problems that reboot the board after some time, but it might be related to my code, lua itself, my sensors or something unidentified as of yet... I've set some debug points and will see tomorrow. 

@jmattsson, on the subject of dev kit, when you say it is not standalone version, would you explain a bit and how can I use it to build the firmware? What else would I need?

TerryE, 2015-11-11:
Please google esp8266.com toolchain and read; also read issue #719.  Thank-you.

jmattsson, 2015-11-11:
Good to hear it's working for you!

Regarding the tool chain, it is precisely what you need to build NodeMCU with :)
On the dev branch the build explicitly uses SDK 1.4.0, which is why the SDK doesn't need to be bundled into/with the tool chain any more (which is what the STANDALONE switch controls).

You might also want to take a look at the Docker image Marcel prepared for building NodeMCU on - https://hub.docker.com/r/marcelstoer/nodemcu-build/

PR #743 (2015-11-09): Fix building with DEVKIT_0_9 defined.
Added lost define, reordered code to fix build error, and made key functions static as they should be. This should fix #736.