nodemcu/nodemcu-firmware #739: hostname in option in wifi.sta.config

binraker, 2015-11-07:
It would be useful to have an option in wifi.sta.config ora related function to provide a hostname to be used when connecting to a WiFi network so that one can use that to connect with rather than having to find out what IP it has been assigned.

TerryE, 2015-11-07:
This should be to do in dev `dev` since the 1.0 SDK added the API function `espconn_mdns_set_hostname()`.  I'll add it to my TODO list in maybe 2-3 weeks unless someone else submits the patch first.

Altsmirsdora, 2015-11-10:
Hi ! I'm new to Github. This enhancement is definetely what I'm looking for. I think that there would be a need and interest in adding this. Could you notify me when this will be added ?

marcelstoer, 2015-11-10:
> Could you notify me when this will be added ?

No, but GitHub will notify you when there's any news on this issue.

marcelstoer, 2016-01-17:
#878 has landed, closing this one.

closed 2016-01-17 by marcelstoer

Altamiradora, 2016-01-17:
Nice to hear about that ! When will it be compiled in the build ? Or is it already ?

marcelstoer, 2016-01-17:
We don't provide pre-built binaries anymore (for the time being), you need to build yourself: http://nodemcu.readthedocs.org/en/dev/en/build/.

Altamiradora, 2016-01-18:
Build it, tried it ! Works like a charm ! Thanks to all that contributed !