nodemcu/nodemcu-firmware #744: wifi.SOFTAP on "dev" makes restart

delinend, 2015-11-09:
Hi.

When using wifi.SOFTAP on Marcel's "dev" builder, the ESP restarts. If I use wifi.STATIONAP it works fine?!?.
Anyone knows this problem ?

Best regards

delinend, 2015-11-09:
Here build info:
NodeMCU custom build by frightanic.com
                             branch: dev
                             commit: 5d036c461bda0f50b4ba3e16ceb1fd4b80f202ee
                             SSL: false
                             modules: node,file,gpio,wifi,net,i2c,tmr,uart,u8g,rtctime,sntp
build                  built on: 2015-11-09 13:32
powered by Lua 5.1.4 on SDK 1.4.0

jmattsson, 2015-11-10:
Hi! I'm unable to reproduce this on 5d036c4 built locally with default settings, running on a devkit 0.9:
> \> print('Wifi mode:', '\n', ({'STATION','SOFTAP','STATIONAP'})[wifi.getmode()])
Wifi mode:
  SOFTAP
  \> if wifi.ap.getconfig then print('',wifi.ap.getconfig()) end print('', wifi.ap.getmac(), '\n', wifi.ap.getip()) print('AP clients:') for k,v in pairs(wifi.ap.getclient()) do print('', k, v) end
AP config:
	DiUS sensor a454ae	nil
	1a:fe:34:a4:54:ae	
	192.168.4.1	255.255.255.0	192.168.4.1
AP clients:
	e8:15:0e:73:80:e8	192.168.4.2

Please provide more information, such as full log of steps taken to trigger issue.

delinend, 2015-11-11:
I'm using the ESPbootloader from Sebastian https://github.com/sebastianhodapp/ESPbootloader

Here the log, after I installed the debug "dev" branche:
Preparing HTML Form
Setting up Wifi AP
del if0
usl
mode : softAP(1a:fe:34:a5:96:4f)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
espconfig
1234567890
bcn 0
del if1
usl
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
Setting up webserver
net_create is called.
TCP server/socket is set.
net_start is called.
TCP port is set: 80.
Setting up Webserver done. Please connect to: 192.168.4.1
Fatal exception 28(LoadProhibitedCause):
epc1=0x40231cf5, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000d0, depc=0x00000000
ò›?¤<1Õ§©é1H)„ÜNÿÐmode : softAP(1a:fe:34:a5:96:4f)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100

_flash_used_end:4026a723
fs.start:4026c000,max:10000
mount res: 0
Task task_lua started.
SIG_LUA received.

NodeMCU custom build by frightanic.com
	branch: dev
	commit: 0e5582620b51dd10dd5a39608b5e43103678686d
	SSL: false
	modules: node,file,gpio,wifi,net,i2c,tmr,uart,u8g,rtctime,sntp
 build 	built on: 2015-11-10 07:53
 powered by Lua 5.1.4 on SDK 1.4.0
lua: cannot open init.lua

jmattsson, 2015-11-11:
Right, it's hitting a hardware exception somehow. Do you have access to the corresponding app/mapfile for your firmware? I'll need that to get any sort of idea what is really going wrong here, since I can't reproduce it myself.

delinend, 2015-11-11:
@jmattsson Yes, I have tryed many times to Flash with the "master" (that works fine) and "dev" that fails, Regarding the app/mapfile, I use Marcel's builder here: http://frightanic.com/nodemcu-custom-build/index.php

jmattsson, 2015-11-11:
Hey, @marcelstoer any chance you could dig out the correct app/mapfile for me here?

jmattsson, 2015-11-11:
@delinend So if I understand this correctly, on the first boot you do have an init.lua which runs this bootstrap thingy, but then it crashes and after the reboot you don't have an init.lua any more? Do you actually connect to the softAP before the crash, or does it crash immediately after the "Please connect to: 192.168.4.1" message is printed?

marcelstoer, 2015-11-11:
@jmattsson Sorry, on the logs are retained, the binaries are gone.

@delinend You'd have to configure and build the firmware yourself (e.g. with [my Docker image](https://hub.docker.com/r/marcelstoer/nodemcu-build/)) in order to provide the `app/mapfile` to Johny.

delinend, 2015-11-11:
@jmattsson No, for testing, I have renamed init.lua to start.lua. So no init.lua.
The last thing I see, is the "Please connect to: 192.168.4.1" message. Then it's reboot.
The strange thing is, if I change from "SOFTAP" to "STATIONAP", it's not rebooting under "dev", but I can't see any content on my browser from http://192.168.4.1. It's just a "blank" html page.
Hmm. So maybe something is different in the "net.createServer(net.TCP)" setup in "master" vs "dev", that mess up the wifi setup.

marcelstoer, 2016-04-10:
Doesn't seem like a bug in the firmware to me but rather something for http://nodemcu.readthedocs.org/en/dev/en/support/.

closed 2016-04-10 by marcelstoer