nodemcu/nodemcu-firmware #748: Document module origin in wiki

marcelstoer, 2015-11-10:
If we ever want to extract non-core modules into their own repository we need to know the origin of those modules. Who contributed it? Who wrote it (if not contributor)? Who has been maintaining it?

TerryE, 2015-11-10:
We also need a standard structure, naming conventions, etc. and possibly some standard host-side scripting language to execute the unpack, install, etc.  I note that eLua requires Lua 5.1 + lua-filesystem to be installed on the host to make eLua (and that's why luac.cross does -- because this is straight eLua funtionality).  I think the we should do likewise and look  at the PECL and CPAN approaches for inspiration.

marcelstoer, 2015-11-19:
WIP: https://github.com/nodemcu/nodemcu-firmware/wiki/Module-origin

closed 2015-11-20 by marcelstoer