nodemcu/nodemcu-firmware #754: Change default value to DIO

TerryE, 2015-11-11:
#752 Refers.  Vowstar:

*    You committed this to `master` not `dev` and as I've raised in #745, we have proposed to stop actioning PRs to `master` so we can sort this mess.
*    You have raised a change, committed yourself and closed the issue without giving time for comment or feedback.  Again I feel it better that we don't do this.
*    You have changes the tool's default to DIO.  Yes, the the QIO-capable board will work in DIO mode, but people will be curious as to why flashing new firmware takes twice as long, so I feel the perhaps we should have discussed the wider implications first.

I consider this an open issue.

marcelstoer, 2015-11-11:
Thanks Terry, I fully agree.

jmattsson, 2015-11-11:
The more issues & fixes that come up in esptool.py, the more I'm inclined to agree with Marcel that it's not a good idea to maintain our own copy of esptool in the tree.

marcelstoer, 2015-11-12:
@jmattsson I picked up work on that old PR-attempt I started a while back to remove `esptool.py`. However, it's definitely gonna take a while because I need to prioritize my non-NodeMCU task list higher at the moment.

marcelstoer, 2015-11-12:
@vowstar Also, I just realized you created a `esptool-dio-patch-1` branch right in _this_ repository. I thought we had agreed to open feature branches only in our own forked repositories? To me this is really bad practice.

vowstar, 2015-11-12:
OK, I'll close it.

vowstar, 2015-11-12:
Some new user don't know how to use esptool.py, I think this problem is come from our documents.

nodemcu, 2015-11-12:
Hi, @vowstar, I know that you are bloody busy these days, and may have missed some discussions and issues. FYI, there is a guidelines for contributors there https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md updated by @TerryE. it's a better rules and practice. and I believe that you are very like me, willing to learn a better way to deal with github and opensource project. so let's learn together my friend.
oh, and please check the discussion on email, there's a lot information there. you are on the CC list, and I hope hear from you.

vowstar, 2016-01-01:
@nodemcu I'm so sorry for late reply, these days I'm to busy on coding. I find this discussion again by searching DIO problem just now. I've carefully study the https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md updated by @TerryE, I think it is very important for this project, and I will follow this guide. I very much admire your technology, hope to be able to work together to make this project a better :smile:

closed 2016-03-17 by TerryE