nodemcu/nodemcu-firmware #757: concurrent PWM outputs count

RafaelKa, 2015-11-12:
How many PWM outputs are available at same time?
According to documentation its 3 PWMs:
https://github.com/nodemcu/nodemcu-firmware/wiki/nodemcu_api_en#description-70

but in your code are 6 possible:
https://github.com/nodemcu/nodemcu-firmware/blob/bb9dd62882b89d40f5d0df2c009c972f64b7f965/app/driver/pwm.c#L30
https://github.com/nodemcu/nodemcu-firmware/blob/bb9dd62882b89d40f5d0df2c009c972f64b7f965/app/include/driver/pwm.h#L4

Which is also correct?

PS: I do not have currently ESP8266 devices to test it!

TerryE, 2015-11-12:
Sorry but please see #719.

closed 2015-11-12 by TerryE

RafaelKa, 2015-11-12:
So i see now in [wiki](https://github.com/nodemcu/nodemcu-firmware/wiki), there are 6 PWMs
```2014-11-11
add file.seek() api to file module
now max 6 pwm channel is supported```
Despier of #719 this should be fixed in the docs!

TerryE, 2015-11-12:
@RafaelKa, you can create the PR (Pull Request) as easily as we can.