nodemcu/nodemcu-firmware #758: Bug with MTU negotiation for sending messages on TCP sockets

apESP8266, 2015-11-12:
Hello.

In the wake of the creation of a Web server using the controls below, I noticed that it responded correctly to a client connected to the local network, failing to respond in case it was connected via a VPN.
    srv=net.createServer(net.TCP);       
    srv:listen(80,function(conn)         
        conn:on("receive", function(conn, req)
           process(conn, req);
            end)
        conn:on("sent",function(conn)           
            conn:close();           
            conn = nil;              
            end)
        end)      

In exploring the reasons for the malfunction in the connection via VPN I noticed that prior to the exchange of messages was shown the MTU 1360 client-side and MTU 1460 server-side.

Apparently the ESP8266 not only don’t adopt the MTU of the lowest client value but also conveys messages with more than 2000 bytes, resulting in truncated messages.

This problem is already solved? 

It will be easy to include a configuration command in "net" or "wifi" modules where we can change the default MTU? (apparently de default is 1460)

thank you

karrots, 2015-11-12:
Typically one would not adjust the MTU unless your first hop gateway (router) has a lower MTU. What does need to adjust is the MSS on the TCP packet. This is typically done automatically using [Path MTU Discovery](https://en.wikipedia.org/wiki/Path_MTU_Discovery). If ping is blocked on the path this can have a hard time adjusting. Many VPNs clamp the MSS to the tunnel size. If this is already done and traffic on regular OSs work fine then it may be a SDK issue and not something the NodeMCU team can fix. 

See #719 this seems like a forum issue.

TerryE, 2015-11-12:
Humm, I agree that this is unlikely to be a nodeMCU firmware issue and more an SDK / LWIP one, but @apESP8266 might be doing the wider community a befit by researching this and posting with Espressif so we can piggyback on SDK 1.5 or whatever.

apESP8266, 2015-12-09:
The problem I detected actually was checked on the version 0.9.5 and is no longer present in version 1.4.0.

In version 1.4.0, if sent more bytes than those supported by the channel to the client, the message is split so as not to exceed the negotiated MSS, and consequently the malfunction via VPN is resolved.

I noticed, however, that in the case of attempting to send more than 1460 bytes the sending fails without errors.

marcelstoer, 2016-03-01:
> I noticed, however, that in the case of attempting to send more than 1460 bytes the sending fails without errors.

In case this is still an issue pls open a separate issue.

closed 2016-03-01 by marcelstoer

AdamLuter, 2017-02-21:
Hey. Does someone has already found a solution to this problem? It still exists even in the SDK V2.0.0 Regards.

apESP8266, 2017-02-22:
Hey AdamLuter

The solution I found was to reduce the maximum packet size of each packet
sent.

First I saw the quantity of bytes added by the VPN in each packet and then
reduced those bytes to the maximum packet size.

Not knowing the number of bytes added in each packet sent over the VPN, I
suggest gradually reducing the maximum packet size until the messages pass
uncut.

In my case it works well with a maximum packet of 1350 bytes (it already
includes a small margin for any overhead that may arise)

AdamLuter, 2017-02-23:
Can I do this when I send a long html to the browser? I'm using Arduino SDK. I should use the "setContentLength" and "sendContent" instead of just "send"?

marcelstoer, 2017-02-23:
> I'm using Arduino SDK.

Huh? Then you're wrong here. This is the NodeMCU firmware project.

AdamLuter, 2017-02-23:
Ouu.. Sorry, I'm first time here...