nodemcu/nodemcu-firmware #760: "dev" build not booting anymore.

delinend, 2015-11-13:
The last days, I can't get "dev" builds to work on my ESP’s.
I have tried 3 times today and yesterday, to make "dev" build's from Marcel great Cloud builder. Downloaded the "integer" and Flashed with NodeMCU-Flasher. But on power up of the ESP, I only get a lot of garbage on my 9600 baud console :-(
If I Flash an old bin from 9/11-2015, it boot's fine. Any ideas whats going wrong the last days? It had worked many timed for me, but not now.
Here the 11 modules that I use: node file gpio wifi net i2c tmr uart u8g rtctime sntp

dnc40085, 2015-11-13:
My guess would be that you are using an ESP module that has a 512k flash, If so... then it is most likely that with the modules you have selected the firmware is too big now.

On a side note, I can only successfully get my 512k to boot with these modules enabled: `node file gpio wifi net tmr adc uart`.

TerryE, 2015-11-13:
+1. We've talked with Marcel about adding extra checks to the builder to report on flash size, bit the one that tips your over the top is probably u8g.  A good hint is the size of the bin files. But the truth is i you want more than a minimal core set of models then you need 1024Kb min Flash -- a module such as the 12E.  But sorry to say that this isn't a bug with the firmware and so this issue falls under #719, and I will close this.

closed 2015-11-13 by TerryE

delinend, 2015-11-14:
Hmm Thanks. It's hard to figur out, when the Flash/bin file, is under 512K. Ex. the bin I made, has size (421 KB (431.555 byte)). ?!? Is there a max bin size, that I can calculate. Is the reason for confusion, that the bin file is not containing RAM bank's used for WiFi setting and more ?