nodemcu/nodemcu-firmware #762: Basic http client not sending data over client

drio, 2015-11-14:
I am using nodemcu `branch: dev` on a esp8266. The device is acting as a basic http client. Right now I am testing thing manually to try to figure what I am doing wrong so my lua code looks like:

```lua
wifiUp = function()
  wifi.setmode(wifi.STATION)
  wifi.sta.config("thewinetwork", "xxxxxx")
  wifi.sta.connect()
end

check = function()
  gpio.mode(3, gpio.OUTPUT)
  ip = wifi.sta.getip()
  if ip ~= nil then
    print(ip)
    gpio.write(3, gpio.LOW)
  end
end

sendData = function()
  conn=net.createConnection(net.TCP, 0)
  conn:on("receive", function(conn, payload) print(payload) end)
	conn:connect(80,'192.168.1.212')
	conn:send("POST /proxy HTTP/1.1\r\n")
	conn:send("Host: localhost\r\n")
	conn:send("Accept: */*\r\n")
	conn:send("User-Agent: Mozilla/4.0 (compatible; esp8266 Lua; Windows NT 5.1)\r\n")
	conn:send("\r\n")
	conn:on("sent",function(conn)
						  print("Closing connection")
						  conn:close()
					  end)
	conn:on("disconnection", function(conn)
			  print("Got disconnection...")
  end)
end
```
I run the first two functions to make sure I have internet connection. Once I know I have an IP, I fire up the sendData function.

TerryE, 2015-11-15:
Sorry, but you need to read #719 and use these forum resources for this type of query

closed 2015-11-15 by TerryE

drio, 2015-11-15:
I found the problem. It was really a combination of things. I believe this is a very idiomatic nodemcu code. I hope it helps to others:

```