nodemcu/nodemcu-firmware #763: Bridge Access Point and Station network.

vvikas, 2015-11-15:
I am using STA to connect to WiFi and AP to create a new WiFi connection.
I want to amplify my original WiFi signal by using STA network via AP network,
but there seems to be no way to bridge these two interfaces.

vvikas, 2015-11-16:
@nodemcu

dnc40085, 2015-11-19:
Are you trying to use an ESP8266 as a wifi repeater/range extender?

vvikas, 2015-11-20:
Thanks, will try it.

evgi9, 2015-12-07:
@vvikas  Any news on this issue?

dnc40085, 2015-12-14:
@vvikas Did it work?

vvikas, 2015-12-19:
I tried it. It worked but only for a small duration then it just disconnects from the Station Network. Nothing conclusive but maybe the device power needs to be increased.

davemaster, 2015-12-19:
Try sending a periodic data to the ES8266. This will keep connection alive.

closed 2016-08-07 by marcelstoer