nodemcu/nodemcu-firmware #768: Problem with booting "dev" builds, on 1Mbyte moduls

delinend, 2015-11-16:
Hi.

First: Wow.. The Issues in here, is getting "Closed" very quickly. It's a bad thing.

I made this Issue: https://github.com/nodemcu/nodemcu-firmware/issues/760 regarding booting problems, and yes... my problem is building U8G with Marcel Cloud builder and Flash it into a 512K ESP.
But even when using a 1M ESP modul, I still can't get it to boot. I have selected this moduls: node file gpio wifi net i2c tmr uart u8g rtctime sntp
I have tryed to make my one "dev" build, on a Rasberry Pi2, SDK 1.4.0 and end up with the 0x00000 and 0x10000 file, wich I Flashed with the NodeMCU-Flasher (230400/1MByte/40MHz/DIO), and this works fine !?!
What am I doing wrong, when flashing the build from Marcel's Cloud builder on a 1M ESP? Is there a special setting in NodeMCU-Flasher ?

Thanks and best regards

jmattsson, 2015-11-17:
This is quite possibly related to the problem observed after #729 was merged, for which there is a pending fix in #770. You could try a build with SSL included in the meantime.

As for issues being closed, that is discussed in #719. The volunteers have quite limited amount of time available, and if people are not willing to provide the requested & necessary information to help us help you, the issues will be closed.

delinend, 2015-11-17:
Wow. This solved the boot problem, as I included the SSL in the build.
Thanks jmattsson, to be so kind and take the time to respond.

closed 2015-11-17 by delinend

TerryE, 2015-11-18:
@delinend, I realise that it may seem harsh closing issues that we can't fix, but in the end this is fairer than just leaving them open and ignored as was the de facto practice previously.