nodemcu/nodemcu-firmware #778: Query re wifi.sta.config()

alexgian, 2015-11-19:
This may not be an issue but just ignorance on my part - if so, my apologies in advance!
I am using NodeMCU 0.9.6 with Lua 5.1.4, build 20150704.

I can connect fine to networks that use normal text string passwords, but with networks where the password is a five-byte string I cannot.

I have tried a few different formats with no success.
Say the password is A1B2C3D4E5.  I have all tried the following, including their lowercase versions:

"A1B2C3D4E5"
"A1 B2 C3 D4 E5"
"A1-B2-C3-D4-E5"
"A1:B2:C3:D4:E5"
and even
"\161\178\195\212\229" (in which case it printed "stdin:1: pwd:0,8~64" but still didn't work)
Is there something I am missing here, or there some problem?

Thanks

karrots, 2015-11-19:
You'll want to try a newer build from the dev branch. See #719.

http://frightanic.com/nodemcu-custom-build/

dnc40085, 2015-11-19:
I believe the minimum password length for WPA/WPA2 is 8 characters. 
Is it possible to change the 5 byte password to an 8-63 byte password?

alexgian, 2015-11-19:
OK, thanks, dnc40085.
The WLANs I was trying were protected with WEP-64, which I will now assume is not compatible with NodeMCU.
Fortunately, I have the capability of changing them over to WPA2 (I'm not really sure what the difference is, anyway). I will do that and not waste more time.  

Does this mean, however, that someone stuck with a WEP-64 system would not be able to use NodeMCU?

karrots, 2015-11-19:
WPA2 should be supported by any 802.11g or better device.

alexgian, 2015-11-19:
OK, thanks, all done, closing this now.

closed 2015-11-19 by alexgian

dnc40085, 2015-11-19:
NodeMCU should be capable of connecting to a WEP encrypted network provided that the password is 8-63 bytes long. However...

WEP encryption should not be used by anyone since it is [no longer secure](http://www.isaac.cs.berkeley.edu/isaac/wep-faq.html). WiFi networks secured with WEP are [ridiculously easy to crack](http://www.freehowtohackwifi.com/wep-attacks/fern-wifi-cracker/), all you need is a backtrack or Kali Linux live cd, a wifi adapter that supports packet injection and about 30 minutes and it will spit out the password needed to connect to the network.

The same goes for wpa/wpa2 networks that have the [WPS feature enabled](http://www.howtogeek.com/176124/wi-fi-protected-setup-wps-is-insecure-heres-why-you-should-disable-it/), which is even worse for security since [once you have the WPS pin](https://scotthelme.co.uk/wifi-insecurity-wps/) the router gives you the password even if it has been changed. 

TL;DR: You should use WPA/WPA2 with the WPS feature disabled.

alexgian, 2015-11-19:
Thanks for taking the trouble to clarify that, even after I closed it!
Really appreciated.
It's been long overdue getting up to speed on WiFi security  ;-)

wordsforthewise, 2016-02-02:
Even with my WEP key as hex, it's not working.  Anything else I should be doing to connect?  Here's what I'm doing now:

wifi.setmode(wifi.STATION)

tmr.stop(1)
tmr.alarm(1,1000,1,function() print(wifi.sta.status()) end)

SSID = "hide your kids hide your wife"
pass = '11111122222233333344444499'

wifi.sta.config(SSID, pass)
wifi.sta.connect()

alexgian, 2016-02-02:
I found that WEP does not work *at all* for me.  
(My router interface does not support 8-bit passwords, it's one of those pre-formatted forms.)
YMMV, but if you can, try WPA/WPA2.

TerryE, 2016-02-03:
This is all standard SDK functionality.  This discussion belong on a support forum, not this issues list.

wordsforthewise, 2016-02-03:
What's the harm?  When I googled the issue this is one of the few results that turned up.

TerryE, 2016-02-03:
We are not a support forum, and we want to focus on development, not Lua developer support.  See #719.  We will delete posts which deliberately abuse the scope of this list.