nodemcu/nodemcu-firmware #784: Waiting answer from ESP - Timeout reached. Command aborted. on Dev Branch

confile, 2015-11-21:
I use the latest Dev Branch commit: b216e6a809edd035d70536a691c4a07ac4b256a9

I build all modules from the dev branch using this tool: http://frightanic.com/nodemcu-custom-build/

When I try to upload a script in ESPlorer with Send to ESP the I get the following error (I turned debug on): 

```
> Task task_lua started.
Task task_lua started.
Task task_lua started.
Waiting answer from ESP - Timeout reached. Command aborted.
```

Using the latest master I had no problems.

devsaurus, 2015-11-21:
I can confirm such issues with ESPlorer at the moment. I'll investigate what's been broken.

steffimueller, 2015-11-21:
+1 I have the same problem. Not working on dev branch

devsaurus, 2015-11-21:
Ok, checked the sources. ESPlorer seems to use CR&LF for most obvious things per default, but falls back to CR for other functions like file management and file transfer. Don't have a clue why. However, it's been #777 which broke things.
@jmattsson seems that we didn't have such creativity on the radar screen. I'd propose to handle all possible newline combos in `readline()` - PR will follow.

devsaurus, 2015-11-22:
@confile @steffimueller if possible, please check whether #786 improves things for your ESPlorer setup.

eyaleb, 2015-11-22:
Tested #786 and now both `screen` and `luatool.py` work.

confile, 2015-11-23:
I tested but I got the same problem. 

```
Waiting answer from ESP - Timeout reached. Command aborted.
```

steffimueller, 2015-11-23:
Same problem here. The problem is not fixed.

eyaleb, 2015-11-23:
Repeated a `git pull` of dev, build fw, flash it (OK) upload programs with `luatool.py` (OK) attached UART to `screen` (OK). I change `user_version.h` every time and did check the bootup message and it really runs the latest image.

BTW I do my own builds and it is as simple as typing 'make'.

confile, 2015-11-23:
I built the dev branch float version with this tool: http://nodemcu-build.com/

Then I use the nodemcu flasher from here: https://github.com/nodemcu/nodemcu-flasher to flash it. 

Upload .lua files with ESPlorer not working.

devsaurus, 2015-11-23:
> Upload .lua files with ESPlorer not working.

I'm not able to generate any issues with current `dev` here. All functions of ESPlorer that failed prior to #786 are definitely working now for me.
Flashed both [float](http://nodemcu-build.com/builds/nodemcu-dev-7-modules-2015-11-23-21-23-48-float.bin) and [integer](http://nodemcu-build.com/builds/nodemcu-dev-7-modules-2015-11-23-21-23-48-integer.bin) builds from frightanic and uploaded .lua files with ESPlorer 0.2.0-rc2 on Linux. All of these were failing before and are fine now:
* "Reload" button -> ok
* "Upload..." button -> ok
* "Save to ESP" button -> ok
* Compile and remove with File Manager -> also ok

Is there any difference to your setup? Older/newer ESPlorer version, different OS? Which specific functionality throws this error - am I missing something in my list? Do you see this error also with the images linked above? Does it happen only for big files or also for [telnet.lua](https://github.com/nodemcu/nodemcu-firmware/blob/dev/lua_examples/telnet.lua)?
The previous failure mode with CR-only newlines was pretty obvious when inspecting ESPlorer's source. Now I'd really need a local testcase to dig in deeper.

confile, 2015-11-24:
@devsaurus Did you get the dev firmware generated from here: http://nodemcu-build.com/

Have you tried that?

confile, 2015-11-24:
I use ESPlorer 0.2.0-rc2 on MAC.

devsaurus, 2015-11-24:
> Did you get the dev firmware generated from here: http://nodemcu-build.com/
> Have you tried that?

Yes. The build images I linked in my previous post have been generated with the nodemcu-build.com service.
Have you tried them?

I still need more details of your use case to reproduce the issue: What is the exact sequence of actions you perform until you get the "Timeout reached..." error?
Please post a .lua script you that use in this sequence.

What items are selected in the "Settings" tab? I have nothing selected except "NodeMCU" in the "Select firmware" group. "Delay after answer" is 0ms.

TerryE, 2015-11-24:
@confile if you start esplorer from a terminal window then you get a load of diagnostics that might help.  This might be some OSX-specific issue.

confile, 2015-11-25:
@devsaurus I tried you [image float version](http://nodemcu-build.com/builds/nodemcu-dev-7-modules-2015-11-23-21-23-48-float.bin) With this I can upload without any problems. But, I use this [init.lua](https://gist.github.com/confile/e9b794008a990482d789) to connect to wifi. I do not get a wifi connection when I use the firmware from the dev branch. When I use the firmware from the master branch I get a connection without any problems. 

Second example. I tried a firmware from the dev branch using all 31 modules. [Here is a link to this firmware](https://dl.dropboxusercontent.com/u/47067729/nodemcu-dev-31-modules-2015-11-23-08-05-27-float.bin). When you try this firmware and upload my init.lua script you get the "Timeout reached..." error. 

Steps to reproduce: 
1. Flash 31 modules dev branch firmware
2. Open ESPlorer
3. Open init.lua
4. Click "Send to ESP"

The tabs that are selected are these:

[image]

@TerryE The only message in the terminal is: 

```
INFORMATION: Waiting answer from ESP - Timeout reached. Command aborted.
```

devsaurus, 2015-11-25:
@confile thanks a lot for your feedback! Very valuable for me since this confirmed that the issue is somehow dependent on the module selection or size (my image is also from dev branch). And furthermore it shows that I can't reproduce at the moment - still waiting for an ESP12E which can accommodate a full 31-module image. All ESPs I have atm are 512kB.

Nevertheless, I'll keep on digging into ESPlorer sources. Got the impression that there are still timing topics to be resolved.

You could try to set "Delay after answer" to 100ms as a workaround. It might deliver more stable uploads.

confile, 2015-11-25:
@devsaurus Still if you check my init.lua you see that even with your 7 module dev firmware you get a different result. You do not get a wifi connection.

confile, 2015-11-25:
Where can I set "Delay after answer"?

TerryE, 2015-11-25:
Try the settings tab.  Bottom left corner,

devsaurus, 2015-11-25:
> Second example. I tried a firmware from the dev branch using all 31 modules.

Download doesn't work - **Error (404)**

Please try my experimental build of ESPlorer from here: [Snapshot of 0.2.0-rc2 with interlock for send/receive](https://github.com/devsaurus/ESPlorer/releases/tag/0.2.0-rc2_interlock). It should work standalone, but you might need to overwrite the original .jar file to get this one to run.
I've identified some race conditions in the source code which lead to problems when uploading code directly from the editor with "Send to ESP". Not 100% your issue since there's no Timeout error message, but it might be related. Let me know how this performs.

devsaurus, 2015-12-01:
@confile @steffimueller do you see any improvement with [Snapshot of 0.2.0-rc2 with interlock for send/receive](https://github.com/devsaurus/ESPlorer/releases/tag/0.2.0-rc2_interlock)? The PR has been accepted for upstream and I'd like to figure out whether this issue is solved now.

confile, 2015-12-02:
@devsaurus I tested you ESPlorer Snapshot but it does not change anything. The issue still does exist.

steffimueller, 2015-12-02:
@devsaurus @confile same for me. Does not fix issue.

devsaurus, 2015-12-02:
@confile too bad, it fixed the issue with "Send to ESP" for me. I got an ESP-12E in the meantime and can flash the build with all modules enabled as you did. But well, the only hiccups which remain beyond my patch are reported here: 4refr0nt/ESPlorer#23. There's no fix available at the moment and it's not related to "Send to ESP" at all.

@steffimueller What are your steps to get the error? Same or different from https://github.com/nodemcu/nodemcu-firmware/issues/784#issuecomment-159546827?

shimosaurus, 2015-12-03:
I commented line `NODE_DBG("Task task_lua started.\n");` in *user_main.c*. That was many prints in terminal after each Lua command. It broken communication if DEVELOP_VERSION is defined.

devsaurus, 2015-12-06:
Thanks for the reminder @shimosaurus! These `Task task_lua started.` messages definitely break the communication with ESPlorer. IMO there's *either* debug mode *or* the fancy ESPlorer features :unamused:

nickandrew, 2015-12-06:
@devsaurus I deleted the task_lua message and disabled the net debug messages, and I have no problem now with ESPlorer in debug mode.

devsaurus, 2015-12-09:
I got the impression that this issue hit a dead end now, so I'm going to close it with the following conclusions:
* It's not recommended to run ESPlorer on a debug build of the firmware. The additional messages interfere with the command-response concept that's been deployed by ESPlorer to implement high level functionality.
* Support for CR-only newline signalling by ESPlorer (and other tools) has been re-added by #786. I consider this to be a reliable solution wrt timing.
* Change in nodemcu's micro-timing behaviour triggered a race condition in the implementation of ESPlorer's command-response concept. PR 4refr0nt/ESPlorer#22 to fix this in ESPlorer has been submitted and was merged.
* Further improvements have been identified and reported as per 4refr0nt/ESPlorer#23. This issue is still waiting for a solution.

closed 2015-12-09 by devsaurus

npashine, 2016-04-11:
Kindly help to resolve the attached error.
[image]

pjsg, 2016-04-12:
I suspect that this is because you are running the dev branch and have not set the speed to 115200.

npashine, 2016-04-12:
so what all the things i should do to make it correct.
And for your information it was running fine with 9600 before after some time this error started coming and also now it is happening with another esp8266 which i used so i decided not to waste another.
kindly help as this is major part of my project.

TerryE, 2016-04-12:
@npashine, as Philip says, the dev branch runs at 115200 baud not 9600. I use explorer with no problems. This isn't a nodeMCU bug.

Pointhairedboss, 2016-06-17:
How can I get an older version without having to build it myself?
On the download site, there isnt even a version number!

marcelstoer, 2016-06-17:
> How can I get an older version without having to build it myself?

You can't, see https://github.com/nodemcu/nodemcu-firmware#releases.

arakus, 2016-10-19:
If you have problem with communication ( Waiting answer from ESP - Timeout reached.  ), first check power supply of ESP8266, add decoupling caps on power line etc...

adamtal3, 2020-04-13:
Simply clicking on "FS Info" after getting the timeout error worked for me.

PR #786 (2015-11-22): Re-add detection of CR as newline
#777 removed newline detection for CR-only case as discussed in #765. This case is now supported again but requires a more complex filter logic to suppress multiple newlines for CR&LF and LF&CR sequences.
Fixes #784 and #785.

I'd propose to merge this as soon as possible once there's positive feedback from the bug issues.
commit 6488c954 (2015-11-22): Merge pull request #786 from devsaurus/fix-cr

Re-add detection of CR as newline.
Fixes #784 and fixes #785.