nodemcu/nodemcu-firmware #789: Rework dht_lib.lua

TerryE, 2015-11-24:
THe comment thread on 3fd7396 refers.  Just opening this as an issue so that I can assign it to myself.

TerryE, 2015-11-24:
@Ritterbrater, added you just so you can track any development of this in future.  Any review comments welcome.

Ritterbrater, 2015-11-25:
@TerryE thank you!
Something is now totally weird :disappointed: 
I updated my firmware (I was running on Feb 2015 until now) and now I run into timing troubles.
I dump the bitStream[] array and the last three "bits" have a value of 10 --> so something is too slow. 
I will continue tomorrow. 
Although my problem is independent from the byte conversion. So maybe somebody else should test now.
New firmware:

NodeMCU custom build by frightanic.com
	branch: master
	commit: d7626961199a7375ca27ec81e5dc6db7571a9909
	SSL: false
	modules: node,file,gpio,wifi,net,tmr,uart,ow,bit
 	built on: 2015-11-25 00:33
  powered by Lua 5.1.4

TerryE, 2015-11-25:
This might be as a result of the changes to use an exception handler so that unaligned RO data within the firmware can be read from Flash.  I might have a look tomorrow.  Also try an integer build as the FP is done in S/W and moving from an integer to a FP build will fundamentally impact timing.

UncleRedz, 2015-11-25:
What is the difference between the dht_lib.lua compared to the app/modules/dht?

 I've been using using the modules version on both master and dev (from around the time when WDT fixes for 1.4 was committed) and have not experienced any issues when using integer builds.

Ritterbrater, 2015-11-25:
@TerryE 
* integer build from dev trunk used: working fine again. (Although I nearly burnt the sensor testing it at 5 V arduino again)
* please add somewhere the hint, that the flash should be formatted after loading the firmware. I had several file corpses of strange file sizes after loading. I thought, they are from the new firmware. 
* float build from dev trunk: timing problems again. The only real difference I see at the float firmware happens at uploading with ESP8266 LuaLoader (maybe it has something to do with this unaligned writing):

> `> >` dofile("LLbin.lua")
cannot open LLbin.lua
stack traceback:
	[C]: in function 'dofile'
	stdin:1: in main chunk

This does not happen normally. (sorry for bad formatting)
IMHO the problem of the float build is hidden somewhere in the function switching the GPIO pin from output to input. I may take so much time that the first bits are not read. But I have no knowledge about the different timings in both firmwares. I have an idea for a hack that might solve the problem: the ESP8266 does following:
* pin output to high
* pin output to low for a long time to wake up the DHT22
* pin output to high
* pin set to input
* pin waiting for input low

What about not setting the pin to high? There is a pull up resistor in the ESP8266 itself (- as far as i read the C routine) and normally one should put a pull up resistor also between GPIO and VCC. When switching from output low to input, the level should change to high automatically. And the DHT22 does not do anything until there was a high level on the data pin. But I don't know if this is forbidden to do: because there is a change from output to input AND a level change.

vowstar, 2015-12-05:
The speed of lua is not fixed in different version, it is a challenge.

@UncleRedz dht_lib.lua is a greatwork which uses lua to read timing strict dht sensor, need adjust in different version. The app/modules/dht is universal dht lib uses C to read dht sensor.

TerryE, 2016-02-17:
See #1048

closed 2016-02-17 by TerryE