nodemcu/nodemcu-firmware #790: ws2812 buffer

Alkorin, 2015-11-24:
Dunno if it's only me, but coming from C, playing with immutable strings with the ws2812 library is painful: when you have a long strip of leds, it's easier to have a buffer which represent your current strip and be able to set any byte in it.

I've written a PoC (my first hack with a lua library...) of a such buffer:
https://github.com/Alkorin/nodemcu-firmware/compare/dev...Alkorin:ws2812-buffer

Current API:
```
buffer = ws2812.newBuffer(nbLeds)
ws2812.setBuffer(buffer, led, G, R, B)
ws2812.writeBuffer(pin, buffer)
```

Next step, be able to call set on the buffer and have:
```
buffer = ws2812.newBuffer(nbLeds)
buffer.(led, G, R, B)
ws2812.writeBuffer(pin, buffer)
```

Any advice are welcome :)

devsaurus, 2015-11-26:
I like your idea of having a persistent and structured buffer.

You allocate the buffer as Lua userdata - `ws2812_writergb()` could detect and accept this as payload parameter for the LEDs in parallel to the current string type. Then there's no need to have `ws2812.writeBuffer()` at all.

Merging the set functionality into the buffer object is for sure quite elegant and keeps the API lean.

Alkorin, 2015-11-26:
Thanks @devsaurus will do it :)

Alkorin, 2015-11-27:
Done, new API:

```
buffer = ws2812.newBuffer(nbLeds)
buffer:fill(G, R, B)
buffer:set(led, G, R, B)
ws2812.write(pin, buffer)
```

Simple led chaser:

```
i=0;
a = ws2812.newBuffer(16);
tmr.alarm(0, 100, 1, function() i=i+1; a:fill(0, 0, 0); a:set(i%16, 0, 255, 0); ws2812.write(4, a); end)
```

Still have to understand how to declare my metatable without `LUA_OPTIMIZE_MEMORY`

jmattsson, 2015-11-28:
@Alkorin Try [this link](http://www.eluaproject.net/doc/v0.9/en_arch_ltr.html) for the eLua LTR internals.

TerryE, 2015-11-28:
@Alkorin, Thomas, a small Lua style optimization.  You should avoid using globals as locals / upvalues generate shorter and more efficient runtime code. Also have you thought of provided a `write(pin)` method to the buffer so that this could be written:

```Lua
local i, a = 0, ws2812.newBuffer(16)
tmr.alarm(0, 100, 1, function() 
     i=i+1
     a:fill(0, 0, 0); a:set(i%16, 0, 255, 0); a:write(4)
   end)
```

BTW the **;** are syntactic sugar for correctly written Lua and are entirely optional.  

As far as metatables, go just look at any of the modules which use them, e.g. net.c and its use with sockets.

Alkorin, 2015-11-28:
I'm really new to Lua, that's why I post here to have some advices :)

Thanks for the pointers, will take a look at them

Alkorin, 2015-11-29:
@TerryE write method added as well as a getter.

Fading led chaser:
```lua
local i, a = 0, ws2812.newBuffer(16); a:fill(0,0,0); tmr.alarm(0, 100, 1, function()
        for i = 0, (a:size()-1) do
                g, r, b = a:get(i)
                a:set(i, g/2, r/2, b/2)
        end
        i=i+1
        a:set(i%a:size(), 0, 255, 0)
        a:write(4)
end)
```

Will now fix metatables

closed 2016-03-27 by Alkorin