nodemcu/nodemcu-firmware #791: not connecting to router after reset

YonatanHanan, 2015-11-25:
Hi,

If the router is shutdown and the esp is trying to connect to it,
and i tern the router back on the esp dosent manege to connect.

but if the router is on and the esp tryes to connect it works stragiht away.

what can the problem be?

here is the code

```javascript

file.open("settings.lua", "r")
lines = {}
for i = 0, 7, 1 do
    lines[i] = file.readline()
    lines[i] = lines[i]:gsub("%\n", "")
end
file.close()

wifi.setmode(wifi.STATION)
wifi.sta.config(tostring(lines[3]),tostring(lines[6]))
wifi.sta.autoconnect(1)

function check_wifi()
 wifi.sta.connect()
 local ip = wifi.sta.getip()

 if(ip==nil) then
   print("Connecting...")
 else
  tmr.stop(1)
  print("Connected to AP!")
  print(ip)
 end
 
end

tmr.alarm(1,20,1,check_wifi)

print("Yofi")
```

tertuis, 2016-01-06:
I've had the same problem since starting with Nodemcu in January 2015. Not sure if it was fixed in the latest version.

TerryE, 2016-01-06:
```Lua
 wifi.sta.connect()
 local ip = wifi.sta.getip()
```
You are banging the SDK with a connect() every 20mSec and issuing a getip() immediately after this.  Not wise.  Please retry with a SDK 1.4 or SDK 1.5 build or #719 applies anyway.

closed 2016-06-29 by marcelstoer