nodemcu/nodemcu-firmware #792: One Wire DS2438 scratchpad reading error

MarcoH2013, 2015-11-26:
Hello community,

I'm testing with latest dev (2015-11-26) with a DS2438 One Wire chip.
http://datasheets.maximintegrated.com/en/ds/DS2438.pdf

Here is my code:
```lua
--------------------------------------------------------------------------------
-- Local used variables
--------------------------------------------------------------------------------
-- DS18B20 dq pin
local pin = 3

local DS2438_TEMPERATURE_CONVERSION_COMMAND = 0x44
local DS2438_VOLTAGE_CONVERSION_COMMAND = 0xb4
local DS2438_WRITE_SCRATCHPAD_COMMAND = 0x4e
local DS2438_COPY_SCRATCHPAD_COMMAND = 0x48
local DS2438_READ_SCRATCHPAD_COMMAND = 0xbe
local DS2438_RECALL_MEMORY_COMMAND = 0xb8
local DS2438_PAGE_0 = 0x00

local DS2438_CHA = 0
local DS2438_CHB = 1

local DS2438_MODE_CHA = 0x01
local DS2438_MODE_CHB = 0x02
local DS2438_MODE_TEMPERATURE = 0x04

local DS2438_TEMPERATURE_DELAY = 10
local DS2438_VOLTAGE_CONVERSION_DELAY = 2

--------------------------------------------------------------------------------
-- Implementation
--------------------------------------------------------------------------------
function gettemp(addr)
    present = ow.reset(pin)
    ow.select(pin, addr)
    ow.write(pin, DS2438_TEMPERATURE_CONVERSION_COMMAND, 0)
    tmr.delay(DS2438_TEMPERATURE_DELAY)

    present = ow.reset(pin)
    ow.select(pin, addr)
    ow.write(pin, DS2438_VOLTAGE_CONVERSION_COMMAND, 0)
    tmr.delay(DS2438_VOLTAGE_CONVERSION_DELAY)

    present = ow.reset(pin)
    ow.select(pin, addr)
    ow.write(pin, DS2438_RECALL_MEMORY_COMMAND, 0)
    ow.write(pin, DS2438_PAGE_0, 0)

    present = ow.reset(pin)
    ow.select(pin, addr)
    ow.write(pin, DS2438_READ_SCRATCHPAD_COMMAND, 0)
    ow.write(pin, DS2438_PAGE_0, 0)

    data = ow.read_bytes(pin, 9)

    crc = ow.crc8(string.sub(data,1,8))

    for i = 1, #data do
        print("byte:"..data:byte(i))
    end

    print("crc:"..data:byte(9))
    print("crc:"..crc)

    temperatur = bit.rshift(bit.bor(bit.lshift(data:byte(3), 8), bit.band(data:byte(2), 0x0ff)), 3) * 0.03125
    print(string.format("%03.0f\n", temperatur))

    present = ow.reset(pin)
end
```

If I read the scratchpad, I'm getting the following bytes back:

Byte | Value
---- | -----
1|11
2|240
3|22
4|151
5|1
6|0
7|0
8|0
9|104

Calculated crc8 is 43 but byte 9 what is the CRC byte says 104.

I tested the same code logic with the same chip on an Arduino:

Byte | Value
---- | -----
1|10
2|224
3|22
4|204
5|1
6|3
7|0
8|0
9|146

Calculated crc8 is 146 what match byte9.

I tested with http://www.datastat.com/sysadminjournal/maximcrc.cgi to calculate the checksum my self. It tells me that there is nothing wrong with the calculation it self. It just looks like the ow.read function does not work correctly.

I have also played with a DS18B20 where reading the scratchpad works fine.

Anyone and idea what is going on?

Kind regards,
Marco

nickandrew, 2015-11-28:
Your 3rd argument to onewire.write should always be 1, as the current firmware will drive the output pin low if it is zero, and you do not want that.

I don't know if that's the cause of your error but it's a start.

nickandrew, 2015-11-28:
Note that in your first example, the hex string for input to the CRC calculator is:

```
0BF0169701000000
```

If you flip the LSB of the first byte, i.e. submit `0AF0169701000000` then the calculated CRC8 is 0x68 which matches your example CRC and backs up my assertion that the ow.write calls are leaving the bus in a bad state - it's corrupting the first bit of data read back.

MarcoH2013, 2015-11-28:
Bingo! That solves it. Thanks Nick
You said that the current firmware works that way. Is it planed to change it our should my code now be fine for the future?

nickandrew, 2015-11-29:
I'll submit a PR someday to fix this. I expect my PR will implement use of an additional pin to control a strong pullup for power, which is why it will take a bit more thought before hacking code.

nickandrew, 2015-11-30:
I think this one can be closed, @TerryE.

closed 2015-11-30 by TerryE