nodemcu/nodemcu-firmware #798: DHT11 temperature integer only

tgreening, 2015-11-30:
I noticed that we are only getting the integer value for the temperature.  Should there be a decimal portion too?  

I am using 0.9.6 float and a 4k7 resistor too.

nickandrew, 2015-11-30:
Change to firmware built from `dev` branch and try again. 0.9.6 isn't supported any more.

tgreening, 2015-11-30:
I am sorry I am using 0.9.6-dev_20150704.

nickandrew, 2015-11-30:
Ok, please upgrade your firmware (e.g. using the build service) and try again.

marcelstoer, 2015-11-30:
> I am sorry I am using 0.9.6-dev_20150704.

Take note of #719 and build the firmware with http://nodemcu-build.com/.

tgreening, 2015-12-01:
I built from the dev branch:
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 33e41a633ec0ad9e2438877515839cf80e1e3dfc
	SSL: false
	modules: node,file,gpio,wifi,net,i2c,tmr,adc,uart,ow,bit,u8g,cjson,dht
 build 	built on: 2015-11-30 14:31
 powered by Lua 5.1.4 on SDK 1.4.0

And still have the the same issue:
status,temp,humi,temp_decimial,humi_decimial = dht.read(1)
   if( status == dht.OK ) then
     print(temp_decimial)
     print("DHT Temperature:"..temp..";".."Humidity:"..humi)
   elseif( status == dht.ERROR_CHECKSUM ) then
     print( "DHT Checksum error." );
   elseif( status == dht.ERROR_TIMEOUT ) then
     print( "DHT Time out." );
   end

temp_decimial is always nil.

nickandrew, 2015-12-01:
@tgreening Thanks.

I looked through the dht module code and it looks okay offhand.

Are your temp and humi values reasonable?

Instead of calling `dht.read(1)`, try calling `dht.readxx(1)` and tell us the results.

nickandrew, 2015-12-01:
Looks like app/dhtlib/dht.c could do with
  * Removal of duplicate code
  * Early check the checksum, and do it once rather than scattered through the code.

nickandrew, 2015-12-01:
Oh, I just checked the DHT11 and DHT22 datasheets. If you have a DHT11, it only has integral temperature resolution anyway.

tgreening, 2015-12-01:
Nick,
That's interesting the datasheet I have said it was floating point precision.  I guess it was a mistake.
Sorry!
Tony

nickandrew, 2015-12-01:
Yeah, DHT11 is 8 bits and 1°, whereas DHT22 is 16 bits and 0.1°

Please close this issue.

closed 2015-12-01 by tgreening