nodemcu/nodemcu-firmware #799: Upgrade to SDK 1.5.0

jmattsson, 2015-11-30:
I'll have a look at doing the upgrade to SDK 1.5.0 over the next few days. Going by the release notes we might have a naming conflict with -lcrypto that will need to be resolved somehow.

TerryE, 2015-11-30:
Johny, if this a good opportunity to mirror the approach taken in the [Open SDK Makefile](pfalcon/esp-open-sdk/blob/master/Makefile) with its `$(VENDOR_SDK_DIR)/.dir` and patch targets?  This way the firmware make would simply fetch and unpack the SDK from the **bbs.espressif.com** site rather than have the files in this repo.

`make clean` would leave the SDK untouched, but we could have a new target `make clean-sdk` which would allow you to clean and refetch the SDK if, say, the patch had changed.

Doing these changes are pretty straightforward but still on my todo list.  How about rolling this approach into your 1.5 rebaseline?  This will get rid of a whole load of denormalised binaries from this repo.

marcelstoer, 2015-11-30:
> the firmware make would simply fetch and unpack the SDK from the bbs.espressif.com site rather than have the files in this repo.

I agree that this would be a sensible approach - in general.

However, the build will become less stable. Before #499 it would fail every now and then just because Travis failed to download `esp-open-sdk.tar.gz` from GitHub. Only if everything we need is in the repository we can guarantee consistency when the build starts (Travis does `git clone` first).

So, based on that experience I'd rather not change that.

jmattsson, 2015-11-30:
Do you have any experience how well Travis handles git submodules? Might that be a middle ground?

marcelstoer, 2015-11-30:
I don't, sorry. We've first been discussing Git submodules in #595. Since you'd essentially tell _Git_ to fetch or "resolve" submodules from either within the `Makefile` or `.travis.yml` (no ideal because that's not used outside Travis) it'd be rather a Git-issue than a Travis-issue. That in turn probably means we can expect more stability.

ghusson2, 2015-11-30:
Hello guys, I am waiting since 2 month the last SDK to be integrated in nodemcu (beacuse I had serious bugs over 0.9.5 one). I was asking if nodemcu was going to fall...
A fresh SDK will boost nodemcu usage as it is very simple to code (ex : a telnet console, in parallel with normal scripts : http://fil.crepp.org/pages/view/51/nodemcu).

I won't be able to help you. But I encourage you really !
Good luck jmattsson and helpers ! This would be a nice Xmas gift, really :)

nickandrew, 2015-11-30:
@ghusson2 The dev branch is already SDK 1.4.0 so you should try it until a 1.5.0 build becomes available.

madhurjain, 2015-12-04:
I've built a project using NodeMCU 0.9.6 SDK which uses WiFi. I am to present this project at the University which has a WPA2-Enterprise WiFi. SDK 1.5.0 adds support for WPA2-Enterprise. Is there a possibility of NodeMCU Firmware with SDK 1.5 being released anytime soon? Or is there a way I can integrate SDK 1.5 with the current dev branch of NodeMCU just to add WPA2-Enterprise support?

TerryE, 2015-12-05:
@madhurjain, sorry but we are going as fast we can.  We don't offer a per-user service.

madhurjain, 2015-12-05:
Thanks for your reply @TerryE I am just trying to understand the complexity of adding SDK 1.5 support to nodemcu and if its something I can do quickly by forking this repo just so I can connect my device to WPA2-Enterprise.

karrots, 2015-12-05:
One thing to be aware of is SDK 1.5 only supports certificate based auth. So uf your using PEAP with MSCHAPv2 for the EAP type on your wireless it still won't work.

madhurjain, 2015-12-05:
@karrots Ahh. I am in fact on MSCHAPv2. Thanks for the clarification.

ghusson2, 2015-12-09:
@nickandrew : ok, I didn't know. I think this is because I didn't found the way to look at the BSP number, and because the README.md was not up to date. Now I see main branch is 1.4, I will test it soon, thank guys :) RQ : if the dev branch is on BSP 1.5, please update the readme in this branch ? An other remark, the wiki and changelog on it was not updated since 2015-03-31. I know those docs won't make the project go forward, but it is important for noobs like me to have an idea of the project (status, community, improvements made and futures ones).

marcelstoer, 2015-12-09:
> An other remark, the wiki and changelog on it was not updated since 2015-03-31

There are issues for that. See https://github.com/nodemcu/nodemcu-firmware/labels/documentation.

> those docs won't make the project go forward

Oh, they definitely will! That's why we did and will put more effort into that.

nickandrew, 2015-12-09:
@marcelstoer I could just delete the changelog from the wiki? I don't think these changelogs have any use - not to new users at least - and more accurate info can be found by perusing the closed pull requests.

It's not much but removing the changelog will not mislead new users into thinking the project is stagnant because it hasn't been updated for 9 months.

marcelstoer, 2015-12-09:
> I could just delete the changelog from the wiki?

See #827.

ghusson2, 2015-12-09:
@marcelstoer, you are right. Precisely, doc don't put forward the code, the product. But it helps to give a good representation of the project. Thus it tends to increase its visibility among others similar project and can attract other users and contributors.
If the changelog is not maintained up-to-date, it should be deleted I agree. As a user, I will look at the issues if I have an issue myself.
And what is important for an end user (in addition to the user manual/tutorial) is a roadmap with a following of what has been implemented (ie : what has been improved since the beginning, and what we are working on). A good exemple I know for that is Proxmox VE (simple and efficient : roadmap/release history) : https://pve.proxmox.com/wiki/Roadmap.

jmattsson, 2015-12-11:
Life & work has been very busy the couple of weeks so I haven't had a chance to work on this yet. Next week should be better.

I'm still torn on what the best approach here really is. I agree with Terry that the SDK itself doesn't belong in this repo, nor does a pre-compiled toolchain for linux/amd64. However, at the same time the current situation means that the build is fully contained and reproducible, locked to the expected SDK revision, and relying on no external items.

Right now I'm considering setting up a nodemcu-build repo containing both the scripts to build the toolchain and patch the SDK, as well as the resulting artifacts, then submodule that repo into nodemcu-firmware. That would have the advantage of getting the SDK & toolchain out of this repo while still keeping the CI build well contained, but more or less just pushes the mess into the new repo instead. I *think* that's still preferable though. It might be that I could avoid getting artifacts into that repo, depending on how well I can instruct the CI build to not clean that part between builds.

eyaleb, 2015-12-11:
I agree. We should have a repo with all the build dependencies, not only for the CI but also to be able to say "can you reproduce this problem with the reference build? Using what settings?".
In the past I had my own VM, toolchain (compiler etc.) and got the SDK from espressif. I had no confidence that I am using the correct collection that the fw was built against (the README was not very specific). I now use my usual workstation with everything off the nodemcu-firmware repo, no guessing.

nodemcu, 2015-12-11:
+1, separate toolchain and SDK  is a good idea I think. there will be needs that use different toolchain and sdk, we can setup a default and make SDK=... for alternative.
btw, just received the esp32 beta board from espressif this morning, since esp32 non-os sdk is not out and no plan heard from espressif,  maybe NodeMCU port to esp32 will go for rtos sdk. fortunately rtos sdk has same api with non-os sdk.

jmattsson, 2015-12-11:
@nodemcu lucky you! :) We're still waiting for our beta board!

nodemcu, 2015-12-11:
@jmattsson I'm in a city nearby:)
since the new chip esp32 is much more capable. we will gonna make a new NodeMCU dev-kit.
I raised a issue [Suggestions about new NodeMCU devkit based on ESP32](https://github.com/nodemcu/nodemcu-devkit-v1.0/issues/12) at repo devkit just now.

jmattsson, 2015-12-16:
Just a note to let people know I've started working on the SDK 1.5.0 upgrade & build/toolchain rejig.

jmattsson, 2015-12-17:
Alright, I'm not having any luck getting esp-open-sdk to build successfully on Travis-CI. The regular instances lack a sufficiently recent autoconf, and the Trusty beta instance conks out during the configure of lx106-hal, when it for some unfathomable reason picks the system gcc while looking for xtensa-lx106-elf-gcc.

jmattsson, 2015-12-18:
Okay, I've finally gotten *something* that seems to work. You can have a look at my branch [sdk150_build](https://github.com/jmattsson/nodemcu-firmware/tree/sdk150_build). It now submodules in another repo, [nodemcu-build](https://github.com/jmattsson/nodemcu-build), which holds the SDK (1.4.0 + 1.5.0 supported) as well as the esp-open-sdk toolchain, in source format. The SDK copy which was in the nodemcu-firmware repo has been dropped.

**The good**:
- nodemcu-firmware no longer contains its own toolchain.
- the toolchain gets built from source, so it should run in more places.
- nodemcu-firmware no longer has its own copy of the SDK.
- Tweaks to the SDK are easily done via the sdk-overrides/ directory in nodemcu-firmware (arguably this could become actual patches to apply during the esp-open-sdk build, but it wasn't obvious how to do that easily, so I stuck with the current approach).
- The nodemcu-build repo has a cache of the necessary source packages, so we're not at the mercy of external sites for source availability.
- SDK1.5.0 was fairly painless to switch to (but I have only done minimal run-testing so far).

**The bad:**
- The toolchain is built from source. This takes a lot of time (~20min).
- Cloning the nodemcu-build repo is slow, because it contains 150+ meg of source archives.
- It's hard to build using anything but the submoduled toolchain.

**The ugly:**
- Travis-CI isn't very happy with building the esp-open-sdk. I ended up fudging the autoconf version to make it all run.
- To avoid having to rebuild the toolchain in CI all the time, I'm using the Travis cache functionality. This was very fidgetty to get to play nicely, and I suspect stale caches might bite us sometime in the future.
- Switching to/from branches that have/don't have this change is a pain, and ofter results in having to wait for a toolchain rebuild.
- While the toolchain & SDK have been split off, we're still using the SDK-specific opensource LWIP bundle from Espressif, which is very much part of our source tree (it's an "application template", not a free-standing module). As a result there's a very tight coupling between the nodemcu-firmware and the nodemcu-build repos.
- The whole thing doesn't feel quite right to me. Not that having a pre-compiled toolchain tar'd up in the repo felt right either, but on the whole I think I preferred that, even though the nodemcu-build approach is technically cleaner.

I'm not going to PR this right away, because I'm really not sure this is the right way forward. I think we should have a thorough discussion about it, and test/bash the living daylights out of this approach before we commit to it. If we'd like to get SDK1.5.0 into NodeMCU sooner than that, I can quite easily do that upgrade on our existing structure.

marcelstoer, 2015-12-18:
> The whole thing doesn't feel quite right to me.

Being the original author of the PR which added the `esp-open-sdk.tar.gz` to our repository I *have to* agree with you :wink:.

I believe you won't find anyone who'd disagree that in general the submodule approach, or something similar, is preferred over hosting our own copy of external binaries. In this particular case though the list of bads and uglies is just way too long.

Of course I'm biased. We're currently at over 120 people who use the cloud build service per day. Anything that increases the build time for them is, well, not ideal. Why should we have to build the same toolchain over and over again?

How about the middle ground? We could have a dedicated repository that hosts the binaries and just submodule those ready-made artifacts.

jmattsson, 2015-12-20:
[Here](https://github.com/jmattsson/nodemcu-firmware/tree/sdk150) is a plain SDK+LWIP upgrade without rejigging any of the build stuff. As before, an **unmodified** SDK lives in the sdk/ directory, with a couple of compatibility fixes on the side over in sdk-overrides.

jmattsson, 2015-12-20:
@marcelstoer I'm very tempted to say that we're already at the middle ground with what we have. Splitting off the toolchain tgz into a submodule'd repo just seems to add complexity for the sake of principle. There's still prebuilt binaries in a repo at the end of the day. The only time I think there would be sufficient gain is if we simultaneously filter-branch'd away any historical trace of that tgz from the nodemcu-firmware repo, to reduce the repo size. That's quite a sledgehammer approach though, and I'm not going to advocate for rewriting history to that extent.

TerryE, 2015-12-20:
> . Splitting off the toolchain tgz into a submodule'd repo just seems to add complexity for the sake of principle

Sorry but I differ on this. It is more that a principle. We've got a whole cluster of elements in our repo that (a) we have no control over and (b) clutter up and bloat it for Lua developers that want to fetch the repo to do local builds. Not the Right Thing to do

jmattsson, 2015-12-20:
Sure, but removing the tgz **does not reduce the size of the repo**. It is always part of the history, and you'll spend the time cloning it. You need to filter-branch the whole repo and eradicate it from the repo history if you want it truly gone. Having had to do one of these at $work a few years ago, I am *not* volunteering to attempt that. Plus of course, the entire repo history becomes less than truthful, and it becomes impossible to actually go back in time and rebuild things as they were at the time.

TerryE, 2015-12-20:
> Sure, but removing the tgz does not reduce the size of the repo.
1.5, 1.5.1, 1.6,...

Yes, we have to live with historic blobs because of the reasons that you detail, but this is too simple a generalisation.  Whilst the git diff algos generate extremely space efficient compression of source-style files, they aren't applied to binary files.  It is extremely difficult to do space and time efficient delta compression of binaries without some model of the contents (e.g. is it zip file, tag.gr, db image, etc. because binary files usually contain internal link which relocate from version to version or use adaptive LZW-style encoding).  So in practice if the SDK is roughly 10 Mbyte then the repository will grow 10 Mbyte for each version added.   I think that we should avoid  this growth.

So my previous comments, both (a) and (b), are still valid.

marcelstoer, 2015-12-20:
> Splitting off the toolchain tgz into a submodule'd repo just seems to add complexity for the sake of principle.

I'm with Terry on this (for the reasons he stated). Had I, or those who supported & merged my initial PR, understood all the consequences and had I anticipated back then how frequently Espressif release new SDKs I'd have proposed submodules right from start. Outsourcing the binaries into a dedicated repository doesn't add a lot of complexity IMO. As long as we stick to standard Git features to achieve that I'd argue it's justifiable.

> As a compromise why not stay with including the SDK zip inside our repo for the 1.5 release and look at sorting it out for the next release...

I don't see why we should postpone that. From what I understand Johny has already done most of what'd be required?

TerryE, 2015-12-20:
The proposed approach has:

> As before, an unmodified SDK lives in the sdk/ directory, with a couple of compatibility fixes on the side over in sdk-overrides

This is far worse than a zip of the SDK in our repo because (a) the zip is smaller than the unpacked SDK, and (b) it is a single configuration item that easily verifiable against the source `file.php` from Espressif's pbpBB3 board, rather than the 145 files and 35 directories added by exploding it.

jmattsson, 2015-12-20:
> Whilst the git diff algos generate extremely space efficient compression of source-style files, they aren't applied to binary files.

Git doesn't use delta compression, you must be thinking of svn or some other vcs. Git stores each version of an object, and does on-the-fly diff'ing. And git does compress binary files (and text files), using zlib (just like a .gz). Google git packfiles for the low-down. That, together with the fact several of the libraries in the SDK might not change between versions (737k of libraries stayed unchanged from 1.4.0 to 1.5.0), might make it a false economy to stash the SDK zip file.

Having said that, I'm not necessarily opposed to that approach, and it would be quite easy to change to. If everyone agrees on at least that much I can do another pass to prepare that. It does seem orthogonal to the toolchain argument.

> (b) clutter up and bloat it for Lua developers that want to fetch the repo to do local builds. 

I still don't understand your reasoning here. Whether submodule or in-tree, it *has* to be fetched in order to build. The only time you'd have an actual saving here is if the CI-toolchain was never part of the tree, and only submoduled. Those not needing the toolchain could then opt not to initialise the submodule. For all the scenarios we're currently discussing, there is no time or space saving, only the added hassle of a submodule.

jmattsson, 2015-12-21:
[Here](https://github.com/jmattsson/nodemcu-firmware/tree/sdk150bin) is another branch which has the 1.5.0 upgrade in its zipped form.

TerryE, 2015-12-21:
Johny, sorry.  Yes, I came from `svn` and you are correct git that each stores each object separately and only uses delta compression optimise network traffic. But a 7% reuse of objects / saving  doesn't merit unbundling, IMO.

Maybe I am just stupid but what is wrong with the open-sdk method of just doing a check of the cache based on the presence and SHA1 of the zip file and on failure doing a `wget --tries=10 --timeout=15 --waitretry=30 --read-timeout=20 --retry-connrefused --continue`or a `curl --retry=10 -C` or the like from the Espressif site on a cache miss, and then we don't need the file in the repo at all.  What am I missing here?

But I'll pull your 1.5 branch down tomorrow and give it a run.  Thanks Johny :)
I do a PR for my configure script (I'll stick to bash) as an optional extra.  I will write it anyway, and I suspect that most developers will find it easier and more convenient.  I certainly will. At the moment I have to stash my user_\*.h files and recheck them out before doing a commit / PR. then unstash them again.  What a pain.  Just having a `$ gen_user` which take a small config file and does this for me is far better in my view.  Having in the tools directory makes its use entirely optional, and allow developers to continue editing header files to define their configuration if that is their want.

TerryE, 2015-12-21:
Seems to be working well.  The only issue that I've found so far is that you've forgotten to update `NODE_VERSION_MINOR` in `app/include/user_version.h`.

TerryE, 2015-12-21:
Sorry I thought that you'd left out updating the ` NODE_VERSION_MINOR`, but that was my cockup. I'd saved the user_\*.h files and check the config and modules before writing them back, forgetting that I was also over-writing the version file.  Leaving the `BUILD_DATE` as unspecfied is probably OK whilst were in dev

TerryE, 2015-12-21:
I was going through the 1.5 Rel notes and saw the caveat about makes must include `-lcrypto` and I remembered that we already had a crypto library in the app, so I checked for duplicates, and there are two copies (name clash) of the following libraries `crypto`, `json`, `lwip` and `upgrade`, one in the sdk and one in the  `crypto`, `json`, `lwip` and `upgrade` directories, respectively.  OK, both versions are pathed in separately (except the SDK upgrade library which isn't used in the firmware), but I am still uncomfortable with the name clash.

* **user**.  Using `readelf` there are duplicate entries in both versions.
* **crypto**. No duplicate symbols
* **json** . The app version generate no code because `JSON_FORMAT` isn't defined, but if it was it would generate many duplicate symbol definitions.
*  **lwip** Again duplicate symbols. so which entry are we using?

Confused from Northamptonshire

jmattsson, 2015-12-21:
I'm not quite awake here anymore, but I'll take a stab at some answers:
- upgrade: not currently used. code in our upgrade/ dir isn't referenced from what I've seen. might be a left-over from an early sdk version?
- crypto: two completely separate libraries, unfortunately now with the same name. we were there first.
- json: no idea, might be a left-over from an early sdk version?
- lwip: we're using the open source version, with the appropriate defines in app/Makefile. the sdk version is not used (and there is no -llwip in the app/Makefile)

I'm not overly keen on having our names crowded out by added sdk libraries (there were two this release, crypto and wpa2). Should we simply repack all the sdk libs into one sdk.a and be done with it? An extra second or two in link time might be worth it to avoid name clashes? (I reserve the right to consider this a Really Dumb Idea(tm) tomorrow when I'm awake btw, just sayin')

creationix, 2015-12-21:
For what it's worth, git does use delta compression in it's pack file format.  This is used for both the network protocols and for internal persistence.  Small repos will start out with every object in it's own deflated file, but eventually some threshold will trigger (or you can run `git gc` manually) and it will be packed into a packfile.

That said, I would highly recommend submodules.  They are very elegant from an internal git perspective, it's just the high-level "chrome" that gets them wrong and makes the UX look bad.

TerryE, 2015-12-21:
@jmattsson Johny, so it's your turn to post when almost dead, eh? :sleepy: 

*   **upgrade** If its a fossil then lets remove it.  As you say it's never really removed from the repo, so if and when we need it, we can resurrect it at that point.
*  **crypto**.  Why not just rename our cypto _directory_ to lcrypto or equiv, We can leave the module as crypto.c, but both modules use the crypto namespace convention so it is only by luck that we don't have symbol clashes.  Renaming ours to `lcrypto_*` will avoid this risk.
*   **json**.  There seems to be 100% alignment of entry points in the `app/json` source and the entries in and the SDK `libjson.a`.  It looks as is the `app` version was an earlier version that zeroday dropped in over a year ago, but it's never been used.  Again my vote is to drop it and clean up the source.  _Note that cjson also doesn't work properly and doesn't have a maintainer._

Also on a general point, IMO we shouldn't do a full extract of the SDK Zipfile, buy only what we need
```Makefile
$(TOP_DIR)/sdk/.extracted:
	mkdir -p "$(dir $@)"
	(cd "$(dir $@)" && unzip $(TOP_DIR)/cache/esp_iot_sdk_v$(SDK_VER)*.zip \
           -d $(SDK_VER)/lib  -d $(SDK_VER)/ld -d $(SDK_VER)/include )
	rm  --f $(SDK_VER)/lib/liblwip.c  
	touch $@
```

TerryE, 2015-12-21:
I've also got a few PRs that I want to do but want to wait until we've merged a working 1.5 into dev.

jmattsson, 2015-12-22:
So based on the earlier discussion, should I do any of:
- not have the .zip checked in, but wget it into the cache dir instead
- have the .zip checked in, but in a "cache repo" and submodule that
- repack the sdk into a single libsdk.a to permanently avoid naming conflicts (I've slept on it, and it still doesn't strike me as a bad idea)

Marcel, you're the one most directly impacted if I do the wget I think, what are your thoughts? Can you keep it cached across builds? In Travis we could simply designate the cache dir for Travis-caching, so it's really the cloud builder that I'm concerned about.

Having a bit of further cleanup and tossing out old crypto/json/whatnot sounds good, regardless.

TerryE, 2015-12-22:
Sorry if I sound like a pissant, Johny.  It's just that this stuff does merit proper review and I seem to be the only one willing to do it :cry:  (1) and (2) are really Marcels call because this will impact the stability and timing of cloud builds.  Maybe we take this off the critical path for getting a 1.5 out.  

As far (3) goes, in reality so long as we don't have duplicates like json and upgrade aggregating libraries probably won't make a material impact to timing.  What slight concerns me more is where we and Espressif have chosen colliding namespaces (e.g. cyrpto_\*)  But again, I looked for collisions this time and didn't find any.

BTW I've done `crypto.fromBase64()` but I will leave off pushing it until 1.5 is merged. Ditto the GPIO fix.

marcelstoer, 2015-12-22:
> Marcel, you're the one most directly impacted if I do the `wget` I think, what are your thoughts? 

They haven't changed much... To me it's, now, pretty clear what is "right" and what is "wrong", it was my PR that eliminated `wget` from `.travis.yml`. I did so because the builds kept failing frequently due to `wget` timeouts/hiccups/whatever between Travis and GitHub where we used to get the pre-built toolchain from.
- When I want to build project X I need to be able to check it out (i.e. clone it) from a SCM and thus get all dependencies I need *unless the project uses proper dependency management* (Maven, Ivy, npm, etc.). It should be self-contained that is. When your project uses Git but doesn't have proper dependency management, think NodeMCU, Git submodules are the obvious answer IMO. They allow for a kind of dependency management within the repository which is very elegant. Let Git worry about consistency.
- The toolchain/SDK should be available as a versioned and pre-built binary.
- Using submodules doesn't change much in terms of Git (repo) space efficiency I believe. When you clone this repository Git still clones the repositories of your submodules and checks out the defined revisions.

> ... so it's really the cloud builder that I'm concerned about.

Thanks for caring but the cloud builder *is* Travis :wink:

Ideally I wouldn't have different build scripts for different branches. That shouldn't bother you too much, though. Should I need to change them I'd appreciate if merging an upcoming PR will be a concerted action with me releasing new build scripts at the same time. I wouldn't want to have dozens of failed `dev` builds just because merging happened when I was absent.

TerryE, 2015-12-22:
Given the fact that I would like to do a few other bits and pieces, is this a good opportunity to merge this into a `dev150` branch on nodemcu? It should only take us a few days to sort this out -- or maybe weeks, given the upcoming holidays. We can freeze PRs to dev for a couple of weeks. That way the current dev cloud build will be OK, and we can move `dev150` over `dev` when we are happy.

TerryE, 2015-12-22:
Re `dev150`: I propose to do this tomorrow based on @jmattsson Johny's `sdk150bin` branch.  However, Marcel, I recommend that we do _not_ include it in the cloud-builder selection list or we'll just get premature issues raised by Lua devs.  We can add any new features there and when we are comfortable switch it over to dev.  Please comment back if anyone has an issue with this suggestion.

jmattsson, 2015-12-23:
@TerryE I added the reduced unzip to the sdk150bin branch.

I've also done a test with the dreaded `wget`, available in [the sdk150dl branch](https://github.com/jmattsson/nodemcu-firmware/tree/sdk150dl). The pros include it not having the SDK zip in the NodeMCU repo, and the cons include it not having the SDK zip in the NodeMCU repo. I did however configure Travis to cache the `cache/` dir, so Travis only has to download it once. It seems to work well. @marcelstoer could I get you to have a play with this branch and see if you would be satisfied with its workings? Personally I'm on the fence between the sdk150bin and the sdk150dl branch, both approaches have merits imo.

As for a dev150 branch, do we really need it? `master` is alive and well and we could snapshot dev onto it before we start with 1.5.0, to keep a recent 1.4.0 around. If we do do a dev150, can I suggest we start piling in the new documentation support in there too?

nickandrew, 2015-12-23:
@TerryE, I don't think a 1.5.0 SDK should be added to the repo at any time, i.e. it should be sorted out whether it's obtained with wget or via git submodule before making a dev150 branch.

@jmattsson Would you be able to rework your sdk150dl branch into two commits, please? One to delete sdk/ and the other to contain all the changes you had to make to get 1.5.0 working; that will be much easier to understand (just read the 2nd commit and ignore the 1st).

TerryE, 2015-12-23:
I am happy with either to be honest.  I've got a small preference for the dl version.  If the wget that unreliable when cached and with `--tries=10`, etc.?  Perhaps the only thing that i'd check would be to put an md5 or even just a size check on the download, just to recover from the odd scenario when we did get a brownout from China.  

One other little comment I notice (a) that you are using gz format for the open-sdk tarball.  I know `xz -9` compression is as slow as hades but the resultant compressed tarball is **14Mb** rather than **34Mb** and decompression time is pretty much the same time as gz decompression.  Also, I take it that this is an Ubuntu 12.04 binary build compatible with the Travis CI service so it makes sense only to include the bits that we need.   For example do we need `xtensa-lx106-elf-gdb`, `cc1plus` or `libstdc++.a`  as these combined are another third of the tarball?  Just a though to keep the repo small for pulling the latest version.

TerryE, 2015-12-26:
@marcelstoer, I've been using the sdk150bin branch and haven't found any glaring issues with it.  I haven't tried the dl variant, but I don't foresee issues.  Given the why "dev150"? quotes, what I am suggesting is to update current master to dev so that master is the last dev140 build (there's only 4 minor commits difference anyway), and then move sdk150bin/dl to dev so that this can unlogjam the 1.5 dependent changes that I've for PRs.

Can you flip a coin and tell me which to use?

TerryE, 2015-12-28:
OK, Master is now at the latest change compared to Dev.  There were a couple of minor commits in Master that I had to back-sync first.  Not sure why. But we can now start to add 1.5 changes to dev.

TerryE, 2015-12-28:
OK, I've issued the PR #885 on Johny's behalf, but given the scope of this change, we should dual key this by having another committer accept the merge or offer an alternative

marcelstoer, 2015-12-29:
Still wading through all the NodeMCU emails that piled up over Christmas...

> Can you flip a coin and tell me which to use?

Looks like I'm too late, sorry. Since Git submodules seem no option to either you or Johny it really feels like flippin' an coin. I have no real preference.
According to #884 you seem to be concerned about Git repository size. Still, you choose the bin branch rather than the dl branch for the PR. I'm sure you have valid reasons. Care to elaborate?

TerryE, 2015-12-29:
> Still, you choose the bin branch rather than the dl branch for the PR ... Care to elaborate?

Swapping from one to the other is a minor tweak, so I "tossed" a coin.  I felt it more important to get one version -- either would do -- out as a test-bed and we can always swap. This was the one that I was testing with;  it involves the least change, and would be the most acceptable -- I felt -- to you and Johny.

To be honest my own vote would be for the `dl` version.

TerryE, 2016-01-01:
See my last comment on #885 :tada:

closed 2016-01-01 by TerryE

jmattsson, 2016-01-05:
Surprised you guys ended up going with the "bin" version instead of the "dl" version. Either way, thanks for doing the PR in my absence :)

TerryE, 2016-01-05:
I tossed a coin. Marcel did not say use DL, but it's truss l trivial

marcelstoer, 2016-01-05:
> (Marcel) I have no real preference.
> (Terry) To be honest my own vote would be for the `dl` version.
> (Johny) Surprised you guys ended up going with the "bin" version instead of the "dl" version.

It would be nothing bug logical then to use `dl` I guess :smile:

TerryE, 2016-01-05:
Let's open a new issue on this, but please leave it a few weeks until we sorted out some other architectural stuff.