nodemcu/nodemcu-firmware #800: RFD - Enable module-specific debug

nickandrew, 2015-11-30:
Debugging firmware errors with `#define DEVELOP_VERSION` results in excessive and unwanted debugging messages through the `NODE_DBG()` macro which is defined as `c_printf` when enabled, and nothing when disabled.

I suggest splitting NODE_DBG into one macro per module, which can be individually enabled in `app/include/user_config.h`. There'd be 30-40 of these macros, so a naming standard is required, and I suggest using DEBUG_MODULE_MODULENAME.

To start, I edited net.c:

```c
#define DEBUG_MODULE_NET
```

and bulk-changed NODE_DBG to DEBUG_MODULE_NET

```diff
--- a/app/modules/net.c
+++ b/app/modules/net.c
@@ -59,7 +59,7 @@ typedef struct lnet_userdata
 
 static void net_server_disconnected(void *arg)    // for tcp server only
 {
-  NODE_DBG("net_server_disconnected is called.\n");
+  DEBUG_MODULE_NET("net_server_disconnected is called.\n");
   struct espconn *pesp_conn = arg;
   if(pesp_conn == NULL)
     return;
```

and that had the desired effect of eliminating just those messages, while leaving other debugging (which I added according to issue #795) intact.

jmattsson, 2015-11-30:
Something like this would be a nice quality of life improvement. I never build with DEVELOP_VERSION precisely because it is far too noisy.

Would we really need to put all these macros into user_config.h though? Might it not be better to have the module debugging fully contained inside each module? My reasoning is that if someone is wanting to debug a module, they're probably building locally anyway, and changing a #define in the .c would be just as easy (if not easier).

nickandrew, 2015-11-30:
The macro definitions could be moved into each module, but on the other hand, having them all in the same .h file promotes naming consistency and makes them easier to find.

Rather than clutter `user_config.h`, I could put them all into `user_debug.h`.

The main argument I can see for not clustering the macros is build dependencies; enable debugging on any macro and every module file will be rebuilt.

pjsg, 2015-11-30:
However, rebuilding the entire image doesn't take that long anyway (from clean). I'd stick them in a single file. Also, would you consider putting the lua_assert stuff in there as well to make it easy to include?

nickandrew, 2015-11-30:
@pjsg Might as well. Re #795, do you have code to print a call traceback or stack dump?

pjsg, 2015-11-30:
#define debug_xstr(s) debug_str(s)
#define debug_str(s) #s
#define lua_assert(e) ((e)? (void)0 :  (output_redirect(__FILE__  ":"  debug_xstr(__LINE__) ": " #e "\n"), (void) 0))

I have a more sophisticated version that make node.assertfails() return the count of assert failures since startup. However, that may be more than what you want....

nickandrew, 2015-11-30:
@pjsg I'm just hoping to find out how you got the exact call sequence from tostring to G(L)->frealloc.

pjsg, 2015-11-30:
I did that by looking at the code. The problem was that the assert fired at the top level. However, knowing which assert fires is a *big* step forwards to finding the issue.

In your case with the mysterious closure issues, I would hope that a funky assert would fire, and this might help narrow the problem down.

marcelstoer, 2015-11-30:
My 2 cents...

> excessive and unwanted debugging messages

Excessive yes but unwanted? I never found myself in a situation of needing debug messages but if I did wouldn't I be interested in more than just the messages from the module I _suspect_ of being fishy? Just because my CoAP code fails doesn't necessarily mean I should be looking at CoAP debug messages in particular - or is it? More questions than answers, sorry.

> I suggest splitting NODE_DBG into one macro per module, which can be individually enabled in `app/include/user_config.h`.

What do you guys suggest I do in the cloud build once a PR like this would land? I currently have a single checkbox with a "warning" label that says:

> turn it on only if you're troubleshooting an issue with NodeMCU developers as it will increase the size of the firmware and decrease its performance

We seem to agree that such a feature isn't for the general public at least :wink:

pjsg, 2015-11-30:
One of the problems with the current debug code is that (as far as I can tell) you can no longer upload code to the nodemcu as there is so much debugging output that the uploader gets confused and thinks that the upload doesn't work.

On a bigger platform, I'd have a 32-bit field, and each bit would control debug output from each of 32 subsystems. This way, output could be disabled/enabled individually. Given that this is just debug code, maybe this is the way to go here as well. The extra overhead of testing a bit is not huge. I'd be inclined to still have (up to 32) different DEBUG macros which implicitly perform the bit test and then call the underlying debug output call. I'd also be inclined to include in the debug strings the file and line number.

nickandrew, 2015-11-30:
@marcelstoer The debug messages appear to be of two main types - the lua_assert which are only triggered when something wrong is detected, and NODE_DBG() which are more logging what the system is doing (although they may be inside a conditional and so appear only in error situations).

I think a developer doing debugging would generally want all the assert checks enabled, but not necessarily a whole bunch of log messages which denote normal system behaviour. So I'm just coming into this from the point of view of making the NODE_DBG() messages optional, and doing it on a per-module basis seems the most rational.

For your build service, I suggest removing the debug checkbox, as the messages are for people debugging the firmware, which can't be done effectively unless those people also have the ability to modify the firmware and build it themselves with patches. If a firmware developer needs logging information from a general user (because a problem can't be reproduced without the right hardware configuration, say) then the developer can build a 1-off firmware and send it to the user.

nickandrew, 2015-11-30:
@pjsg Using a bitfield and testing it at runtime means all potential messages must be compiled-in, whereas they just vanish with a disabled macro. Also I expect we'll exceed 32 categories, and what to do when we run out?

Adding file and line number is a good idea, I think it could be done in macros.

pjsg, 2015-11-30:
I would definitely want to leave the debug checkbox present and have the lua_assert messages all enabled. In particular, I have only seen the hardware fault messages with the debug builds that I have done. This seems useful -- in fact so useful, I would want them enabled under all circumstances (unless the code space required is prohibitive). I haven't looked to see where those are generated.

Also, it would be good to fix the various uart speed changes that happen. It appears that the early boot messages come out at 115000 and then (in develop mode) it drops to a lower speed. For non-debug builds the speed is 9600 after some point. [As a side note, the uploader that I use cannot upload at the debug speed (78k??) so I have to switch it to 9600 before doing an upload. If we want to encourage debug builds, then having the speed set to 9600 would be a good start. If (by default) all the NODE_DBG calls are disabled, then there will be no(?) additional output and hence the slow speed won't matter.]

pjsg, 2015-11-30:
@nickandrew Currently, if you do a debug build, you get all the debug messages compiled in. This scheme is only slightly worse (as each message would get the FILE and LINE number prefixed). [The lua_assert macro above does include the preprocessor trickery to get the file an dline number prefixed onto each message]

I guess it depends on what the purpose of the debug build is. My take is that they serve two purposes:

1: A user has a problem with the regular code. A nodemcu dev tells them to build a debug version (using the online builder) and try running their code again. Maybe enable a few categories of debug message -- but that can be done once the user is running the debug firmware. Yes, this may not work if the user has a 512K esp8266. 

2: A dev wants to debug something very particular and is adding extra debug messages/asserts. In this case, they are building their own images. They can disable whatever they like -- they are probably hacking header files anyway.

What is important is that the debug messages should not burn up precious RAM!

nickandrew, 2015-11-30:
@pjsg I think you and I are only seeing the hardware fault messages now because DEVELOP_VERSION sets bitrate to 74880 and that's the same speed the hardware faults are emitted.

Funnily, even though the bitrate is 74880, I see the messages at 38400. I have no idea why, except that 74880 is roughly twice 38400.

nickandrew, 2015-11-30:
This explains why 74880, the non-standard baud rate in particular: https://github.com/arduino/Arduino/pull/3389#issuecomment-115194096

> The 74880 baudrate is directly related to the quartz used on ESP8266, if you use a 40MHz clock the standard baud is 115200, but on all ESP8266 that we have on ebay and other manufacter is used a clock of 26MHz and automatically the baudrate become 74880

I still have no idea why I can see the output at 38400.

jmattsson, 2015-11-30:
> In particular, I have only seen the hardware fault messages with the debug builds that I have done. This seems useful -- in fact so useful, I would want them enabled under all circumstances (unless the code space required is prohibitive). I haven't looked to see where those are generated.

Those come from the SDK. One of the 1.x releases introduced them. They registed a handler for the hardware exceptions, which gives them a chance to dump the details to console. I can't remember whether they print right then and there, or whether they save into RTC memory and print on reboot. Either way, these outputs are always enabled and in no way correlate to other NodeMCU debugging options. So, if you're only seeing these with lua_asserts() enabled, then odds are it's something in the assert path that is crashing out.

nickandrew, 2015-11-30:
@jmattsson Can we override that handler, and dump our own at the current bitrate?

TerryE, 2015-11-30:
> I would definitely want to leave the debug checkbox present and have the lua_assert messages all enabled.

Mine is -1 to that one. My rationale is that the cloud service is a quick start for end Lua application developers.  Anyone serious enough to want to do development work on the Lua core or even a library extension needs to be able to build the firmware locally.  It's not rocket science.  Marcel has described you to do this with application containers, and I've given a step-by-step on how to build your own Ubuntu VM for this ([here](http://www.esp8266.com/wiki/doku.php?id=toolchain#how_to_setup_a_vm_to_host_your_toolchain)).

When I was developing my LCD patch, I followed Bogdan Marinescu's example in his development of the LTR patch, and hammered my code with the Lua test suite.  I was really impressed by this and amazed at the edge conditions that this threw up.  Since then, I've thought that some form of stripped down harness would be invaluable for testing the firmware and libraries.  However, this has t be balanced by the code space hit in flask that pretty makes the type of development testing totally impractical on the 512Kb modules.

My experience of developing one this type of embedded system is that blanket debug statements or even runtime bit masking isn't worth it unless you have a separate fast serial debugging channel and even then you have end up changing realtime performance / behaviour by debug o/p flow control.

IMO better to use the assert approach:

*    An assert identifies a bug
*    Binary chop the test code to produce the smallest reproducible test case.
*    Use it plus enabled debug code to binary chop into the error.

pjsg, 2015-12-01:
Do you think that the asserts should always be enabled or only enabled upon request (at build time)? I would like people to be able to get builds with asserts enabled so that if they have weird behavior, it is quick test for them to be able to figure out if there is some internal problem in the firmware, or whether it is their code (more likely this). i would rather not have them build the firmware just so that they can enable asserts. [I'm not even sure that you *can* build the firmware if you are only running windows -- without a lot of pain involving virtual machines and linux images]

The other question is what should happen when an assert fires? The options are either to continue as if nothing was wrong, or reboot. The problem with the latter is that you may lose the error message fairly quickly!

TerryE, 2015-12-01:
@jmattsson said:

> My reasoning is that if someone is wanting to debug a module, they're probably building locally anyway, and changing a #define in the .c would be just as easy (if not easier).

+1

nickandrew, 2015-12-01:
@TerryE I could put a note in user_debug.h telling developers the "standard" and to go look in the module files (and edit them) to enable individual module debugging.

In this case, user_debug.h would contain the macros for lua_assert (as @pjsg commented above) and any debug-related `#define` from user_config.h

nodemcu, 2015-12-01:
DEVELOP_VERSION is a global debug switch when there are not as many modules as now are. when modules become many, sometime I just roughly add something like #define NODE_DBG c_printf within the module .c file, so I think it's a good idea that put a local macro switch at top of the module .c file.
but put these macros all together in a file maybe not, as a module developer they should know how to find the module file(.c .h) anyway.
Enable a bit mask debug option in a runtime will eat everything I think, and it definitely impact the behavior,  maybe even harder to find the problem.
some "debug" information like indicate that the lua syntax error is useful for lua developer.
someday maybe IDE with virtual-nodemcu-node can do a lot pre-debug for this.

74880 bitrate is same as the chip output at boot stage. so can see every char under this bitrate.
@nickandrew can you see the right char output under 38400? maybe get a oscilloscope to help debug?

btw: FULL_VERSION_FOR_USER macro is of-no-use. it's just something I forgot to remove...

nickandrew, 2015-12-01:
@nodemcu Right now I see almost everything (boot reason, flash loading, nodemcu debug messages, console output, hardware exception) at 38400 without changing speeds ... if there's anything I'm not seeing, then it's not important.

I've had lots of interaction with the device where it will reboot and the bitrate will change and spew tons of gibberish into my terminal window, and I think it would be good for regular users if the different types  of message can come out at a consistent bitrate.

Also that's a good idea to check it out in more detail; I can put my logic analyser onto TX sometime when I get a chance. It's not high priority for me though.

jmattsson, 2015-12-01:
Oh how I wish for a consistent bitrate...

Unfortunately, the boot ROM doesn't set a particular bit rate on its own, it inherits what was already there. This means that for a cold boot it's entirely crystal dependent, but for a warm boot it uses the "left-over" baud rate of whatever was last used, based on the APB clock IIRC. Then during the SDK start-up, it does the PLL adjustment to go from 52MHz to 80MHz, and around that window some of the messages can come out at 115200, before the SDK then readjusts the baud rate back to 74880. After that, NodeMCU finally gets a say, and we get to 9600 or whatever has been configured in user_config.h

TL;DR: no-can-do consistent baud rate :(

marcelstoer, 2016-02-22:
No broad consensus here, is there?

I definitely wouldn't want to loose the ability to turn on "debug statements" globally with a single switch. On the other hand enabling per-module debug makes sense. So, why not combine the two? Keep the current global flag but include it in the per-module macro so that `DEBUG_MODULE_NET` would print if `DEBUG_MODULE_NET` or `DEVELOP_VERSION` are true.

TerryE, 2016-02-22:
@marcelstoer, this is really to heavy IMO.  When I want to have debug in a single module, I am working on it pretty heavily so I just temporarily define NODE_DBG within the module the same as Zeroday does.

It's just not worth adding another layer of macro magic for this.

marcelstoer, 2016-02-22:
Fair enough Terry, does this mean we can close this?

TerryE, 2016-02-22:
my vote would be :+1: so let's close this and if anyone strongly objects they can reopen it.

closed 2016-02-22 by TerryE