nodemcu/nodemcu-firmware #801: Make it easy to enable asserts

pjsg, 2015-11-30:
It seems clear that nobody is running with lua_assert enabled (or, at least, they don't bother to fix the assertion failures). For example, the following test case triggers assertion failures in lvm.c reliably:

```
function badfont()
   local x = u8g.ssd1306_128x64_i2c(0x40)
   
   x:setFont(0)
end

tmr.alarm(0, 100, 0, badfont)
```

with this error:
```
PANIC: unprotected error in call to Lua API (stdin:3: bad argument #1 to 'setFont' (font data expected))
lvm.c:470: base <= L->top && L->top <= L->stack + L->stacksize
Fatal exception 29(StoreProhibitedCause):
epc1=0x4024adc5, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000004, depc=0x00000000
```

Now, this should generate a PANIC as I did pass in a bad argument to setFont, but it isn't actually clear what should happen next. However, I would argue that an assertion error followed by a hardware trap is incorrect.

I am willing to produce a PR to make it easy to enable assertion checking if people think that that is interesting.

You might also argue that the problem here is the hardware exception, and I would agree, and I'll try and fix this as well (once I understand what _ought_ to happen as a result of a PANIC)

pjsg, 2015-11-30:
It looks as though @nickandrew is going to add the enablement of asserts.... I'll probably use another ticket to fix the PANIC issues.

jmattsson, 2015-12-01:
Looking at that `29(StoreProhibitedCause)` and `excvaddr=0x00000004` I'd say something in the assert code tried to write through a null-pointer.

nickandrew, 2015-12-01:
> it isn't actually clear what should happen next.

'Programming in Lua' section 8.4 (which I just read for the first time) indicates that a panic is appropriate, because the code is not being executed under pcall(). What I'd *like* to see here, as well as the panic, is a stack traceback.

I'm guessing the firmware is soldiering on regardless after the panic, and that's why the null pointer exception happens.

pjsg, 2015-12-01:
The panic is appropriate and the lua manual indicates that the app should exit. I don't know what that means in an embedded environment. It could cleanly restart the firmware (reboot), or it could just abort that particular call stack -- essentially leaving the app running.

I think that the requirements for embedded systems are different to that of a standalone program. In a system which mixes C and has a bit of lua where there is significant work done by the C code (e.g. a firewall that supports LUA for customization), it would be very unfriendly to reboot the firewall because somebody has a typo in their lua code. 

I also agree that a stack trace would be nice as well.

pjsg, 2015-12-01:
@jmattsson I think that the error actually is as a result of code downstream from the assert failure. Currently, the implemented lua_assert code just prints out the assert failure message and lets execution continue. An alternative of making it reboot the board would make the firmware unusable until we deal with the most common assertion failures.

I suspect that the reason that @nickandrew and I are seeing the hardware fault messages is that our uarts are set to the funky 77k baud rate -- though there are some extreme oddities there. I noticed before, when I was running stock firmware, that crashes caused a random strings of garbage to be output (when receiving at 9600).

nickandrew, 2015-12-01:
@pjsg wrote:

> it would be very unfriendly to reboot the firewall because somebody has a typo in their lua code.

I see 2 use cases:
  1. Device is an IoT thing; it's embedded in something and is inaccessible to the user. Some unlikely event has occurred, but the device has to *keep running* and the correct way to do that is to reboot and proceed from a known state. Big bonus if the programmer can define an exception handling function which could log or report the error. Otherwise, it should go to console.
  2. There's an end user talking to it over a console, playing with lua commands, and in this case the device has to print an error and return to the command prompt.

I assume it's possible to wrap an exception handler around the code which runs init.lua. I had a quick look and I found where it is run in the firmware, but didn't understand it in detail.

pjsg, 2015-12-01:
Since many of the errors that cause panics are out-of-memory errors, it probably is not viable to run any specific code. However, I could relate to having a boolean option -- e.g. node.rebootonpanic(true/false)

The default probably should be true as that is the safest -- but I'm open to arguments for and against.

However, the code is really gnarly. The original call to c_exit() in ldo.c was just commented out (as compared with lua-5.1.5). Before this change, luaD_throw never returned. Callers of this function don't expect it to return! Rebooting the platform under deliberate control seems to be the only option. 

Another option is to have a version of lua_call that establishes an error handling function and can deal with errors. Then go through the code base and change lua_call into lua_call_handle_error (or whatever). 

Thoughts?

jmattsson, 2015-12-01:
Ah, I thought I'd pushed this back upstream but I see now that I haven't. We're running with an added 'while(1){}' at the end of panic() in lauxlib.c, to let the watchdog recover for us. This both gives us a chance to see the panic message during development/test, and have it do something sensible in production - all while running the same code path.

pjsg, 2015-12-01:
Excellent -- thanks (in advance) for fixing that.

nickandrew, 2015-12-01:
@jmattsson Ah … do you know how long the watchdog takes to fire? I could suggest resetting the watchdog just before your while loop, to make a consistent time delay before reboot.

Also is there any way for lua code to find the boot reason (watchdog or power up (or CH_PD?))? That would be handy if an IoT thing could actually report upstream that it had experienced a fatal error. I already have mine do POST to /reboot but the node doesn't know why it rebooted.

jmattsson, 2015-12-01:
From memory it's ~4-5 seconds once it's been programmed by the SDK (during the first part of the boot it's about 11sec).

Try node.bootreason() to read out the low-level boot reason from `rtc_get_reset_reason()`. I don't think we have exposed the new `system_get_rst_info()` yet.

nickandrew, 2015-12-01:
@jmattsson Goodo :-)

pjsg, 2016-03-10:
This appears to be resolved

closed 2016-03-10 by pjsg