nodemcu/nodemcu-firmware #802: Spontaneous reset doing multi-device onewire temp reporting

nickandrew, 2015-12-01:
I'm branching my issue off from #795 as it is not identical, but maybe similar.

My hardware setup is an ESP-01 with 0, 1, 2 or more DS18B20 sensors attached to GPIO0, a serial console monitor, and WiFi. Power supply is regulated and a couple of smoothing capacitors are attached.

Firmware is latest `dev` branch, commit 33e41a6 with or without DEVELOP_VERSION, with or without the concat patch by @pjsg in #797. In other words, it fails in all these situations.

My lua code on top reads onewire temp sensors and POSTs the temperatures. If there's 0 or 1 sensor attached it works fine; two or more sensors will all be reported the first time through the loop, then the device resets.

The reset seen can be a fatal exception 28 in reallymarkobject:

```
Fatal exception 28(LoadProhibitedCause):
epc1=0x402566d5, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00004655, depc=0x00000000
```

or a lua PANIC:

```
PANIC: unprotected error in call to Lua API (tn-ds18b20.lua:40: bad argument #1 to 'reset_search' (number expected, got nil))
PANIC: unprotected error in call to Lua API (ow 0 does not exist)
```

It's not a problem with 2 or more sensors per se - if I disable wifi then the problem goes away; if I stop POSTing the results then the problem goes away; if I replace the sensor read functions with functions which return dummy results, the problem goes away. The problem only manifests when all the functionality is in place, and so I have had no success isolating it to a code fragment I can post here.

The problem also goes away if I send lua commands to the device through the serial interface after reporting the first set of results. If I send 1 command then the error may change (indeed it doesn't always fail in exactly the same way). If I send 2 commands the error may not occur, and it can report temperatures for hours on end.

The lua PANIC is around this fragment of code:

```lua
function conv2()
    local pin = 3
    startConversion(pin)
    tmr.alarm(0, 1000, 0, function() conv1(pin) end)
end
```

The panic indicates that conv1(nil) is being called, which means that the value of pin has been corrupted in that 1-second interval. If I hardcode 3 in both function calls, of course the problem goes away.

Enabling debug and lua assertions, I see this 18 times before the panic:

```
lgc.c:85: !iscollectable(uv->v) || (ttype(uv->v) == (uv->v)->value.gc->gch.tt)
lgc.c:531: !iscollectable(uv->v) || (ttype(uv->v) == (uv->v)->value.gc->gch.tt)
```

This is also in garbage collection, but I'm pretty much at a loss about how to debug it further. I don't know how lua is getting there.

pjsg, 2015-12-01:
This is a problem in the GC when dealing with upvalues. upvalues are (AFAIK) references to variables in an enclosing frame. This is exactly the problem that you are facing.

Will your code work in my environment? I want to run some tests....

nickandrew, 2015-12-01:
If you have 2 or more DS18B20 temp sensors, then it will work.

UncleRedz, 2015-12-04:
Could you provide the code for your sensor sample "loop"? I had an issue which behaved very much the same, when using multiple sensors and posting over mqtt. The issue was that reading all sensors and posting in one timer callback took too long time, which caused Wifi / network to behave bad. If I split it up so that only one sensor and post was processed each time the timer callback triggered, the problem went away. This allowed the ESP to do its house keeping in time between my callbacks.

nickandrew, 2015-12-05:
This is called from a timer every 10 seconds:

```lua
function conv1(pin)
    readTemperatures(pin, function(a,t)
        print(string.format("A: %s T: %s", a, t))
        postTemp(a, t)
    end)
end

function conv2()
    local pin = 3 -- gpio0
    startConversion(pin)
    print("Pin value first " .. pin)
    tmr.alarm(0, 1000, 0, function() conv1(pin) end)
end
```

I currently have modified ow.c to print the contents of the lua stack on each call.

The 2nd print statement `print("Pin value first " .. pin)` makes the crash *go away* however it does not always print "Pin value first 3" as expected. Sometimes it prints "Pin value first" and no value, but the reporting works nevertheless!

```
nick2: ow_reset Number read was 3
LUA Stack trace from 1 to 1
Type of index 1 is 5 (number)
  3
lobject.c:88: (((t1)->tt) == 4)
lobject.c:88: (((t2)->tt) == 4)
lobject.c:88: (((t1)->tt) == 4)
lobject.c:88: (((t2)->tt) == 4)
lobject.c:88: (((t1)->tt) == 4)
lobject.c:88: (((t2)->tt) == 4)
Pin value first 
nick2: ow_reset_search Number read was 3
LUA Stack trace from 1 to 1
Type of index 1 is 5 (number)
  3
nick2: ow_reset Number read was 3
LUA Stack trace from 1 to 1
Type of index 1 is 5 (number)
  3
A: 28.000002f4fdfc6b T: 22.8125
```

Huh?

Also my additions to ow.c check the type of the stack value before returning back to the original code:

```c
static int ow_reset_search( lua_State *L )
{
  nick2(L, "ow_reset_search");
  unsigned id = luaL_checkinteger( L, 1 );
```

Function nick2() does this:

```c
void nick2(lua_State *L, char *msg) {
  if (!lua_isnumber(L, 1)) {
    unsigned int i = 0;

    NODE_DBG("nick2: %s It is not an integer\n", msg);
    nick_luastack(L);
  } else {
    unsigned int number = (unsigned int) lua_tonumber(L, -1);
    NODE_DBG("nick2: %s Number read was %d\n", msg, number);
    nick_luastack(L);
  }
}
```

and the stack printing function is this:

```c
void nick_luastack(lua_State *L) {
  int i;
  int top = lua_gettop(L);
  c_printf("LUA Stack trace from 1 to %d\n", top);

  for (i = 1; i <= top; i++) {
    int t = lua_type(L, i);
    c_printf("Type of index %d is %d (%s)\n", i, t, lua_typename(L, t));
    switch(t) {
      case LUA_TSTRING:
        c_printf("  '%s'\n", lua_tostring(L, i));
        break;
      case LUA_TBOOLEAN:
        c_printf("  %s\n", lua_toboolean(L, i) ? "true" : "false");
        break;
      case LUA_TNUMBER:
        c_printf("  %g\n", lua_tonumber(L, i));
        break;
      default:
        c_printf("  Something else.\n");
        break;
    }
  }
}
```

And the console output just before a PANIC is:

```
lobject.c:88: (((t1)->tt) == 4)
lobject.c:88: (((t2)->tt) == 4)
nick2: ow_reset_search It is not an integer
LUA Stack trace from 1 to 1
Type of index 1 is 0 (nil)
  Something else.
lgc.c:85: !iscollectable(uv->v) || (ttype(uv->v) == (uv->v)->value.gc->gch.tt)
lgc.c:531: !iscollectable(uv->v) || (ttype(uv->v) == (uv->v)->value.gc->gch.tt)
```

So the "iscollectable" assertions fire *after* the ow_reset_search() function is called, and while it is checking the type of the argument. My tests run after lots of "lobject.c:88" assertions, which do not seem to harm anything, but they show that at the point ow_reset_search() is called, the stack is somehow invalid.

pjsg, 2015-12-06:
A possibility is around asynchronous callbacks. Try adding the following code to user_config.h
```
char lua_locked;
#define lua_lock(L)   (lua_assert(lua_locked == 0), lua_locked++)
#define lua_unlock(L)   (lua_assert(lua_locked > 0), lua_locked--)
```
The default implementation of lua_lock is nothing. The standard code assumes that we are running in a single threaded environment. However, I'm a little vague as to how timers and interrupts are handled. It seems possible that these might interrupt the existing code. Adding the code above will trigger an assert if ever the code violates this assumption.

This code does ignore the state pointer. I think that the nodemcu only has a single state object, so this shouldn't matter. If there are multiple states, then the lua_locked vairable should be moved into the state and the code adjusted accordingly.

I'm on the road, so I can't test this -- but it does build!

nickandrew, 2015-12-06:
Interesting ... it causes resets; we don't get as far as a PANIC.

```
NodeMCU 1.4.9+debug build 1.4.0-20151201-30-g8e1ccb4 powered by Lua 5.1.4 on SDK 1.4.0
lapi.c:528: lua_locked == 0
lapi.c:566: lua_locked == 0
lapi.c:164: lua_locked == 0
```

This set of 3 assertions (with the same line numbers) can repeat up to 3 times before the device resets.

pjsg, 2015-12-07:
That isn't good (for system reliability, but good for diagnosing your problem). This indicates that the LVM is being called recursively (I think) and then all bets are off. Getting a complete stack traceback for the first assert should show the problem -- I think that there was another patch provided in one of these threads that did a stack traceback. 

It *could* be that there are multiple Lua_state objects, but I don't think that that is the case. 

@TerryE -- you seem to have a lot more experience with this stuff. My feeling is that if the lua_lock assert ever fires, then either something is severely wrong, or someone is about to violate an assumption. I think that you can *only* invoke lua_call when the LVM is not running (lua_call calls luaD_call which calls luaV_execute. If that is already running, then I think that we are doomed).

Not clear what the solution is -- probably need to have a list of pending calls, and then run them at appropriate times.

By the way, what guarantees does LUA make about when callbacks can run? I.e. is the following defined?

```
function test() 
  local value = 0
  tmr.alarm(0, 100, 0, function() value = value - 1 end)
  for i = 1, 100000 do
     value = value + 1
  end
  print(value)
end
```
Is the value printed guaranteed to be 100000, or could the timer have fired and the callback run during the execution of the loop? I suspect that the answer is that it is guaranteed to be 100000 by the spec (though I didn't find it with a quick google search).

TerryE, 2015-12-07:
I think that you are on false track here. The Lua lock mechanism has a specific context.. Remember that the Lua engine is designed to be embedded in other applications, and is often used as the scripting engine in games. Such applications can be multi-threaded, and different threads might each have their own Lua engine instance. The lock mechanism prevents reentrancy issues in the usecase. The Lua environment on the ESP is definitely single instance, so this should be replaced by `((void) 0)` to get optimised away. It serves no purpose on the ESP. I also came across some counting issues in the compiler when testing my packed lineinfo patch where it was being botched, so I am not to air hood solid it is anyway.

My Unofficial FAQ is desperately in need of a complete rework, but my description of how it interacts with the SDK is still a fair summary. The SDK is single threaded. It calls C tasks sequentially, and each runs to completion. Each task should complete in less than 10 msec, or the WiFi stack can fall over. Lua just sits within this framework, so each C library routine is a task and will typically call the corresponding Lua callback.

There are true interrupt driven drivers, but these interact with the SDK, and messing with these is not for the faint hearted. GPOI uses one, but I can't remember if OW does. I'll check tomorrow on my laptop. (I am on my tablet now).

nickandrew, 2015-12-07:
Are timer tasks called directly by the SDK?

nickandrew, 2015-12-07:
Also, in the for loop mentioned by Philip, does the SDK regain control at
any point before the loop has finished? I'm just trying to wrap my head
around how to ensure that lua tasks complete in under 10 msec.

TerryE, 2015-12-07:
Read the FAQ. In Philip's example, function test runs to completion, and then returns control to the SDK. (Well it won't in this case because the WDT will fire and reboot the CPU.) The SDK will then invoke the next runnable task which might be the timer library callback. This will then execute the registered Lua function which increments the upvalue **value**. Read the FAQ.

pjsg, 2015-12-07:
@TerryE Your message describes what I thought ought to happen -- i.e. the lua code always runs to completion. 

The fact that the assert in the lua_lock fires indicates that (in @nickandrew  case) this is not what is happening. In his case, the callback is firing while the main interpreter is running. I am not suggesting that the lua_lock function needs to be implemented. 

I don't know where your 10ms number comes from. The uart module can fire an event much more often than that. At 9600 bps, it could fire an event every millisecond.

TerryE, 2015-12-07:
Philip,

> The fact that the assert in the lua_lock fires indicates that (in @nickandrew case) this is not what is happening.

This inference is flawed.  As I tried to explain, I first developed and tested my packed lineinfo patch on standard Lua run on my dev laptop (natively in Ubuntu), and this does implement the lock by default.  I found that some API calls issue a lock /unlock sequence even though they can be called from within locked code so the locked code was doing unlock, call API, lock which is actually nonsense because this introduces a couple of unlock//lock points in what is supposed to be atomic code.  I gave up and just went for the `((void) 0)` rather than adding such crap to my patch.  The nodeMCU libraries have been developed with locks disabled so _may_ just be doing this sort of double lock.  I know one of my algos does.

  > I don't know where your 10ms number comes from. 

See the SDK 1.3 release notes optimization point 4:

> Optimize software timer to solve the a connecting problem.
> _Please do not call “os_delay_us” or “while” or “for” to occupy CPU more than 10 ms in timer callback._

and current SDK Overview (I'lI leave you to translate the "Chinese English"):
> *    Using non-OS SDK which is single-threaded, the CPU should not take long to execute tasks:
>     *     If a task occupies the CPU too long, ESP8266 can't feed the dog, it will cause a watchdog
reset;
>    *     If interrupt is disabled, CPU can only be occupied in us range and the time should not be more than 10 us; if interrupt is not disabled, it is suggested that CPU should not be occupied more than 500 ms.
>*     We suggest using a timer to check periodically, if users need to call `os_delay_us` or function `while`, or function `for` in timer callback, please do not occupy CPU more than 15 ms.

The SDK uses the same callback mechanism for time-crtitical Wifi and TCP functions.  If aapplication callbacks don't honour this guideline then the wifi or TCP stacks can fail and then I am answering Qs about "why does my netwroking fail?" instead.  

My idea for writing a **_Frequently_ Asked Questions** is that that I get tired of answering the same Q over and over again. :wink:

pjsg, 2015-12-13:
I now understand the lock message from @nickandrew run.
```
lapi.c:528: lua_locked == 0
lapi.c:566: lua_locked == 0
lapi.c:164: lua_locked == 0
```
These are as a result of the compile_stripdebug code in lparser.c -- this is code that is not present in lua 5.1.5.

The parser code is (AFAIK) called under the lua_lock and hence it mustn't call any of the external APIs that take out and release the lock as part of their implementation. Now, the nodemcu code does not actually implement the lock, so this is (I think) a harmless error. I did find problems in the gpio module (which I filed as another issue), but @nickandrew 's code doesn't use that module, so that isn't his problem.

TerryE, 2015-12-13:
Philip, I could have done what some of the other routines did to botch the lock
```
// Start of atomic code
lua_lock();
// ...
lua_unlock();                                           /* unsafe window follows */
API-call_which takes out a lua_lock();    /* unsafe window follows */
lua_lock();
// ...
lua_unlock();
```
But look on the bright side -- your compiled source files are only slight larger than the **lc** versions. :smiley:

pjsg, 2015-12-13:
Maybe I'll hack up that function to do that. Not really the safe way of doing things -- but given that I just want to see if/where any lua_lock violations are happening, maybe it won't matter. It will solve the false positives that clutter up the output at present. Thanks for the suggestion.

nickandrew, 2015-12-16:
I tried @TerryE's suggestion in #851 of disabling `#define GPIO_INTERRUPT_ENABLE` to see what happens.

But first, a recap. The problem I am trying to debug occurs in this bit of code:

```lua
function conv2()
	local pin = 3 -- gpio0
	startConversion(pin)
	tmr.alarm(0, 1000, 0, function() conv1(pin) end)
end
```

The second time it runs, conv1() is called with a nil or invalid argument. I'm detecting this through additional checking in `app/modules/ow.c` and subsequent lua asserts fire:

```
lgc.c:85: !iscollectable(uv->v) || (ttype(uv->v) == (uv->v)->value.gc->gch.tt)
lgc.c:531: !iscollectable(uv->v) || (ttype(uv->v) == (uv->v)->value.gc->gch.tt)
```

leading at last to a PANIC:
```
PANIC: unprotected error in call to Lua API (tn-ds18b20.lua:40: bad argument #1 to 'reset_search' (number expected, got nil))
```

This error (the whole thing) is fragile. Changing bits of debugging code (which should not affect the outcome) can make it disappear, or morph. Adding a memory dump (which outputs to the UART) into ow.c made the error disappear this morning. So, unfortunately, making the error *go away* is not proof that the underlying cause has been solved.

Now onto the results of my test. After disabling GPIO_INTERRUPT_ENABLE, the error remains and is unchanged. I agree with Terry's conclusion:

> we've effectively ruled out that it is an interrupt related issue.

at least as far as GPIO interrupts are concerned. Code which isn't being compiled can't be blamed for a runtime error.

One way to make the problem go away is to do additional debugging which outputs to the UART.

Another way is to send Lua commands to the console after the first time through the main loop. Just a few repetitions of `=gpio.read(2)` are enough to disrupt this fragile error and everything works as it should, permanently.

jmattsson, 2015-12-16:
Input on the console "fixing" errors is something I've seen mentioned in another issue here. Hmm.

TerryE, 2015-12-16:
As I mentioned in #851, Johny's implementation of console input has an ISR passing control back to the Lua world by issuing a task post -- that is the way that you are supposed to do it. As does the handling of console input. I would suugest that another way of phrasing Nick's comment:

> One way to make the problem go away is to do additional debugging which outputs to the UART. Another way is to send Lua commands to the console after the first time through the main loop. 

is: one at of making the problem go away is to break the processing thread by doing atask post to reschedule.

TerryE, 2019-06-18:
The rules for calling to the Lua world from ISRs are noew clear: _don't_; use task posting.  We have fixed the GPIO handing post this thread, so I am closing this as definitely stale.

closed 2019-06-18 by TerryE