nodemcu/nodemcu-firmware #803: node.input() should this work?

lorneb, 2015-12-01:
Should this code execute and assign the value of bar into the variable baz?

```
foo = "bar"
baz = "bazza"
cmd = "baz = foo"
node.input(cmd)  
print("this is foo "..foo)
print("this is baz "..baz)
```

Would expect output to be:
this is foo bar
this is baz bar

It doesn't is there something I'm missing? TIA

nickandrew, 2015-12-01:
node.input() doesn't seem to be working at all for me. Hang on, the API docs say "never use node.input() in console, no effect" and I'm doing it in the console.

lorneb, 2015-12-01:
Ahh ok, I was trying it saved to a file and "dofile" ing it.
The longer term plan was to use the method in a mqtt subscribe event handler so it should work in there

What about node.output( func, debug)  does that only work while not in the console too?

nickandrew, 2015-12-01:
Try it; you might break your device until next reboot.

'dofile'ing from the console might have the same problem as pasting it into the console, as I did. Try your dofile from init.lua.

lorneb, 2015-12-01:
Even doing this in an empty init.lua doesn't work so I think it must have to be used in a socket call back or similar.

tried

```
function launch()
  foo = "bar"
  baz = "bazza"
  cmd = "baz = foo"
  node.input(cmd)  
  print("this is foo "..foo)
  print("this is baz "..baz)
end

tmr.alarm( 0 , 15000 , 0 , launch )

``

Got

```
> this is foo bar
this is baz bazza
```

lorneb, 2015-12-01:
Ahh hah moment,  after looking at the source it appears that a timer is used to execute the command,  so after waiting a little bit I do get the expected result.  This works

```
function launch()
  foo = "bar"
  baz = "bazza"
  cmd = "baz = foo"
  node.input(cmd)  
  print("this is foo "..foo)
  print("this is baz "..baz)

  tmr.alarm( 1 , 50 , 0 , somethingElse )
  
end

function somethingElse()
  print("this is foo "..foo)
  print("this is baz "..baz)
end

tmr.alarm( 0 , 15000 , 0 , launch )
```

marcelstoer, 2015-12-01:
In general, take note of #719. Why did you ask the same question on SO at http://stackoverflow.com/q/34012453/131929 _after_ it seems to be resolved (based on your last comment)?

lorneb, 2015-12-01:
@marcelstoer The reason why I also asked on SO was because after I posted here I found and read #719 . Then this thread picked up so I continued my thoughts and findings here.  When I get a chance I will update also update SO with my findings, close this issue and keep everyone happy

closed 2015-12-01 by lorneb