nodemcu/nodemcu-firmware #805: memory leak if use wifi.sta.setip

nyuszis, 2015-12-01:
Only change one line in my code 
if wifi.sta.setip is call before connect then memory decrease every connect
if use dhcp memory remain

jmattsson, 2015-12-01:
I just tested a default `dev` build and I am not able to reproduce this issue:
```
> =wifi.sta.getip()
192.168.2.190	255.255.255.0	192.168.2.2
> =node.heap()
40640
> =wifi.sta.getip()
192.168.2.190	255.255.255.0	192.168.2.2
> =node.heap()
40640
> =wifi.sta.disconnect()
> =node.heap()
41128
> =wifi.sta.connect()
> =node.heap()
40640
> =wifi.sta.disconnect()
> wifi.sta.setip({ip="192.168.2.190",netmask="255.255.255.0"})
> =node.heap()
41128
> =wifi.sta.connect()
> =node.heap()
40800
> =wifi.sta.getip()
192.168.2.190	255.255.255.0	0.0.0.0
> =node.heap()
40800
> =wifi.sta.disconnect()
> =node.heap()
41128
> =wifi.sta.connect()
> =node.heap()
40800
> =wifi.sta.disconnect()
> wifi.sta.setip({ip="192.168.2.190",netmask="255.255.255.0",gateway="192.168.2.2"})
> =node.heap()
41128
> =wifi.sta.disconnect()
> =node.heap()
41128
> =wifi.sta.connect()
> =node.heap()
40800
> =wifi.sta.connect()
> =node.heap()
40800
> 
```

nyuszis, 2015-12-01:
the build i use release eMCU 0.9.6 build 20150704  powered by Lua 5.1.4

marcelstoer, 2015-12-01:
See #719.

closed 2015-12-01 by marcelstoer