nodemcu/nodemcu-firmware #809: Custom build from source does not work with nodemcu devboard v1.0

tasdomas, 2015-12-01:
I'm trying to build nodemcu from source, because why not ;-]

Unfortunately, once I flash the built image to the devboard, it does not work - I can connect to the serial port, but get not prompt and can see the board periodically restart (at least that's what I assume it does, since the led blinks every ~10 seconds).

My order of commands is as follows:
git clone ...
git checkout dev
tar -xzvf tools/esp-open-sdk.tar.gz
export PATH=/[DIR]/nodemcu-firmware/esp-open-sdk/xtensa-lx106-elf/bin:$PATH
make
make flash
picocom -b 9600 /dev/ttyUSB0
(at this point I get a blank serial session and can't send any commands to the dev board)

The log of my session (verbose) is here: http://pastebin.com/feP4fLKm

If I flash the stock nodemcu image or a custom built one from http://nodemcu-build.com/, it works just fine.

jmattsson, 2015-12-02:
Are you flashing with DIO mode? The devkit 1.0 hardware changed from the old QIO (quad IO lines to/from the flash chip) to DIO (dual IO lines) in order to free up two more pins for GPIO. Flashing in QIO mode means things don't boot. So, if your esptool isn't already set to DIO by default, try adding -fm dio to the write_flash command.

eyaleb, 2015-12-02:
A bit OT: I do not need the extra pins (GPIO9/10) and most of my current modules do not have access to these, so how do I continue to use flash in QIO mode?

jmattsson, 2015-12-02:
AFAIK it's not possible, short of soldering directly onto the appropriate pins (and possibly removing any external pull-ups/down on said pins). I'd be happy to be proven wrong though.

nickandrew, 2015-12-02:
Hi @eyaleb !

As I understand it, QIO connects 2 more GPIO pins (9 and 10) to the flash. for a total of 6 connections, so if your device has those connections then you (must?) use QIO mode; otherwise you must use DIO mode.

I believe the ESP-01 modules use QIO and the recent NodeMCU dev boards are DIO.

jmattsson, 2015-12-02:
In particular, the ESP-12E modules became DIO, whereas the original ESP-12 were QIO. The devkit-1.0 I have has a 12E.

eyaleb, 2015-12-02:
So, if I understand correctly, old modules that have all the flash lines connected for QIO **will** be used in QIO mode, and newer ones where only two data lines connect the flash **will** use DIO.

I originally thought (wrongly) that
*The flash is still configured (and soldered) with all 4 data lines, two of which also may connect to the module (new) edge. It is then with software that one can switch the flash to DIO to allow using the external pins. As such, at boot time QIO is used until the user selects another mode*

jmattsson, 2015-12-02:
@eyaleb Alright, after discussing this with a colleague here I think you're right about all the lines being hooked up even on an ESP-12E. The module layout images we've seen certainly suggests this. That then leaves the DIO vs QIO as something which depends on the actual flash chip loaded. I just got reminded that the last batch of ESP-12 (not 12E) modules we got did not support QIO, which again supports the argument that it is flash-chip-dependent.

The boot ROM starts out using single (or maybe DIO) mode, reads the first sector which contains the flash configuration flags (speed, dio/qio, size), and then uses whatever it finds there.

devsaurus, 2015-12-02:
I can back @jmattson's statement. Recently got a NodeMCU v1.0 clone that can be flashed with either dio or qio mode in esptool.py. I decided for QIO, thus trading in pins for speed. Might change in the future if I need the additional pins.

tasdomas, 2015-12-02:
You're absolutely right, @jmattsson  - was using QIO mode when I should have been using DIO - now it works. Many thanks!

jmattsson, 2015-12-02:
You're welcome!

closed 2015-12-02 by jmattsson

randmor, 2017-05-27:
I also had this problem. After success with Adafruit and WeMos boards, I decided to buy 10 Amica brand NodeMCU boards from Banggood/GeekCreit. Boy was I disappointed to find they would not flash correctly with MicroPython. I even verified that they worked okay under Arduino C/IDE w/ESP8266 Core. The ran the Arduino apps just fine. So, I started searching the Internet for others who had this problem. Up on the MicroPython ESP8266 forum I found someone who had this problem and a fix, so I tried it, then documented my findings in detail. You may want to read that set of postings. Here's the URL:

https://forum.micropython.org/viewtopic.php?f=16&t=2958

Now I can run MicroPython just fine (maybe a bit slower than Quad-I/O SPI, but I'll have 2 more GPIO pins when I figure out which pins GPIO9 and GPIO10 correspond to on the board.