nodemcu/nodemcu-firmware #810: Clean up modules for consistent LTR approach

jmattsson, 2015-12-02:
Note to self: As discussed in #796, I need to do a sweep of all the modules and tidy up the LTR aspects to get some consistency, and make it easier for new devs to Do The Right Thing.

TerryE, 2015-12-12:
@jmattsson Johny, this is a big patch. I need to review it properly, and not when I am having a Saturday morning cuppa in bed and working on my tablet! :laughing: 

But, I've replied on the issue, because I wanted to air the principle for wider comment amongst the other active contributors. I am sure that most will trust that we can age the details between us, following those agreed principles.

I suggest that these principles are:

*     We are mandating LTR at optimisation level 2 for all standard modules within our distribution. What individual developers do for their own add-on libraries I'd up to them.
*     We are including your non-aligned flash memory patch, and making us of it by moving all default .text segments into flash.
*     We are including your refactoring of the common string and allocation macros to remove unnecessarily duplicated routines.
*     We are including my LCD (packed info) patch at a default stripdebug level 2. That is source compiles dump local and upvalue debug info, and retain packed line debug information. As this drops the overhead of running from source compared to **lc** files to less than 10%.

Since your are tidying up the source, I would like to propose another principle, and that is that we kept the differences between the eLua code base and nodeMCU to a necessary minimum.  An example here is that nodeMCU has an almost global replace of strlen to c_strlen, but to avoid pulling in two different strlen functions we now have strlen as a linker synonym for c_strlen, so why do this bulk edit and make 100s of unneeded difference?

Anyway, I would appreciate your and others thoughts.

jmattsson, 2015-12-12:
The patch is mostly large in size rather than content. A pattern will stand out very quickly once you start reading it... but a tablet might not be the best device for that!

As for the listed principles, I'm in agreement. Plus I'd love to see the addition of the EGC changes/tweaks you're working on once they're ready - the performance improvement seemed quite remarkable from what you mentioned so far.

I'm all for keeping differences minimal between NodeMCU and upstream eLua (hence my comment in the PR). When large changes can be avoided through smaller changes via the pre-processor or linker-script, that's typically where I'll lean.

TerryE, 2015-12-13:
Johny, This patch is sufficiently complex that it merits some review comments and possibly a second iteration.  

*    The broad strategy that we seem to be implementing is that for the core elements which are also in the eLua code base, we are adopting a minimum change goal, so leaving in the `LUA_OPTIMIZE_MEMORY`  + `MIN_OPT_LEVEL` defines and their use unchanges, save for the additional per file `#undef MIN_OPT_LEVEL` statements to allow these to compile with the files to compile with  `MIN_OPT_LEVEL` specified at a compile macro level.
*    All of the modules will be built assuming `MIN_OPT_LEVEL==2` and `LUA_OPTIMIZE_MEMORY==2`.  Given this, we should include a standard `#if ... #error ... #endif` check to throw a compiile error is the developer dicks around with these and breaks this assumption.
*    I support ditching the extra complication module defines in the `auxmods.h` file, but given that we've done this, the only stub left in this is the macro definition for `MOD_CHECK_ID` so we should move this definition somewhere else and get rid of this header file entirely.

I've added this into my 5d653df.  Please see what you think.

jmattsson, 2015-12-15:
My turn to say "whoa, that is a big patch!" ;)

Looks good to me, the only change I'll propose is `s/build/built/g`. Would you like me to cherry-pick this onto my PR, or would you prefer to raise another PR and close mine?

TerryE, 2015-12-15:
No, just cherry pick what you want. Happy with any changes that you propose.

jmattsson, 2015-12-16:
Alright, I ended up being inspired and took the next step in terms of simplifying module registration.

Rather than trying to shoehorn everything using the pre-processor as was the case, I've reworked it to use a bit of pre-processor and a bit of linker script. The end result is that:
  - There is no longer any need for `modules.h` and its copy/paste proclivity.
  - Modules are now self-contained. Just include the new `module.h` file and add a `NODEMCU_MODULE(MYMOD, "mymod", mymod_map, luaopen_mymod);` line and you're set to go.
  - The `NODEMCU_MODULE` macro automatically honors the corresponding `LUA_USE_MODULES_MYMOD`, no extra coding needed.
  - Support both `#define LUA_USE_MODULES_MYMOD` as well as `#define LUA_USE_MODULES_MYMOD 1`, so should be fully backwards compatible.
  - The `luaopen_mymod()` function is optional - no more unnecessary empty functions.
  - No constant checking of Lua mem/optimize settings - you have to try *really* hard to do the wrong thing now.
  - `app/modules/linit.c` looks a lot neater as well.

I'll leave it to you guys to decide whether I've gone overboard with this. Personally, I think it's nice :)

devsaurus, 2015-12-16:
+1

Gone will be the days when you added a new module and it refused to show up. Could take longer to track this down than adding all the functionality.

We shouldn't postpone the merge for too long - PRs are already beginning to pile up.

jmattsson, 2015-12-16:
I'm hoping this might also bring @TerryE 's dream of drop-in'able modules a bit closer to reality.

TerryE, 2015-12-16:
Johny, I am on my tablet at the mo. I'll go through this at the desk, do a build and test and it is ok as I expect, then I will do the merge.

closed 2015-12-20 by jmattsson

TerryE, 2015-12-20:
Johny, I might want to reopen this, because as I felt when meeting the PR comments, this is a major step in the right direction, and works, so the PR was. Well worth merging into dev. However, I suspect that there is a bit of more tidying to be done -- lower priority stuff. Let me go through the final solution, and if I do think that we should do white then I'll reopen the issue as a placeholder.