nodemcu/nodemcu-firmware #811: GPIO state doesn't change after restart

vudangngoc, 2015-12-02:
I'm not sure this is normal behavior or a bug. When I pull down GPIO2 (wire it with GND) and GPIO.read(4) it return 0, and I restart my ESP and GPIO.read(4) still return 0. I did other test, wire GPIO2 with 3.3V, GPIO.read(4) return 1 and still return 1 after I restart my ESP.
This is first time I work with GPIO. Thank you for any support in advance <3

karrots, 2015-12-04:
Sounds like its reading the correct values. What are you expecting?

Also see #719

vudangngoc, 2015-12-04:
Actually, I don't know what is normal behavior of GPIO reading. Should GPIO change to default status (0 or 1) after restart?

karrots, 2015-12-04:
GPIO will read whatever state the GPIO is in. In your example it reads 0 when pulled to ground and 1 when pulled high. This is normal behavior. You may want to check out the forums for FAQs or beginner information.

vudangngoc, 2015-12-04:
thank you for your information :)

closed 2015-12-04 by vudangngoc