nodemcu/nodemcu-firmware #813: Assert lvm.c:463: L->top == L->ci->top || luaG_checkopenop(i)

pjsg, 2015-12-04:
This is running essentially the tip of dev with some extra changes to enable asserts: The relevant information is:
```
L->top=0x3fff0d54, L->ci->top=0x3fff0d4c, i=8388638, check=0
lvm.c:463: L->top == L->ci->top || luaG_checkopenop(i)
```
I.e. neither clause is true (hence the assert). I printed out all the values....

My repro steps:

* reset the platform
* use nodemcu-uploader.py to upload the httpserver.lua that I attached to the other issue
* once complete, I use esplorer and press 'open' so that it connects to the platform 
* it automatically fires in '=node.heap'
* The assert fires.

```
NodeMCU firmware detected.
=node.heap()
L->top=0x3fff0d54, L->ci->top=0x3fff0d4c, i=8388638, check=0
lvm.c:463: L->top == L->ci->top || luaG_checkopenop(i)
> 
```
The node.heap() output is *not* displayed. Otherwise things appear to work fine.

TerryE, 2015-12-04:
I went through your steps and I have debug in lvm.c and got the following:

```

Communication with MCU...
Got answer! AutoDetect firmware...

NodeMCU firmware detected.
=node.heap()
41776
> =node.heap()
41776
> dofile"httpserver.lua"
> =node.heap()
41504
> 
----------------------------
httpserver.lua  : 6230 bytes
test.lua        : 72 bytes
----------------------------
Total file(s)   : 2
Total size      : 6302 bytes

Total : 52961 bytes
Used  : 7028 bytes
Remain: 45933 bytes
```
The server won't get very for are the other components aren't on SPIFFS.

pjsg, 2015-12-04:
Hmm. I'll add some more logging to my build (i'll save each character as received in a circular buffer) just in case there is some noise/bogus characters being inserted during the switchover between the uploader.py and esplorer starting. I'll aim to get a reproducible test case that doesn't involve any switching from one environment to the other.

I'm out of town for the next few days so I won't be able to do this till the back end of next week). I didn't bring my boards with me!

You don't need to run the webserver at all to experience the problems (at least for me).

TerryE, 2015-12-05:
@pjsg, @jmattsson I though that you might be interested in my  enhanced developers lua_assert in my repo (see 676831d).  It's a bit of a hack, hence it's nowhere near dev but this will give you a lot more information on an assert failure.  

As you will see I've added a couple of extra defines in `user_config.c` (ignore the other changes which just save space on a minimal build).  You enable `lua_assert()` on a per module basis -- as per [these lines](TerryE/nodemcu-firmware/commit/676831dc0afa8b9a5d7cfd75ccb167f3d2527879#diff-4f88f55aa387440543ca826102471be5R17) in `lvm.c`.  The extra change is to force an assert failure in in the **^** operator for testing purposes.

The main assert code is currently in `lua.c` but this could probably be shifted into ldebug.c. The assert generates `((assert_exp)? (void)0 :  (lua_assert_traceback(L, __FILE__,__LINE__), (void) 0))` .  I've dropped printing the assert expression since we get the file and line no, but added some extra content for example: 

```
NodeMCU 1.4.0 build 20151006 powered by Lua 5.1.4 on SDK 1.4.0
lua: cannot open init.lua
> function a(i)
>> b = i*i
>>  return b, i^2 end
> function c() local x = a(2) end
> c()
Assert failure at lvm.c:553 at 0x00404091 (3 of 6)

stack traceback:
        stdin:1: in function 'c'
        stdin:1: in main chunk

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 29792, room 16 
tail 0
chksum 0x94
load 0x3ffe8000, len 2268, room 8 
tail 4
chksum 0xf0
load 0x3ffe88dc, len 8, room 4 
tail 4
chksum 0x0d
csum 0x0d
```
You get:
*  The C file and line number
*   A Lua traceback
*   The instruction being executed and the PC \#i of \#n for the current function.

This last plus a `luac.cros -l` will give you the actual VM instruction being executed at the failure.  These should help to narrow the failure.

TerryE, 2018-06-23:
@pjsg I think that this is dead, but can I ask you to close it.

closed 2019-04-10 by TerryE