nodemcu/nodemcu-firmware #816: Initial rBoot bootloader implementation

raburton, 2015-12-05:
After discussion in #806 (and as previously requested in #477) I have prototyped rBoot integration into nodemcu. You can find it here: https://github.com/raburton/nodemcu-firmware/tree/rboot

This allows you to have two versions of nodemcu-firmware on the device at the same time, switch between them and perform over the-air-updates (OTA).

You can use the following commands to interact with it:
`print(rboot.rom())` - prints the current running rom number (0 or 1).
`rboot.swap()` - change the active rom and reboot into it
`rboot.ota("url")` - perform an OTA update, note: currently url is ignored and update is fetched from a hard coded url! actually using the url can be implemented later...

You will need:
- A nodemcu devboard or other esp8266 device with 2mbyte flash or larger. Support for 512k devices is impossible (no way to fit two 450k roms on a 512k flash), 1mb devices is just about possible but requires a slightly different approach and more changes to the build system.
- esptool2 from https://github.com/raburton/esp8266/tree/master/esptool2 add it to your path or define the env var esptool2 to point to the binary.
- A decent webserver to serve the OTA rom from (Apache/IIS would be fine, some people have had trouble with python web server modules).

Instructions:
- Checkout the rboot branch from my nodemcu-firmware fork.
- Edit app/rboot/rboot-ota.h and set the ip address of your ota webserver.
- Build as normal.
- Flash rboot.bin to 0x00000 and eagle.app.v6.bin to 0x02000.
- Copy eagle.app.v6.bin to the root of your webserver and rename it to rom0.bin.
- Run `print(rboot.rom())` and check it returns zero.
- Connect to wifi.
- Run `rboot.ota("url")` and wait a minute, a second copy of the rom will be flash and the device will reboot into this rom.
- Run `print(rboot.rom())` and check it now returns one.

Not yet implemented:
- Actually using the url passed in the ota command (just pass anything for the time being).
- Spiffs selection. At the moment it will probably always use the spiffs at the end of the first rom, a little extra code needs adding to add an offset to choose the second spiffs when booting the second rom.

raburton, 2015-12-12:
Ok, finished off the prototype. Sorry for the delay, I've been pretty busy.

Spiffs is now supported, each of the two roms gets it's own spiffs after the rom (position configurable) and the correct one is mounted depending on the rom in use. I've removed the url parameter from the ota method, I realised it didn't really make sense to include that. There are pro and cons, but mostly it's not needed and doing it properly would require a better understanding of the internals of the lua interpreter than I currently have. The instructions above have been updated.

Not actually had any feedback though, so I wonder if people are really that interested!

nickandrew, 2015-12-13:
@raburton I'm very interested, just not in a position to try anything at this time.

robertfoss, 2015-12-14:
I'm very tempted to try this feature out.

Is it possible to include a SPIFFS filesystem with the upgrade? And also, would that filesystem be R/W?

raburton, 2015-12-15:
Yes, there are two ways to add the spiffs to the OTA.
1) Easiest. Append the spiffs to the nodemcu firmware (with sufficient padding between them for the spiffs to land in the right place on the flash). The position of the spiffs is defined in user_config.h so it would be a matter of adding a little padding to the end of the rom up to this position, then appending the spiffs bin file (remember the starting point for the rom is 0x2000, not at zero, on the flash). Then the spiffs will be updated automatically in a single operation, along with the rom.
2) slightly more effort. Add a second call to the ota function (slightly modified) to make a separate update of the spiffs area of the flash. This should probably be on a separate call e.g. rboot.otaspiffs() so that you have the option of just updating one or the other. Actually not that big a deal to implement.

Any spiffs written this way would be read write as normal, but remember that if you are going to update the spiffs in this way each time you will loose anything that has been written to it by the user.

raburton, 2015-12-17:
@robertfoss I've added an rboot.otafs() method that will pull a spiffs bin file and write it to the other roms spiffs area (not the spiffs of the currently running rom), so you can do that alongside the actual rom update. Also fixed a little bug where the spiffs for the second rom was in the wrong place on a 4mb flash.

robertfoss, 2015-12-17:
@raburton Those are killer features. Thank you for taking the time to implement them.

raburton, 2016-01-12:
It's been a month and I've had no feedback. Doesn't look like anyone is very interested.

jmattsson, 2016-01-12:
That was the impression I had in my discussion thread too. I think it's just too early at this time. If you don't mind, keep your changes around and maybe down the track interest will grow/resume/something?

SeanBannister, 2016-01-19:
Was just searching for this feature, great to see someone working on it :) unfortunately I'm not familiar with building NodeMCU so haven't had a chance to test it out.

someburner, 2016-01-19:
Hi @raburton . I am very interested in this. I am using my own flavor of NodeMCU + Esp-ginx + SPIFFS. I just got done implementing the file-serving methods using SPIFFS and am now working on implementing OTA updates, so I will definitely be looking into this. I was actually pretty concerned that I would have to figure out how to implement this myself until I saw this.. I should have feedback for you sometime this week.

jmattsson, 2016-01-19:
@someburner do feel free to have a play with the competing approach too, over in #806 (branch ref [in this comment](https://github.com/nodemcu/nodemcu-firmware/issues/806#issuecomment-161465798)).

someburner, 2016-01-19:
@jmattsson thanks for the heads up. Is there an example of loading either of these without LUA modules? Do I need to use rtctime_early_startup() or is there a way I can call (link?) the bootloader from user_main() ? I'm still a novice when it comes to the linker and bootloader.

raburton, 2016-01-19:
> Was just searching for this feature, great to see someone working on it :) unfortunately I'm not familiar with building NodeMCU so haven't had a chance to test it out.

I think this is the problem, nodemcu is a distributed as a binary which users flash once and forget about, only working with their apps afterwards. Unlike, for example sming or sdk apps, where the framework forms part of the same rom. Sure, you can get nodemcu as source, but unless you are looking to fix it or add to it why would you? And perhaps much of it's target audience running are just learning a bit of programming with lua scripts so wouldn't be in a position to do these things anyway.

To be useful it doesn't need to be something users play with, it needs to be built in to the official rom and the ota updates put on the projects own server so that users can just use it when a new version is released.

raburton, 2016-01-19:
> Hi @raburton . I am very interested in this. I am using my own flavor of NodeMCU + Esp-ginx + SPIFFS. I just got done implementing the file-serving methods using SPIFFS and am now working on [...]

Should be easy enough to add the spiffs ota as well, most if not all of the code you need to do it will already be in that branch. (Although another option to make it even easier is to append the two roms and update as a single unit.) Let me know if you need a hand.

Sming uses rBoot to do ota of the app and spiffs filesystems. There is a sample [sming app]( https://github.com/SmingHub/Sming/tree/develop/samples/Basic_rBoot) that demonstrated the use of this using the Sming [rBootHttpUpdate class](https://github.com/SmingHub/Sming/blob/develop/Sming/SmingCore/Network/rBootHttpUpdate.cpp), itself a wrapper for rboot-api, if you need more examples of use.

And as @jmattsson says there are other ways too, see what suits you best if you're doing something custom.

willd, 2016-01-25:
Heya!

I seemingly succeeded to both build and flash your fork, but after "Booting rom 0" I get nothing more on baud 74880 and nothing on 9600.

This is the unit I'm trying to run it on: http://www.aliexpress.com/item/rgb-strip-WiFi-controller-1-port-control-15-rgb-lights-communicate-with-Android-phone-to-dim/32301423622.html.

That said, I haven't been able to run anything but the original firmware and v0.9.2.2 AT on it so far.
esptool.py flash_id says it is a e0 4014 flash, which according to this: https://github.com/esp8266/Arduino/commit/5d7a3e939c00706d6318272a0c0216d1e426b2eb is a 1Mbyte flash.

I would love some help on the matter :)

raburton, 2016-01-25:
> That said, I haven't been able to run anything but the original firmware

That's probably not a good starting point. This devices uses a non-standard pin for the serial port, so you'll probably need to modify nodemcu to get any serial output once it's booted. Funnily enough I have one of these and had to do the same with sming, before I discovered how bad pwm is on the esp. I'm in the process of designing my own board, with hardware pwm and bigger mosfets. PCB almost done, just waiting for parts from China to test a breadboard prototype.

willd, 2016-01-25:
How does that work with the bootloader? I can flash, and verify that it works. Does the bootloader push RX on all pins during download mode?

raburton, 2016-01-25:
Bootloader, rom boot messages and rBoot output will go to both serial lines, but once the sdk has initialised the serial ports you need to output to them individually (as I understand it), which is what NodeMCU will be doing under the covers. Had same problem with Sming, as soon as I got past rBoot it looked like it had hung, but it was just that all output was going to uart0 while uart1 was actually wired to the header pins. rBoot runs fine on them though (no reason it shouldn't of course), had that and Sming going fine, only gave up on them when I discovered how flaky pwm is (also probably need a bit more current. Once I have the device finalised I'll publish the designs. With cheap boards from dirtypcbs and a few parts off aliexpress, probably work no more than $10 a piece (I'm not planning on selling them, making them will be up to anyone who wants one).

raburton, 2016-02-07:
FYI, I've updated the rBoot branch in my fork (https://github.com/raburton/nodemcu-firmware/tree/rboot) to latest master + rBoot implementation (obviously). I thought it was worth keeping it up to date, although few seem interested at present it might come in handy at some point in the future.

jfollas, 2016-02-15:
@raburton FYI: I've been keeping a close eye on this as a potential strategy for my device. Initially, I was convinced that device updates could be managed with just Lua files, but that locks out future enhancements (or bug fixes) that may require new binaries. So, it's good to have an OTA update strategy for firmware itself when devices are deployed to the field.

devyte, 2016-03-07:
@raburton @jmattsson I'm also keeping an eye on this, as well as #806. At the end of my project I'll have something like 40 nodemcu devices deployed around my house, so doing serial updates would be extremely time consuming. Having OTA updates would allow flashing all devices via some automated process.

someburner, 2016-05-11:
@raburton 

I just got time to start playing with this. Here's where I'm at: I'm not using the latest NodeMCU source and mine has branched off significantly since I'm not interested in any Lua features. I updated my rboot folder (apis as well as the main rboot folder), and have everything compiling and running on a 4MB ESP12 with rBoot 1.4, split config. I re-wrote the OTA API to use my own implementation and it *kind of* works. But my issue is not with the OTA I don't think.

I have read through your blog and readmes several times but am still having trouble wrapping my mind around the memory mapping. The actual addressing seems straightforward enough, and I've playing around quite a lot with SPIFFS in the past. But basically, whenever I try to switch ROMs, do an OTA, or anything like that, it crashes-- although I'm pretty sure there is a good reason for it:

I looked through rboot-bigflash.c and I'm fairly certain that Cache_Read_Enable_New is not being linked properly because uncommenting the ets_printf doesn't give me anything. But since it's all mapped I'm not sure if there's another way I can check whether that is indeed the case.

Is there any chance you could shed a little more light on the linking process for Cache_Read_Enable? From what you've written it doesn't sound like there's a definitive way to accomplish it. But perhaps you have some new info on that?  The other thing I'm thinking is that I'm still a novice when it comes to makefiles and I'm not sure where to add in extra things (if it's even necessary). 

I noticed in the more recent NodeMCU builds there is a prototype for Cache_Read_Enable(uint32_t b0, uint32_t b1, uint32_t use_40108000) and I'm curious about that as well. Any help on any of this would be greatly appreciated, and thanks again for making the port!

TriAnMan, 2016-05-31:
@raburton @marcelstoer
I'm very interested in OTA feature and trying to build NodeMCU with it.
I'm using a docker image [provided in docs](http://nodemcu.readthedocs.io/en/dev/en/build/#docker-image) but get an error:
```
CC rboot-api.c
CC rboot-bigflash.c
CC rboot-ota.c
AR .output/eagle/debug/lib/rboot.a
xtensa-lx106-elf-ar: creating .output/eagle/debug/lib/rboot.a
CC .output/eagle/debug/image/eagle.app.v6.out
/opt/esp-open-sdk/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot find -lmain2
collect2: error: ld returned 1 exit status
make[1]: *** [.output/eagle/debug/image/eagle.app.v6.out] Error 1
make: *** [.subdirs] Error 2
```
I assume this error arises because I must add `esptool2` somehow to the docker image.
Could you help me to do this?

raburton, 2016-05-31:
Someone else had this same issue with docker, but we never did work out why. It's not esptool2 related, although you will need that too. libmain2.a is created from libmain.a using the objcopy command (it just has one symbol weakened to allow a function to be overridden). The make file should be doing this for you, and if you look higher up the log you should see it happen, but for some reason it wasn't getting found. Have a look in the build dir and you should find the file. One easy option it so just copy it to the sdk lib folder, along side the original libmain.a, and it should find it then.

someburner, 2016-05-31:
@TriAnMan  I think it's because you don't have libmain2 copied over, or it's not in the right directory. Here's how mine looks:

```
# Base directory for the compiler
OPENSDK_ROOT 		?= /home/jeffrey/esp-open-sdk-1.5.3
XTENSA_TOOLS_ROOT ?= $(OPENSDK_ROOT)/xtensa-lx106-elf/bin/
SDK_ROOT 			?= $(OPENSDK_ROOT)/sdk/
PROJ_ROOT = $(abspath $(dir $(CURDIR)/$(word $(words $(MAKEFILE_LIST)),$(MAKEFILE_LIST))))/

# Base directory of the ESP8266 SDK package, absolute
SDK_BASE				?= $(abspath $(OPENSDK_ROOT)/esp_iot_sdk_v1.5.3)

............

.PHONY: all .subdirs rboot

ifndef PDIR
# targets for top level only
RBOOT=bin/rboot.bin
LIBMAIN_SRC = $(SDK_ROOT)lib/libmain.a
LIBMAIN_DST = $(PROJ_ROOT)sdk-overrides/lib/libmain2.a
endif

all: .subdirs $(OSPIFFY) $(LIBMAIN_DST) $(RBOOT) $(OBJS) $(OLIBS) $(OIMAGES) $(OBINS) $(SPECIAL_MKTARGETS)

$(LIBMAIN_DST): $(LIBMAIN_SRC)
	@echo "OC $@"
	@$(OBJCOPY) -W Cache_Read_Enable_New $^ $@

............

bin/rboot.bin: rboot/firmware/rboot.bin
	$(Q) cp $^ $@

rboot:
	$(Q) mkdir -p ${PROJ_ROOT}/sdk-overrides/lib;
	$(OBJCOPY) -W Cache_Read_Enable_New ${SDK_ROOT}lib/libmain.a ${LIBMAIN_DST}
	$(Q) $(MAKE) -C ${RBOOT_SRC_DIR}
```

The "............" is to show some separation. But these are the relevant sections that I have in my Makefile to automatically copy libmain and make a lib directory in sdk-overrides.

TriAnMan, 2016-05-31:
I've just manually copied `$(SDK_ROOT)lib/libmain.a` to `$(PROJ_ROOT)sdk-overrides/lib/libmain2.a` and got another error. (:
```
CC rboot-api.c
CC rboot-bigflash.c
CC rboot-ota.c
AR .output/eagle/debug/lib/rboot.a
xtensa-lx106-elf-ar: creating .output/eagle/debug/lib/rboot.a
CC .output/eagle/debug/image/eagle.app.v6.out
../sdk-overrides/lib/libmain2.a(app_main.o): In function `Cache_Read_Enable_New':
(.text+0x3c): multiple definition of `Cache_Read_Enable_New'
rboot/.output/eagle/debug/lib/rboot.a(rboot-bigflash.o):/opt/nodemcu-firmware/app/rboot/rboot-bigflash.c:37: first defined here
collect2: error: ld returned 1 exit status
make[1]: *** [.output/eagle/debug/image/eagle.app.v6.out] Error 1
make: *** [.subdirs] Error 2
```

raburton, 2016-05-31:
libmain2 isn't an exact copy of libmain, there wouldn't be much point in that. See the part of the Makefile that generates it, you need to use the objcopy command with the -W option, as below:
```
$(LIBMAIN_DST): $(LIBMAIN_SRC)
    @echo "OC $@"
    @$(OBJCOPY) -W Cache_Read_Enable_New $^ $@
```

TriAnMan, 2016-05-31:
@raburton my steps with docker was:
1. Install docker (I've removed an old image just to be shure with `docker rmi -f marcelstoer/nodemcu-build`)
2. Clone `git@github.com:raburton/nodemcu-firmware.git`
3. Checkout rboot branch
4. Run `docker run -e INTEGER_ONLY=1 --rm -ti -v `pwd`:/opt/nodemcu-firmware  marcelstoer/nodemcu-build`
So you can try to reproduce my error.

> you need to use the objcopy command with the -W option
I'll try. (:

TriAnMan, 2016-05-31:
I've made an objectcopy and got another error:
```
CC rboot-api.c
CC rboot-bigflash.c
CC rboot-ota.c
AR .output/eagle/debug/lib/rboot.a
xtensa-lx106-elf-ar: creating .output/eagle/debug/lib/rboot.a
CC .output/eagle/debug/image/eagle.app.v6.out
E2 ../bin/eagle.app.v6.bin
CC rboot-stage2a.c
LD build/rboot-stage2a.elf
/opt/esp-open-sdk/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot open linker script file ../ld/eagle.rom.addr.v6.ld: No such file or directory
collect2: error: ld returned 1 exit status
make[1]: *** [build/rboot-stage2a.elf] Error 1
make: *** [.subdirs] Error 2
```

I ran docker with `docker run -e INTEGER_ONLY=1 --rm -ti -v `pwd`:/opt/nodemcu-firmware -v /var/git/esptool2:/var/git/esptool2 -e PATH=/var/git/esptool2:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin marcelstoer/nodemcu-build`

raburton, 2016-05-31:
This is found relative to the lib path too, so looks like something in your environment (library path related) is a bit screwy.

oyooyo, 2016-06-02:
I just want to point out that I too am extremely interested in this.

I consider OTA updates to be the one killer feature still missing in the official nodemcu-firmware, the one missing feature that would make the difference between nodemcu-firmware basically being a toy for developers with too much spare time, or something generally ready to be seriously used, even by common people/a mass market.

ESP8266 with nodemcu-firmware is about the most fascinating combination of hard- and software I've come across in many years, and OTA is the one missing piece to finally make it the perfect IoT platform. Keep up the great work, @raburton , this has to find it's way into the official nodemcu-firmware as quickly as possible!

raburton, 2016-06-02:
@TriAnMan All works fine on a normal Linux system, so I don't know what the problem is with the docker environment, but I'm sure you can move around the missing file till it finds it.

@oyooyo thanks, hope you find it useful, but as you say unless it finds it's way into the main source, or an alternative solution, it's not nearly as handy

oyooyo, 2016-06-02:
@raburton I've seen that @jmattsson has provided another solution for OTA updates in #806. Now I'm a bit undecided which solution I should try out, I'm not even sure if I really understand the main differences between jmattsons approach and yours...

The one major difference that I believe to see between your solutions is that your approach takes care of the whole actual OTA transfer (and requires the URL to download the OTA image from to be specified at compile time), while jmattsons solution does not handle the actual OTA transfer at all, is really agnostic to how the firmware image is being transferred, and instead basically just provides low-level functions for flashing a new firmware-image via LUA. Did I get that right?

If so, I actually prefer jqmattsons agnostic approach, as it seems more flexible to me.

On the other hand, I got the impression that your rBoot bootloader has more features and is generally a little more advanced than the bootloader used by jqmattson. Would you agree on that?

The thing is, I'm wondering if it would be possible to combine your (seeming advanced) rBoot  with jqmattsons approach of a low-level OTA API that is agnostic to the actual transport, with relatively little effort.
Because I currently imagine the optimal solution to be something like this:
rBoot bootloader + a module written in C that pretty much just provides low-level access for flashing/switching between images, but is agnostic to the OTA transport (so similar to jqmattsons "otaupgrade" module) + an **optional** high-level module (written in LUA?), that works similar to your rboot module by providing a convenient function like rboot.ota(<firmware_image_url>) for simply downloading and installing a firmware file via HTTP, sufficient for most users. I believe that would combine the best of both worlds, the convenience and simplicity of your rboot.ota function while still allowing perfect flexibility for those who for some reason prefer a different approach for fetching the new firmware image.

Does that idea even make sense, or is it a stupid idea for some reason?

raburton, 2016-06-02:
> The one major difference that I believe to see between your solutions is that your approach takes care of the whole actual OTA transfer (and requires the URL to download the OTA image from to be [...]

rBoot doesn't care how the ota is done, it just works with what's on the flash. I do supply sample code for the SDK to perform OTA updates, and have wrapped that sample OTA code with a lua class to allow it to be used easily from nodemcu. You can write your own OTA method, using the rBoot API or from scratch, if you wish to do things differently. You could extend my sample implementation to take parameters instead of hardcoding the OTA details if you wanted. What I've provided is fully working, but it is just a starting point which I'd expect to be developed and brought more in line with the standards and requirements of nodemcu, if anyone ever wanted to integrate it into nodemcu properly. That's just the OTA part though. rBoot itself, the underlying bootloader, shouldn't need touching.

> On the other hand, I got the impression that your rBoot bootloader has more features and is generally a little more advanced than the bootloader used by jqmattson. Would you agree on that?

Yes, but I'm biased ;-) It is quite widely used though, I'm aware of projects in some well known large companies using it and it's the standard bootloader/OTA solution in Sming. rBoot has continued to be developed since I did this nodemcu integration too, so it now has even more features.

> The thing is, I'm wondering if it would be possible to combine your (seeming advanced) rBoot with jqmattsons approach of a low-level OTA API that is agnostic to the actual transport, with relatively little effort. Because I currently imagine the optimal solution to be something like this:
rBoot bootloader + a module written in C that pretty much just provides low-level access for flashing/switching between images...

I think that's pretty much how this is laid out already. rBoot is there as the bootloader, rBoot API (written in C) provides APIs to write to flash and interact with the rBoot config sector, and then on top of that is the sample rBoot HTTP OTA code (also written in C) and on top of that is a small lua module to call the OTA code. You can replace any number of those layers in order (top down), e.g. lua module, lua and OTA or lua, OTA and API layer and still be using rBoot. I'd suggest keeping the API layer and improving the OTA/lua code to be more lua-ish and use more of the existing nodemcu framework. As I don't use nodemcu myself I'm not the best person to write these top two layers, but I did write a basic version. As I said I did expect, if it were adopted, this would get rewritten by someone more familiar with the internals of the framework.

djphoenix, 2016-06-03:
Feature proposal: OTA update with user-specified URIs (from optional lua-level arg). That will make OTA setup simple without firmware rebuild. Also, update from spiffs may be loaded (e.g. from device connected locally, not only http).

djphoenix, 2016-06-04:
So, using latest esp-open-sdk (1.5.2) I failed to run both original repo and merged changes to current master. Compilation success, but when run I received this log:

```
load 0x40100000, len 1828, room 16 
tail 4
chksum 0xb4
load 0x3ffe8000, len 832, room 4 
tail 12
chksum 0x80
csum 0x80

rBoot v1.2.1 - richardaburton@gmail.com
Flash Size:   4 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: irom chksum

Writing default boot config.
Booting rom 0.
system param error
00 00 00 00 00 00 55 00 55 00 00 00 55 00 00 00 55 00 00 00 55 00 00 00 55 00 00 00 55 00 ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
35 81 00 00 38 e1 00 00 00 01 73 74 61 74 65 2e 6c 75 61 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 92 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
35 01 00 00 7c 6c 6f 63 61 6c 20 73 74 61 74 65 20 3d 20 7b 7d 0a 0a 73 74 61 74 65 2e 6e 61 6d 
65 20 3d 20 27 41 44 27 20 2e 2e 20 6e 6f 64 65 2e 63 68 69 70 69 64 28 29 0a 73 74 61 74 65 2e 
74 6f 6b 65 6e 20 3d 20 63 72 79 70 74 6f 2e 68 61 73 68 28 27 4d 44 35 27 2c 20 6e 6f 64 65 2e 
63 68 69 70 69 64 28 29 20 2e 2e 20 77 69 66 69 2e 73 74 61 2e 67 65 74 6d 61 63 28 29 29 0a 0a 
73 74 61 74 65 2e 6f 6e 20 3d 20 66 61 6c 73 65 0a 73 74 61 74 65 2e 73 70 65 65 64 20 3d 20 31 
0a 73 74 61 74 65 2e 63 6f 6c 6f 72 20 3d 20 30 0a 73 74 61 74 65 2e 61 6d 62 69 61 6e 63 65 20 
3d 20 66 61 6c 73 65 0a 73 74 61 74 65 2e 64 69 6d 20 3d 20 31 30 30 0a 0a 72 65 74 75 72 6e 20 
73 74 61 74 65 0a ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
55 80 00 00 78 fa 00 00 00 01 74 6f 6f 6c 73 2e 6c 75 61 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 94 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
55 00 00 00 fc 6c 6f 63 61 6c 20 74 6f 6f 6c 73 20 3d 20 7b 7d 0a 0a 6c 6f 63 61 6c 20 63 6f 6d 
70 69 6c 65 20 3d 20 66 75 6e 63 74 69 6f 6e 28 29 0a 20 20 6c 6f 63 61 6c 20 6c 69 73 74 20 3d 
20 66 69 6c 65 2e 6c 69 73 74 28 29 0a 20 20 6e 6f 64 65 2e 73 74 72 69 70 64 65 62 75 67 28 33 
29 0a 20 20 6c 6f 63 61 6c 20 70 72 6f 74 65 63 74 65 64 20 3d 20 7b 0a 20 20 20 20 27 69 6e 69 
74 2e 6c 75 
00 00 55 00 00 00 55 00 00 00 55 00 00 00 55 00 55 80 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
55 80 00 00 78 d6 06 00 00 01 74 6f 6f 6c 73 2e 6c 75 61 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 94 00 95 00 97 00 99 00 9b 00 9d 00 9f 00 ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
[... 18 lines cut ...]
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
55 00 08 00 fc 72 6c 65 6e 20 3d 20 36 30 2a 36 30 2a 32 34 2a 33 36 35 20 65 6e 64 0a 20 20 65 
[... cut ...]
20 6d 6f 6e 74 68 2c 20 73 65 63 20 3d 20 6d 6f 6e 74 68 20 2b 20 31 2c 20 73 65 63 20 2d 20 6d 
6f 6e 74 68 6c 65 6e 0a 20 20 20 20 69 66 20 6d 6f 6e 74 68 20 3d 3d 20 �������*�64 20 79 65 
61 72 20 25 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

... (some resets...) ...

rBoot v1.2.1 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: irom chksum

fail 9: 210, 215
Rom 0 is bad.
fail 4
Rom 1 is bad.
No good rom available.
user code done
```

raburton, 2016-06-05:
> 
rBoot v1.2.1 - richardaburton@gmail.com
Flash Size:   4 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: irom chksum

>Writing default boot config.
Booting rom 0.
system param error
... lot of hex values...

You have the flash size set to 4mb. The hex dump you are getting is from the sdk because it has found invalid data in it's config area (last 4 sectors of flash, based on the size set), probably part of the rom because you've set a flash size so small. You need to make sure you flash with the -fs 32m option.

djphoenix, 2016-06-05:
OK, now I booted into rom, rboot OTA seems to work. So I anyway want to boot latest master cause 1.4.0 is outdated. Will you spend some time to rebase changes on master?

djphoenix, 2016-06-05:
Merged on master, received this:
```
load 0x40100000, len 1808, room 16 
tail 0
chksum 0xe4
load 0x3ffe8000, len 832, room 8 
tail 8
chksum 0xd0
csum 0xd0

rBoot v1.2.1 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  80 MHz
rBoot Option: Big flash
rBoot Option: irom chksum

Writing default boot config.
Booting rom 0.
sp 0x3ffef5b8 
sf_dump[0] a0: 0x4010fca5  a1: 0x3ffef5b8  a2: 0x2c002800  a3: 0x34003000 
epc1=0x40259078, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000002, depc=0x00000000
Fatal exception (28): 
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00007ff0, depc=0x00000000
Fatal exception (28): 
epc1=0x40001800, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00007ff0, depc=0x00000000
```

djphoenix, 2016-06-07:
Another side effect of OTA is separated filesystems. In case of store some user data (e.g. persistant configuration) in flash, after update, we cannot access it directly. @raburton do you have an idea about file transfer between roms?

someburner, 2016-06-07:
@djphoenix 

Just map SPIFFS to a fixed spot that will not get overwritten and use for all ROMs. Or after switching ROMs copy the old filesystem to a new location on first boot

djphoenix, 2016-06-07:
@someburner

Good idea, but when we need to update both binary and lua/lc files (preserving config files) it fails...

someburner, 2016-06-07:
@djphoenix Ahh. Yeah I don't use Lua so I don't know how that works. But anything in SPIFFS should copy over just fine

raburton, 2016-06-09:
@someburner I've updated the code to latest master, the problem you were hitting was caused by a change to the alignment of the irom segment in master (for reasons I cannot imagine it was made to align to 0x1000 bytes which was pushing it out of place).

djphoenix, 2016-06-10:
@raburton perfect! I have wrote some features like file transfer between roms, and (partially) duplicate current rom (for update just fs or binary, preserving current as other part). Will share it later after some tests.

djphoenix, 2016-06-13:
@raburton after some debugging: net.cert.verify is not works (reports "bad offset"). I tried to patch nodemcu.ld to align cert pages to sector, invocation successes, but after reset rboot tells me:
```
> net.cert.verify(cert)
.... all ok ....
.... reset ....

Communication with MCU...
..Got answer! AutoDetect firmware...
Communication with MCU established.
Can't autodetect firmware, because proper answer not received (may be unknown firmware). 
Please, reset module or continue.
  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 1392, room 16 
tail 0
chksum 0xa2
load 0x3ffe8000, len 688, room 8 
tail 8
chksum 0xe8
csum 0xe8

rBoot v1.4.1 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  20 MHz
rBoot Option: Big flash
rBoot Option: irom chksum

Rom 0 is bad.
Rom 1 is bad.
No good rom available.
user code done
```

raburton, 2016-06-13:
I have no idea what net.cert.verify does, but if it causes rBoot to reject the rom on next boot it must be writing something to the flash within the rom area.

djphoenix, 2016-06-13:
@raburton there is some interesting: [docs](https://nodemcu.readthedocs.io/en/dev/en/modules/net/#netcertverify), [implementation](https://github.com/nodemcu/nodemcu-firmware/blob/master/app/modules/net.c#L1593).
It actually rewrites a sector of code.

devyte, 2016-06-13:
I seem to remember @pjsg adding certificate verification not too long ago. Maybe he can provide some insight?

raburton, 2016-06-13:
Wow, it rewrites a sector of the rom! That's crazy and will obviously break any checksum of the rom. Either disable irom checksum, which which is a bad idea, or find a better way to handle the certificate issue. I'd suggest embedding it at compile time (which appears to be an option, and can still be updated once deployed by ota update of the whole rom) or if you want it to more dynamic save the data to a spiffs or to a sector separate to the rom image.

devyte, 2016-06-30:
So I'm confused. Is this working or not? 

I'm looking for an OTA upgrade method. Upgrading firmware for all my devices is currently a huge pain...
Currently I could get the new firmware bins onto the spiffs over wifi via an http server. So what I'm missing is something that will take the bins and write them to flash, then reboot.

raburton, 2016-06-30:
> So I'm confused. Is this working or not? 

Yes, the only thing that doesn't is the net.cert.verify function (because it rewrites part of the rom). Better to embed the ca info at build time instead of trying to do it dynamically.

pjsg, 2016-07-01:
Yes -- the espressif sdk is somewhat borked and requires that the certificate (and the private key if needed) is accessed directly from the ROM. However, I think that this may be fixed in SDK 1.5.4 -- there is a way to pass in pointers to the certificate data. 

In the short term, I'd use the option to build the certificate into the ROM (during build) and not use the runtime flavor. In the future, if the SDK supports it, then we could make the runtime setting of the certificate not have to modify the rom.

jmattsson, 2016-07-01:
Additionally, if we switch to mbedTLS things should be substantially more sane.

ymxcc, 2016-07-01:
I have a try, the code for firmware update is pretty good running on my nodeMCU.

However, I wonder that how to update the script through cloud.

@raburton , if possible, could you help me about that!

Thanks so much!

someburner, 2016-07-01:
@pjsg 

For me it was working fine for SDK 1.5.2 & 1.5.3.1 but got compile errors with 1.5.3.2 & 1.5.3.3

```
home/jeffrey/esp-open-sdk-1.5.3.3/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libpp.a(pm.o): In function `pm_set_sleep_time':
(.irom0.text+0xd0): undefined reference to `rf_cal_sec'
../sdk-overrides/lib/libmain2.a(user_interface.o): In function `wifi_softap_dhcps_stop':
(.irom0.text+0x1338): undefined reference to `get_fpm_auto_sleep_flag'
../sdk-overrides/lib/libmain2.a(user_interface.o): In function `wifi_station_dhcpc_start':
(.irom0.text+0x13ef): undefined reference to `get_fpm_auto_sleep_flag'
collect2: error: ld returned 1 exit status
```

1.5.4 also works fine. Given that they've now updated 1.5.3 several times for things related to rf_cal and sleep, I think it's probably a good idea to avoid using 1.5.3 altogether. 

@ymxcc

There are several ways to do it. @raburton has an example for using HTTP for updates. I've written MQTT and Websockets based versions of it and have actually used public MQTT brokers to OTA ESPs for fun haha. If anyone's interested I could make a demo repo.

ymxcc, 2016-07-01:
@someburner, thanks for your reminding. For me, I am very interested in your program using MQTT and websockets based versions of @raburton . It sounds good. If possible, could you share it to me?

I found some solutions for that like so:

function load_file (fname, ftxt, cmpl)
file.remove(fname)
file.open(fname,"w")
file.write(ftxt)
file.flush()
file.close()
if string. sub(fname,-3,-1)=="lua" and cmpl==true then
node.compile(fname)
file.remove(fname)
end

I found that the two main functions used for the remove and flash the new script to the nodeMCU with node.write() and node.compile(), am I right?

You said your version is based on @raburton 's one. Could you give a link for me?

Thanks so much, Jeff!

someburner, 2016-07-01:
@ymxcc 

My [esp-rfm69](https://github.com/someburner/esp-rfm69) project is loosely based on NodeMCU firmware (started ~SDK 1.4.1) and I have a websocket uploader that you can drag-and-drop files to the ESP. I haven't pushed up any code for rboot OTA version of this and I cut out Lua completely, so I'd have to take some time to make a Lua version. But I think it shouldn't be too hard. Again, never used Lua before, can someone comment on what is required for loading Lua scripts? Are they just stored in SPIFFS? 

Are you looking more for MQTT OTA or websocket file upload? I think MQTT would probably be better for most cases, especially with public brokers being available.

ymxcc, 2016-07-01:
My task is to provide the Lua Script Language for customers, you know they could operate it like node.js, whatever they like to do, it is so easy. To achieve the goal, for my work, I have to integrate all modules into the firmware. Including the OTA not only for the firmware update but the script. I am very lucky, I found @raburton 's code for firmware update. Sure, the most important thing is let the firmware could update using OTA from cloud, It worked now.

The left is how to update the script through cloud. In fact, I just want to find a shortcut, I mean it is better for me if I could find a good code from Github, I could save more time.

Correct! I used the MQTT to publish a message to inform the Node who need to update now, then updating.... Maybe, like @raburton 's code, the node receive the request, then to start up a request to webserver for the correct downloading script.

mikewen, 2016-07-01:
This OTA is soooo useful for large projects or products that target end
consumers.

@raburton, thanks for your great work.

Anything I can do to help? Maybe I can start to add custom URL?

I would like to stick with NodeMCU for my WiReboot
<https://www.kickstarter.com/projects/786298545/wireboot-keep-your-wifi-on-24-7-automatically-rebo>
project.
(I was thinking to switch to Arduino just because OTA update feature.)

[22 more replies]

closed 2019-08-04 by stale[bot]