nodemcu/nodemcu-firmware #820: Tag the repository

nickandrew, 2015-12-06:
Now that 1.4.0 is in master branch, I suggest a tag so git-describe will be useful.

I tagged https://github.com/nickandrew/nodemcu-firmware with 1.4.0-20151201 at the current master commit.

```
tag 1.4.0-20151201
Tagger: Nick Andrew <nick@nick-andrew.net>
Date:   Sun Dec 6 15:06:59 2015 +1100

NodeMCU 1.4.0 release 2015-12-01

This is the commit at which the work-in-progress 'dev' branch
replaced the 0.9.6 'master' branch.

```

So now:

```
$ git describe
1.4.0-20151201-29-g5977633
```

marcelstoer, 2016-06-29:
We started tagging the repository. However, since we did that on GitHub (so far) the tags are not annotated and this in turn means `git describe` still shows 0.9.6. Either we start tagging with `git` and then push the tags or your have to run `git describe --tags` (or `--all`).

marcelstoer, 2016-06-29:
While it is possible to [convert a lightweight tag to an annotated tag](http://sartak.org/2011/01/replace-a-lightweight-git-tag-with-an-annotated-tag.html) it's a no-go once the tag has been released into the wild. So, I just made a mental note that we should use annotated tags from now on (`git tag -a 	
1.5.1-master_20160731 -m ""` and `git push`).

closed 2016-06-29 by marcelstoer