nodemcu/nodemcu-firmware #822: wrong in http_server

chegongcai, 2015-12-07:
stack traceback:
[C]: in function 'createServer'
http_server.lua:118: in main chunk
[C]: in function 'dofile'
stdin:1: in main chunk

TerryE, 2015-12-07:
@chegongcai, please see #719. What build are you running? You have to provide more information, and clearly show what is output. Otherwise we will close this issue .

chegongcai, 2015-12-07:
OK，I will provide more information as soon as possible.

chegongcai, 2015-12-08:
Hi, there are more information:
You successfully commissioned a NodeMCU custom build from the master branch. You selected the following 10 modules: node file gpio wifi net tmr adc uart mqtt hx711.(I have choose SSL)
And then I test  a simple http server:
srv=net.createServer(net.TCP,30) 
srv:listen(80,function(conn) 
conn:on("receive",function(conn,payload) 
print(payload) 
conn:send("Hello, NodeMCU.")
end) 
end)
And then the result is:
dofile("web.lua")
web.lua:2: only one tcp server allowed(it seems dowload the second time should show this)
stack traceback:
	[C]: in function 'createServer'
	web.lua:2: in main chunk
	[C]: in function 'dofile'
	stdin:1: in main chunk
what should I do about the stack traceback?

vudangngoc, 2015-12-08:
"web.lua:2: only one tcp server allowed" --> maybe you started another tcp server before, only one tcp server allowed

chegongcai, 2015-12-08:
@vudangngoc  I know the reason of  "web.lua:2: only one tcp server allowed",but  I don't know about stack traceback.

vudangngoc, 2015-12-08:
root cause doesn't come from this line of code, it because you started a tcp server before, close it or restart your esp before create other one

chegongcai, 2015-12-08:
@vudangngoc  while I do test on "nodemcu_float_0.9.6-dev_20150704",it just show:only one tcp server allowed,and then I can connect my router by web(192.168.4.1). when I  test on the "a NodeMCU custom build from the master branch",it always show:stack traceback.I don't know why?

chegongcai, 2015-12-08:
@TerryE what should I do can resolve the  problem about:
 stack traceback:
[C]: in function 'createServer'
web.lua:2: in main chunk
[C]: in function 'dofile'
stdin:1: in main chunk
I had try many NodeMCU custom build,but the problem still there.

chegongcai, 2015-12-08:
@TerryE  can you help me ?

nickandrew, 2015-12-08:
@chegongcai You are seeing the stack traceback because of a previous error. Not sure if the traceback you showed is for 1.4.0 or 0.9.6.

0.9.6-dev-20150704 is no longer supported.

TerryE, 2015-12-08:
The traceback was added in my patch dddf46e at the end of August.  This simply adds back in standard Lua error reporting functionality that was previously suppressed.  All it's telling you is that the createServer() call on line 2 of your `web.lua` is throwing an error.  If you want to suppress this and handle it programmatically, then use a `pcall()` or `xpcall()`, but you would also be better off tracknig down the underlying source of the error.  Note also #624 which I've just reopened.

chegongcai, 2015-12-09:
@nickandrew  this wrong will not happen when I test the http_server on 0.9.6-dev-20150704,  I don't think it is because of a previous error.

jmattsson, 2015-12-09:
The stack trace merely tells you the call path that led to the actual error. It is not itself an error. It is a debugging aid. It was not present in the old 0.9.6 version. It is a new feature to help users locate the actual error.

chegongcai, 2015-12-09:
@jmattsson  If it is not itself an error, why I can success when I used "nodemcu_float_0.9.6-dev_20150704" to connect my router by web(192.168.4.1). when I use "a NodeMCU custom build from the master branch", I can not to log in 192.168.4.1(the same lua). there are the information after I reset:
NodeMCU custom build by frightanic.com
	branch: master
	commit: 432f698fd80e0ff90ea4f209b66c77d5105ea55a
	SSL: true
	modules: node,file,gpio,wifi,net,tmr,adc,uart,mqtt,hx711
 build 	built on: 2015-12-08 01:06
 powered by Lua 5.1.4 on SDK 1.4.0
192.168.4.1	255.255.255.0	192.168.4.1
please open 192.168.4.1 then input ssid and password

TerryE, 2015-12-09:
@chegongcai, it does sound as if you have a version dependent issue here. But the traceback is just a diagnostic and not the actual issue itself. What you seeing is a _standard_ Lua feature. ALibrary is throwing a Lua error, in this case ` only one tcp server allowed` and you are not catching it with a `pcall()` or `xpcall()`. 

This isn't a bug. It's standard Lua.

closed 2015-12-09 by TerryE