nodemcu/nodemcu-firmware #823: u8glib draw functions doesn't work inside loop

pastukhov, 2015-12-07:
Here is my code:
```lua
    local strCoords = {}
    strCoords[1] = {x=30,y=32}
    strCoords[2] = {x=30,y=46}
    strCoords[3] = {x=5, y=60} 

    disp:firstPage()

repeat
      while data.desc:len() > 0 and strNumber < 4 do
            data.desc = data.desc:sub(13)
            disp:drawStr(strCoords[strNumber].x,strCoords[strNumber].y,data.desc:sub(1,12))
            strNumber = strNumber +1
      end
until disp:nextPage() == false
```
And it is draw nothing, is is a bug?

Any advice how to split an long string and print it?

devsaurus, 2015-12-07:
You're modifying the image before it's actually drawn.
Don't change parameters in the drawing loop which affect the information that's being displayed. U8glib functions expect they're called exactly the  same way between `firstPage()` and `nextPage() == false`.

rickvanbodegraven, 2015-12-07:
What I don't understand is how people keep opening up issues and swamping the maintainers (or even the collaborators) with work that is completely unrelated to actual issues with the codebase. Is it not clear to people that Github issues are not the place for this type of question? *sigh* sorry for the rant, but I am subscribed to this repo and I keep finding notifications like this in my inbox and it is making me a bit sad.

marcelstoer, 2015-12-07:
See #719.

closed 2015-12-07 by marcelstoer